zephyrproject-rtos/zephyr #44262: mimxrt1050_evk: build time too long for this platform

LixinGuoX, 2022-03-28:
**Describe the bug**
After this PR(#44150) merged, the total build and run time for mimxrt1050_evk has been extended a lot, from approximately 4~5 hours to 7~8 hours.
And found the the biggest change is in the build stage, time cost almost doubled.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p mimxrt1050_evk --device-testing --device-serial /dev/ttyACM0 -T tests/ -T samples/ - j 1

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit 40526f853e0643ab76fb401ba5c8ae50cd926bd2

LixinGuoX, 2022-03-28:
Hi @danieldegrasse, any idea on this?

nashif, 2022-03-28:
"-j 1" is really bad, you should try without -j1, not directly related to the issue and commits above, but this will speed up things considerably, i.e. build on process while you are flashing and testing on another.

nashif, 2022-03-29:
@gmarull  @carlescufi  FYI

LixinGuoX, 2022-03-29:
Without '- j 1' , it can speed up to 90min, will then reset back and see real effect of the patch above.

mbolivar-nordic, 2022-03-29:
@danieldegrasse have a look at `/omit-if-no-ref/` here: https://elinux.org/Device_Tree_Source_Undocumented and see if adding it to your pinctrl nodes might work for you

danieldegrasse, 2022-03-29:
@mbolivar thanks for the hint- build time seems to improve with that change, from 49 to 37 seconds. `devicetree_unfixed.h` goes from 103959 SLOC to 41148 SLOC. I've opened #44319 to include the new pin control headers, with that property present.

PR #44319 (2022-03-29): [DNM] manifest: hal_nxp: update NXP hal to use new pin control headers
Pin control DTSI files need /omit-if-no-ref/ property to reduce
generated devicetree size and improve build times.

Fixes #44262

Dependent on #43826, due to the ordering of commits in NXP's HAL
commit dcac0ce0 (2022-03-29): manifest: hal_nxp: update NXP hal to use new pin control headers

Pin control DTSI files need /omit-if-no-ref/ property to reduce
generated devicetree size and improve build times.

Fixes #44262
closed 2022-03-29 by mbolivar-nordic