zephyrproject-rtos/zephyr #44281: Bluetooth: Use hardware encryption for encryption

LingaoM, 2022-03-29:
I noticed that `bt_encrypt` uses `tc_aes` is used for software encryption. Why not use HCI to hand over to the controller for hardware encryption? This saves a lot of CPU occupation and stack space, and can improve the sending and receiving performance of mesh

carlescufi, 2022-03-29:
cc @trond-snekvik @jhedberg @joerchan

carlescufi, 2022-03-29:
As far as I know, no particular reason other than no one has implemented it yet. On Nordic devices we do use the hardware encryption, but directly and not via HCI.

jhedberg, 2022-03-29:
@LingaoM I'd be surprised if HCI over a physical transport is faster than doing the calculation on the local CPU. It sounds like the cipher is not properly optimised or you have a slow CPU. Normally, when optimizing execution one tries minimize IO-bound operations, but this would do the opposite.

joerchan, 2022-03-29:
What kind of setup is this anyway?

In zephyr we have a method for faster encryption / rand through the BT_CTLR_CRYPTO instead of BT_HOST_CRYPTO.
Only use-case for this feature request is in a host-only build where you want save code-space by not having encryption in the host image.
Offloading encryption to the controller chip in this case is highly unlikely to result in any sending or receiving performance increase.

closed 2022-10-07 by LingaoM