zephyrproject-rtos/zephyr #44299: Backport Request of PR (#38720): drivers: virt_ivshmem: Allow multiple instances of ivShMem devices to v2.7-branch

michael1schmidt, 2022-03-29:
**Describe the bug**
Backport PR (#38720): drivers: virt_ivshmem: Allow multiple instances of ivShMem devices to v2.7-branch

multiple instances of ivshmem device are used for Safety systems based on Zephyr. This feature is important for LTS Zephyr.

michael1schmidt, 2022-03-29:
Hi. Actually its a bug fix as the current code if ivshmem allows only one instance of a ivshmem device to be enumerated by the OS. If the zephyr OS has to handle several instances this won't work with current code.

If there is a multi VM system setup it is a common use case to use several ivshmem channels to exchange data between different VMs. In additional to this data exchange it should also be fulfilled that there is isolated between channels that are not intended for a specific VM. This can be reached only with several instances of ivshmem devices.

In general there should not be a limitation by OS to enumerate only one instance of a device where no further physical limitation is limiting it.

cfriedt, 2022-04-15:
Fixed by #44332

closed 2022-04-15 by cfriedt