zephyrproject-rtos/zephyr #44324: Compile error in byteorder.h

surfskidude, 2022-03-29:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/44190

<div type='discussions-op-text'>

<sup>Originally posted by **surfskidude** March 24, 2022</sup>
I cloned the latest today and I got a compile error when I compiled my existing code, which includes net/socket.h

In file included from /home/build/zephyrproject/zephyr/include/net/net_ip.h:27,
                 from /home/build/zephyrproject/zephyr/include/net/socket.h:27,
/home/build/zephyrproject/zephyr/include/sys/byteorder.h:22:36: error: expected declaration specifiers or ‘...’ before ‘(’ token
   22 | #define __bswap_16(x) ((uint16_t) ((((x) >> 8) & 0xff) | (((x) & 0xff) << 8)))
      |                                    ^
/home/build/zephyrproject/zephyr/include/sys/byteorder.h:26:36: error: expected declaration specifiers or ‘...’ before ‘(’ token
   26 | #define __bswap_32(x) ((uint32_t) ((((x) >> 24) & 0xff) | \
      |                                    ^
/home/build/zephyrproject/zephyr/include/sys/byteorder.h:36:36: error: expected declaration specifiers or ‘...’ before ‘(’ token
   36 | #define __bswap_64(x) ((uint64_t) ((((x) >> 56) & 0xff) | \
      |                                    ^

Build:
west build -p auto -b native_posix_64 myproject
</div>

henrikbrixandersen, 2022-04-05:
Please include further details. Which host OS, which compiler etc. All of this is requested if you fill out the bug report issue template.

mbolivar-nordic, 2022-04-05:
Closing for now due to insufficient information to categorize this as a bug. Please reopen with additional information if you believe this to be a bug, including steps to reproduce.

surfskidude, 2022-04-05:
If you look at the end of the above you see: west build -p auto -b native_posix_64 myproject

surfskidude, 2022-04-05:
Just include net/socket.h in any C file and it will fail with native_posix_64

mbolivar-nordic, 2022-04-07:
Does not reproduce on my machine.

```
$ git diff
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 987728bbf2..c442b25b75 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -5,6 +5,7 @@
  */
 
 #include <zephyr.h>
+#include <net/socket.h>
 
 void main(void)
 {
(west-0.13) scarlet: ~/zp/zephyr (west-project-maintainers) mbolivar
$ west build -p -b native_posix_64 samples/hello_world 
-- west build: making build dir /home/mbolivar/zp/zephyr/build pristine
-- west build: generating a build system
loading initial cache file /home/mbolivar/zp/xtra/initial-cache.cmake
Loading Zephyr default modules (Zephyr base).
-- Application: /home/mbolivar/zp/zephyr/samples/hello_world
-- Found Python3: /home/mbolivar/.virtualenvs/west-0.13/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/mbolivar/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/mbolivar/zp/zephyr)
-- Found west (found suitable version "0.13.0a1", minimum required is "0.7.1")
-- Board: native_posix_64
-- Found dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /home/mbolivar/zp/zephyr/boards/posix/native_posix/native_posix_64.dts
-- Generated zephyr.dts: /home/mbolivar/zp/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/mbolivar/zp/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/mbolivar/zp/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/mbolivar/zp/zephyr/build/zephyr/dts.cmake
Parsing /home/mbolivar/zp/zephyr/Kconfig
Loaded configuration '/home/mbolivar/zp/zephyr/boards/posix/native_posix/native_posix_64_defconfig'
Merged configuration '/home/mbolivar/zp/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/mbolivar/zp/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/mbolivar/zp/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/bin/gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mbolivar/zp/zephyr/build
-- west build: building application
[1/97] Preparing syscall dependency handling

[2/97] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/home/mbolivar/zp/zephyr), build: zephyr-v3.0.0-2355-ga54006612448
[93/97] Linking C executable zephyr/zephyr_pre0.elf

[97/97] Linking C executable zephyr/zephyr.elf
```

de-nordic, 2022-04-12:
`byteswap.h` collides with `byteorder.h` (tested on amd64), because `byteorder.h` defines preprocessor macro named `__bswap_16` and `byteswap.h` tries to define inline function of the same identifier; the order of inclusion causes the identifier in function definition, from `byteswap.h`, to be substituted with replacement-list of macro `__bswap_16`, from `byteorder.h`.
Then compiler kicks in find broken function definition and complains.
Do you really need the `byteswap.h`?

surfskidude, 2022-04-12:
We are using a common code base on multiple platforms. The net/socket is included by a special zephyr porting layer. byteswap.h is included by the generic code.

If you look at my initial comments at the top of this thread, we did not have this problem until recently so something changed in zephyr that broke this.

rlubos, 2022-04-13:
Why did you assign me to this? It's not a networking problem.

de-nordic, 2022-04-13:
@surfskidude What version of Zephyr, which commit are you using as base?
The files, `net_ip.h` and `byteorder.h`, does not seem to be touched in 3 or more years, in lines that could affect this.
Is it possible to just #ifdef inclussion of `byteswap.h` for Zephyr platform?

mbolivar-nordic, 2022-04-13:
> Why did you assign me to this? It's not a networking problem.

@aescolar I assigned you because it is a native posix issue

mbolivar-nordic, 2022-04-13:
> If you look at my initial comments at the top of this thread, we did not have this problem until recently so something changed in zephyr that broke this.

.. or something changed in your underlying platform, etc.

Notice how @de-nordic carefully specified their platform. It would really help if you could do the same

mbolivar-nordic, 2022-04-15:
You may be convinced of that, and you are probably right, but I am trying to explain the difference between 1. you being right and 2. you convincing the upstream Zephyr developers that you are right by providing them with an example that they can actually reproduce instead of just repeating yourself ad nauseam.

andrewleech, 2022-09-19:
For what it's worth, I encountered the same error message when compiling sections of the zephyr codebase on linux x64, gcc 9.4.0 (not using Zephyr SDK).

By comparing compile arguments between standard zephyr / west builds for `native_posix_64` and my custom Makefile I narrowed the issue down to `-std=c11`

If I compile without `-std=c11` it throws these error in `zephyr/sys/byteorder.h`, eg. 
```
../../lib/zephyr/include/zephyr/sys/byteorder.h:20:36: error: expected declaration specifiers or ‘...’ before ‘(’ token
   20 | #define __bswap_16(x) ((uint16_t) ((((x) >> 8) & 0xff) | (((x) & 0xff) << 8)))
```

With `-std=c11` it compiles fine.

mkettn, 2022-11-22:
Hello,
I stumbled upon the same problem also for BOARD=native_posix[_64] ... on the contrary: BOARD=qemu_x86 works.
I get the same error and adding -std=c11 does not work for me, since I am using a C++ compiler.
zephyr is at most recent version on main (9ebb9abab7b0d662e0d5379367fb7f052959c545) and I am using the Zephyr SDK v0.15.1

Since the posix native target is not that crucial for me, I'll just go with qemu. However, the problem still persists.

aescolar, 2024-01-05:
For those facing this issue, and until it is fixed in Zephyr (where it should): One workaround which may work out of the box for you is to target native_sim instead of native_posix and select one of the embedded C libraries (for ex. CONFIG_PICOLIBC=y).

PR #67224 (2024-01-10): net: posix Fix name collision with __bswap
__bswap in zephyr/sys/byteorder.h conflicts with __bswap in host's byteswap.h. byteswap.h from host compiler used in posix_native_64 boards causes a compilation issue.

This commit renames __bswap to __z_bswap to prevent collision.

Before this commit a compilation error can be created by adding #include <byteswap.h> to samples/net/sockets/echo/src/socket_echo.c

This does not change external API to byteorder.h, but does change internal implementation which some other source files depend on.

Fixes #44324
commit 9c1a45cc (2024-01-10): posix: Fix name collision with __bswap

__bswap_ in zephyr/sys/byteorder.h conflicts with __bswap_ in host's
byteswap.h. byteswap.h from host compiler used in posix_native_64 boards
causes a compilation issue.

This commit renames __bswap_ to BSWAP_ to prevent collision.

Before this commit a compilation error can be created by adding #include
<byteswap.h> to samples/net/sockets/echo/src/socket_echo.c

This does not change external API to byteorder.h, but does change
internal implementation which some other source files depend on.

Replaced manual byteswap operations in devmem_service.c with A [... cut]
closed 2024-01-10 by fabiobaltieri