zephyrproject-rtos/zephyr #44327: proposal a "next" branch for fast iteration

ck-telecom, 2022-03-30:
## Introduction

For some reasons, some new feature or API is really slow for upstream, proposal a next branch like Linux for some new feature or new API for experiments, and this didn't affect main branch.

### Problem description

when some feature inspired, there is a more complicated way to integrate into main branch, and pend on pull request and stale

### Proposed change

new "next" branch for fast iteration

## Detailed RFC

each module or subsystem maintainer aware that feature in the next branch, and when is ready it should get cherry-picked from next to main in merge window.

### Proposed change (Detailed)

### Dependencies

### Concerns and Unresolved Questions

## Alternatives

stephanosio, 2022-03-31:
What you are talking about is essentially a feature branch -- there have been and will be ones like that for really major and/or big features.

dbaluta, 2022-03-31:
it's more than a feature branch. it's like the merge of all feature branches. 

> What you are talking about is essentially a feature branch -- there have been and will be ones like that for really major and/or big features.

Is more than a feature branch. It is the 'merge' of all feature branches. This could be useful for testing and speeding up development.

stephanosio, 2022-03-31:
We had and, if necessary, can still create `topic-*` branches (e.g. [`topic-sensors`](https://github.com/zephyrproject-rtos/zephyr/tree/topic-sensors)) -- these topic/feature branches seem to basically what @ck-telecom is describing.

stephanosio, 2022-03-31:
> we are talking branch is a 'merge all feature' branch

We already have `main` for that purpose.

A new feature, either in the form of a PR or a feature branch, is either accepted or not accepted. If you want to test out a new feature that has not been merged, you can check out the PR branch, or a feature branch if there is one.

In case of the Zephyr, during a merge window, `main` is already an unstable branch which is essentially what `linux-next` is.

During a merge freeze, `main` acts as a stable branch and there is not really an unstable branch (i.e. `linux-next`) anymore; but, that is a good thing because that forces us to redirect our resources more towards stabilisation than feature addition.

Linux kernel is an enormous project and the Zephyr Project is not nearly as big and/or resourceful as the Linux kernel project at this time.

In the future, when the project grows to such a magnitude that we can handle both stabilisation and feature additions simultaneously, it may make sense to create a stable `vX.Y-branch` at the beginning of a release merge "freeze", and keep `main` always open as an unstable branch; but as of now, I am opposed to this idea.

ck-telecom, 2022-03-31:
My original idea is about 'New API' is to slow, for a project,  checkout A feature branch to test feature A, checkout b branch to test feature B, and the project is fragment of feature branches

gmarull, 2022-03-31:
I'd personally not create a `next` branch for having experimental/incomplete features. I think we have topic branches for that. One can always have a fork with a branch combining multiple upstream topic branches if needed, or even to include custom patches that are not ready for upstream.

jfischer-no, 2022-03-31:
>For some reasons, some new feature or API is really slow for upstream, proposal a next branch like Linux for some new feature or new API for experiments, and this didn't affect main branch.

Linux has a bit different contribution structure. This suggestion does not make sense as long as we use Github as contribution front-end, here every PR is a branch and can be whatever next.

ck-telecom, 2022-04-01:
After all, I think one must maintain a Zephyr downstream for project

nashif, 2023-08-14:
@keith-zephyr  let's touch on this in the Process WG and close there based on the discussion and other discussions we had before in the TSC.

keith-zephyr, 2023-08-30:
@keith-zephyr Process WG has discussed topic branches and TSC has previously discussed creating a "next" branch during the stabilization period.
@nashif - Decision from TSC was to branch the release early, which opens up the main branch earlier in the process.
@nashif -  impact with LTS was also discussed.
@marc-hb - The Linux model uses "next" to merge many topic branches.  @henrikbrixandersen - Zephyr doesn't have topic branches so the "next" model doesn't apply.

--
Process WG conclusion is that a "next" branch isn't needed for Zephyr and doesn't fit existing workflows.

closed 2023-08-30 by nashif