zephyrproject-rtos/zephyr #44338: intel_adsp_cavs18: multiple tests failed due to incomplete log

LixinGuoX, 2022-03-30:
**Describe the bug**
Twister now has a stricter pass/fail criterion by adding test suite name verification, that exposed some problems of several test cases on  intel_adsp_cavs18/25.
For test cases below, logs are incomplete, missing the first half.
```
tests/lib/p4workq/
tests/subsys/logging/log_immediate/
tests/crypto/tinycrypt_hmac_prng/
```

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p intel_adsp_cavs18 --device-testing --device-serial-pty=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh -T tests/crypto/tinycrypt_hmac_prng/ -vvv

**Logs and console output**
```
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.8+ hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x141a00000 to contain 180224 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - DEVICE: ==
DEBUG   - DEVICE: HMAC-PRNG test_108:
DEBUG   - DEVICE: PASS - test_108 in 0.269 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_109:
DEBUG   - DEVICE: PASS - test_109 in 0.271 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_110:
DEBUG   - DEVICE: PASS - test_110 in 0.274 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_111:
DEBUG   - DEVICE: PASS - test_111 in 0.276 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_112:
DEBUG   - DEVICE: PASS - test_112 in 0.279 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_113:
DEBUG   - DEVICE: PASS - test_113 in 0.281 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_114:
DEBUG   - DEVICE: PASS - test_114 in 0.284 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_115:
DEBUG   - DEVICE: PASS - test_115 in 0.286 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_116:
DEBUG   - DEVICE: PASS - test_116 in 0.289 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_117:
DEBUG   - DEVICE: PASS - test_117 in 0.291 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_118:
DEBUG   - DEVICE: PASS - test_118 in 0.294 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_119:
DEBUG   - DEVICE: PASS - test_119 in 0.296 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: HMAC-PRNG test_120:
DEBUG   - DEVICE: PASS - test_120 in 0.299 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: All HMAC-PRNG tests succeeded!
DEBUG   - DEVICE: PASS - test_hmac_prng in 0.299 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite test_hmac_prng_fn succeeded
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION SUCCESSFUL
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit 199048b2ed45

nashif, 2022-03-31:
use CONFIG_BOOT_DELAY to delay start of test execution.

LixinGuoX, 2022-04-07:
This is a common issue for intel_adsp_cavs platform, CONFIG_BOOT_DELAY didn't help.
All the 3 failed cases share a common characteristic that the serial log is long. And the starting part are missing.

yerabolu, 2022-04-13:
@LixinGuoX : This issue can be closed as the issue with tests/crypto/tinycrypt_hmac_prng/ was resolved by #44669

LixinGuoX, 2022-04-15:
Yes, verified and close this issue.

LixinGuoX, 2022-06-23:
After change on this line(from 0.1s to 1s):
https://github.com/zephyrproject-rtos/zephyr/blob/d130160813fb52ee87a2d0c4a4fb8e57d466f181/soc/xtensa/intel_adsp/tools/cavstool.py#L472
We have this issue again, I have a PR: #46471, this is a workaround in testcase.
We can also apply config boot_delay in twister cmd, but this will extend the total execution time.

PR #47070 (2022-06-30): soc/xtensa/intel_adsp/tools: Dial back stream reset sleep
A full second is too much, and since "Ack local interrupt before
processing IPC" patch, it doesn't seem necessary. This whole second
wait would break tests that log too much content, as the firmware
is already running during the reset - and the host script wouldn't read
the contents, thus some of the logs would be overwritten.

---

Fixes #44338

This should address the (second coming) of https://github.com/zephyrproject-rtos/zephyr/issues/44338 - hopefully, without breaking https://github.com/zephyrproject-rtos/zephyr/issues/44925 agai [... cut]
closed 2022-06-30 by nashif