zephyrproject-rtos/zephyr #44342: Bluetooth: Audio: Codec config macros should not have LC3 prefix

Casper-Bonde-Bose, 2022-03-30:
**Is your enhancement proposal related to a problem? Please describe.**
Many of the macros in lc3.h (#define values) are belonging to Generic Audio assigned numbers and are not limited to LC3.

**Describe the solution you'd like**
They should be moved to audio.h or a (bt_codec.h, bt_assigned_numbers.h etc.) should be made with all the assigned numbers and the corresponding helper macros to use them. 
They should be treated similar to metadata macros, and if moved to a separate file, the metadata macros should also be moved..

**Additional context**
The list of assigned numbers can be found at https://btprodspecificationrefs.blob.core.windows.net/assigned-numbers/Assigned%20Number%20Types/Generic%20Audio.pdf

Casper-Bonde-Bose, 2022-03-30:
Somewhat related to #43202

Casper-Bonde-Bose, 2022-04-01:
When doing this cleanup the values should also be grouped into enums to allow referring to them as groups in the documentation.

Thalley, 2023-09-26:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/43098

closed 2023-09-26 by Thalley