zephyrproject-rtos/zephyr #44347: ACRN: multiple tests failed due to incomplete log

Zhaoningx, 2022-03-30:
**Describe the bug**
I tested some cases by the west, and I checked the log, I find some cases log is not complete(the first part is missing), those cases will be marked as failure cases by the twister even though the case is tested successfully. Using the case tests/kernel/mem_protect/mem_map/ as the demo. See below.

**To Reproduce**
west build -b acrn_ehl_crb tests/kernel/mem_protect/mem_map/ -p always
west flash /opt/remotehw/remotehw-x86-acrn.sh ehlsku11

**Logs and console output**
```
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
00000000000138a
E: RSI: 0x0000000000109bad RDI: 0x0000000000109750 RBP: 0x00000000001173c0 RSP: 0x00000000001173a0
E:  R8: 0x0000000000109330  R9: 0x0000000000109b94 R10: 0x0000000000000001 R11: 0x0000000000129240
E: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x00000000001173a0 RFLAGS: 0x0000000000010246 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x0000000000100924
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_z_phys_map_rw)
Caught system error -- reason 0
 PASS - test_z_phys_map_rw in 0.8 seconds
===================================================================
START - test_z_phys_map_exec
E: Page fault at address 0xff6000 (error code 0x11)
E: Access violation: supervisor thread not allowed to execute
E: PTE: 0xff6000 -> 0x0000000000129000: XD
E: RAX: 0x00000000001173a8 RBX: 0x0000000000129000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x0000000000000000 RDI: 0x000000000011739f RBP: 0x00000000001173c0 RSP: 0x0000000000117388
E:  R8: 0x0000000000000001  R9: 0x0000000000000003 R10: 0x0000000000000001 R11: 0x0000000000129240
E: R12: 0x0000000000000004 R13: 0x000000000010d8a0 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x0000000000117388 RFLAGS: 0x0000000000010202 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x0000000000ff6000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_z_phys_map_exec)
Caught system error -- reason 0
 PASS - test_z_phys_map_exec in 0.8 seconds
===================================================================
START - test_z_phys_map_side_effect
E: Page fault at address 0xff4000 (error code 0x3)
E: Access violation: supervisor thread not allowed to write
E: PTE: 0xff4000 -> 0x0000000000113000: XD
E: RAX: 0x0000000000ff4000 RBX: 0x0000000000129070 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x0000000000000000 RDI: 0x0000000000109750 RBP: 0x00000000001173c0 RSP: 0x00000000001173b0
E:  R8: 0x0000000000000001  R9: 0x0000000000000003 R10: 0x0000000000000001 R11: 0x0000000000129240
E: R12: 0x000000000010ce80 R13: 0x000000000010d8a0 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x00000000001173b0 RFLAGS: 0x0000000000010206 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x0000000000100098
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_z_phys_map_side_effect)
Caught system error -- reason 0
 PASS - test_z_phys_map_side_effect in 0.8 seconds
===================================================================
START - test_z_phys_unmap
E: Page fault at address 0xff2000 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000ff2000 RBX: 0x0000000000129098 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x0000000000000000 RDI: 0x0000000000109750 RBP: 0x00000000001173c0 RSP: 0x00000000001173b0
E:  R8: 0x0000000000000001  R9: 0x000000000010dc60 R10: 0x0000000000000000 R11: 0x0000000000129240
E: R12: 0x000000000010ce80 R13: 0x000000000010d8a0 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x00000000001173b0 RFLAGS: 0x0000000000010247 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x000000000010010a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_z_phys_unmap)
Caught system error -- reason 0
 PASS - test_z_phys_unmap in 0.8 seconds
===================================================================
START - test_k_mem_map_unmap
Free memory: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
mapped a page to 0xff2000
Free memory after mapping: 7118848
Free memory after unmapping: 7122944
E: Page fault at address 0xff2000 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000000000 RBX: 0x0000000000ff2000 RCX: 0x00000000000000f9 RDX: 0x0000000000109764
E: RSI: 0x0000000000109b0d RDI: 0x0000000000109750 RBP: 0x00000000001173c0 RSP: 0x00000000001173a0
E:  R8: 0x00000000001092f0  R9: 0x0000000000109ae3 R10: 0x00000000ffffffff R11: 0x0000000000000037
E: R12: 0x00000000006cb000 R13: 0x00000000006cb000 R14: 0x000000000000000a R15: 0x0000000000000000
E: RSP: 0x00000000001173a0 RFLAGS: 0x0000000000010246 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x0000000000100638
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_k_mem_map_unmap)
Caught system error -- reason 0
 PASS - test_k_mem_map_unmap in 0.20 seconds
===================================================================
START - test_k_mem_map_guard_before
mapped a page: 0xff2000 - 0xff3000
trying to access 0xff1ff8
E: Page fault at address 0xff1ff8 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000000001 RBX: 0x0000000000ff2000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x000000000000000a RDI: 0x0000000000112660 RBP: 0x00000000001173c0 RSP: 0x00000000001173b0
E:  R8: 0x0000000000000001  R9: 0x0000000000000000 R10: 0x00000000ffffffff R11: 0x0000000000000066
E: R12: 0x000000000010ce80 R13: 0x000000000010d8a0 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x00000000001173b0 RFLAGS: 0x0000000000010202 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x0000000000100771
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_k_mem_map_guard_before)
Caught system error -- reason 0
 PASS - test_k_mem_map_guard_before in 0.9 seconds
===================================================================
START - test_k_mem_map_guard_after
mapped a page: 0xfef000 - 0xff0000
trying to access 0xff0008
E: Page fault at address 0xff0008 (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x0000000000000001 RBX: 0x0000000000fef000 RCX: 0x0000000000000001 RDX: 0x0000000000000001
E: RSI: 0x000000000000000a RDI: 0x0000000000112660 RBP: 0x00000000001173c0 RSP: 0x00000000001173b0
E:  R8: 0x0000000000000001  R9: 0x0000000000000000 R10: 0x00000000ffffffff R11: 0x0000000000000066
E: R12: 0x000000000010ce80 R13: 0x000000000010d8a0 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x00000000001173b0 RFLAGS: 0x0000000000010202 CS: 0x0018 CR3: 0x0000000000134000
E: RIP: 0x00000000001006dc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10ce80 (test_k_mem_map_guard_after)
Caught system error -- reason 0
 PASS - test_k_mem_map_guard_after in 0.9 seconds
===================================================================
START - test_k_mem_map_exhaustion
Free memory: 7114752
E: insufficient virtual address space (requested 12288)
Mapped 1256 pages
Free memory now: 1970176
Unmapped 1256 pages
Free memory now: 7114752
 PASS - test_k_mem_map_exhaustion in 0.45 seconds
===================================================================
Test suite test_mem_map succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

You can see the output from the console, the first part of log is missing.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr SDK 13.2
 - Commit: 199048b2ed45b548a911b8c6f5d7ad45b34f78d6

nashif, 2022-03-31:
You should try adding a delay, run twister with 

-x CONFIG_BOOT_DELAY=10 

or a larger number and see if this works

Zhaoningx, 2022-04-02:
> You should try adding a delay, run twister with
> 
> -x CONFIG_BOOT_DELAY=10
> 
> or a larger number and see if this works

Hi @nashif, it doesn't work after adding the option CONFIG_BOOT_DELAY, I tested some values: CONFIG_BOOT_DELAY=10, CONFIG_BOOT_DELAY=100, CONFIG_BOOT_DELAY=2000. The log is still not complete.

Zhaoningx, 2022-04-12:
Hi, this issue can be reproduced manually, I tested the case using `west`, and got logs by reading the console. And as you can see, the log I posted above is not complete. So I would like to update it here, maybe this is not `twister` issue.

nashif, 2022-04-19:
use --disable-suite-name-check

closed 2022-04-19 by nashif