zephyrproject-rtos/zephyr #44349: Nordic BLE fails assertion when logging is enabled

crides, 2022-03-30:
**Describe the bug**

When logging is enabled on a BLE peripheral and data is being sent, and [the assertion here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c#L653) fails.

This still happens even when I enable `CONFIG_FORCE_NO_ASSERT`.

**To Reproduce**
 
I don't have a minimal reproduction yet, and reproducing my exact setup is a bit complicated (honestly I don't expect anyone to fully replicate this, and I'll try to provide as much info so people can quickly filter out the unimportant bits). The full background is that this is a split keyboard using the [ZMK Firmware](https://zmk.dev), which for 2 halves of a board it uses a BLE connection to send the keypress data across. Here's [the repo for the board](https://github.com/crides/zmk/tree/fissure). The specific board name is `fissure_left`. The specific Zephyr version used is `f2e12d55a1` in [this branch](https://github.com/petejohanson/zephyr/tree/v3.0.0+zmk-fixes). When I encounter the problem, I build the board `fissure_left` with logging on, attach the Jlink probe for the debug log, wait for the connection to happen, and press a key on the peripheral side. (log is provided in the corresponding section)

**Expected behavior**

No assertion happens, the split keyboard half communicates fine with the central half, but with logging to the probe.

**Impact**

Major annoyance. Normally logging is not enabled (especially for the peripheral side), but I needed to investigate some issue which leads to spurious resetting. This can certainly be done with backtrace information when the system resets, but having logging will definitely help.

**Logs and console output**

```
*** Booting Zephyr OS build 02c77d4cc96a  ***

[00:00:00.319,122] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.319,152] <inf> fs_nvs: alloc wra: 0, f68
[00:00:00.319,152] <inf> fs_nvs: data wra: 0, f8
[00:00:00.320,739] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.320,953] <dbg> zmk: ble_profiles_handle_set: Setting BLE value profiles/0
[00:00:00.321,105] <dbg> zmk: ble_profiles_handle_set: Loaded CB:C9:A8:EA:32:1E (random) address for profile 0
[00:00:00.326,354] <dbg> zmk: zmk_ble_ready: ready? 0
[00:00:00.326,354] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:00:00.331,817] <dbg> zmk: bvd_sample_fetch: ADC raw 2450 ~ 1794 mV => 3588 mV
[00:00:00.331,817] <dbg> zmk: bvd_sample_fetch: Percent: 7
[00:00:00.331,817] <dbg> zmk: zmk_battery_update: Setting BAS GATT battery level to 7.
[00:00:00.332,458] <inf> zmk: Welcome to ZMK!

[00:00:03.467,224] <dbg> zmk: connected: Connected thread: 0x200037e0
[00:00:03.467,346] <dbg> zmk: connected: Connected CB:C9:A8:EA:32:1E (random)
[00:00:03.468,627] <dbg> zmk: update_advertising: advertising from 0 to 0
[00:00:03.468,627] <dbg> zmk: connected: Active profile connected
[00:00:03.469,604] <dbg> zmk: get_selected_endpoint: Only BLE is ready.
[00:00:03.768,341] <dbg> zmk: split_svc_pos_state_ccc: value 1
[00:00:03.768,524] <dbg> zmk: security_changed: Security changed: CB:C9:A8:EA:32:1E (random) level 2
[00:00:09.039,642] <dbg> zmk: kscan_gpio_read: Sending event at 0,19 state on
[00:00:09.039,733] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 19, position: 36, pressed: true
[00:00:09.039,764] <dbg> zmk: split_listener: 
ASSERTION FAIL [next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:653

[00:00:09.106,262] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000004  r2/a3:  0x00000003
[00:00:09.106,262] <err> os: r3/a4:  0x20003128 r12/ip:  0xdfffebbc r14/lr:  0x0001b691[0m
[00:00:09.106,262] <err> os:  xpsr:  0x41000028
[00:00:09.106,292] <err> os: Faulting instruction address (r15/pc): 0x0001b69c
[00:00:09.106,323] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:09.106,323] <err> os: Fault during interrupt handling

[00:00:09.106,353] <err> os: Current thread: 0x20003bd0 (unknown)
[00:00:09.428,588] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Arch (updated in ~1 week)
 - Toolchain: `GNUARMEMB`
 - Commits used are provided above

**Additional context**

N/A

As suggested by @Emil-Gydesen-Bose in Discord, cc'ing @cvinayak

cvinayak, 2022-03-30:
@crides Lets analyse the issue by lowering the default IRQ priorities, which you can do by having a file `<your application folder>/dts/arm/nordic/override.dtsi` with the following one line in it:

```
#define NRF_DEFAULT_IRQ_PRIORITY 5
```

Please let me know the outcome...

crides, 2022-03-30:
So I'm not sure if I followed the directions correctly, but for ZMK it basically has the top level directory, the `zephyr` directory created from the `west.yml` and the `app` directory for the actual sources (I believe the latter 2 are common for standalone Zephyr projects?). I tried putting that in all 3 locations, and directly changing the default value of `NRF_DEFAULT_IRQ_PRIORITY` in `dts/arm/nordic/nrf_common.dtsi` to 5, all 4 produced the same result as described in the issue.

carlescufi, 2022-03-30:
@crides maybe the simplest way is to just edit `zephyr/dts/arm/nordic/nrf_common.dtsi` and make the change there.

crides, 2022-03-30:
Like I mentioned, I did, and it also didn't change anything

cvinayak, 2022-03-30:
Check your `zephyr.dts` file that you will find alongwith your `zephyr.hex` file that the priorities are 0x5, if that is the case I need to think other reasons that can delay the controller's execution at priority value 1.

[image]

carlescufi, 2022-03-30:
Typically `build/zephyr/zephyr.dts`

crides, 2022-03-30:
Yes, I checked and they are all 5

cvinayak, 2022-04-05:
@crides  It is difficult to analyse without reproducing the issue at my side. 

- Is there any directed advertising active during the assertion?
- Use the below to debug, to rule out interrupt latencies due to irq_lock
  ```
    CONFIG_ZERO_LATENCY_IRQS=y
    CONFIG_BT_CTLR_ZLI=y
  ```

carlescufi, 2022-04-06:
@crides ping

crides, 2022-04-06:
I got the ping, but I'm fairly busy for the last 2 days. I'll try to see if I can get some testing done tonight or tomorrow night

cvinayak, 2022-04-08:
@crides Any updates? I have some code changes which I think could help, unless you are able to reproduce the issue consistently and then test these changes, I will not be able to upstream a fix.

crides, 2022-04-09:
Sorry for the delay. I finally have some time to test.

Using the configs
```
CONFIG_ZERO_LATENCY_IRQS=y
CONFIG_BT_CTLR_ZLI=y
```
doesn't change anything, and there's advertising ongoing when the assert happens, but it's non advertising

cvinayak, 2022-04-09:
@crides Thank for confirming the assertion, please upload full console logs when possible to help me analyze the scenarios. 

Meanwhile I have a draft change (guess work), please cherry-pick the commit to your zephyr base and let me know if the assertion still happens.

PR: https://github.com/zephyrproject-rtos/zephyr/pull/44711

cvinayak, 2022-04-09:
Thanks for confirming and helping to resolve the issue.
No hurry, I would be away too, hence, it is ok if you report back on Monday.

crides, 2022-04-11:
Okay I merged and tested, the logs seem the same. So I set `CONFIG_BT_LOG_LEVEL_DBG=y` and the log is here:
[ble.log](https://github.com/zephyrproject-rtos/zephyr/files/8461190/ble.log)

cvinayak, 2022-04-11:
@crides Please increase `CONFIG_BT_CTLR_RX_PRIO_STACK_SIZE` to some high value like 1024 and check if you still get an assertion. I am suspecting that enabling logging needs more stack size in each active thread, you may increase all `CONFIG_*_STACK_SIZE`  Kconfig to high value and check using the following to ensure stack usage is within limits.

```
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_NAME=y
```

cvinayak, 2022-04-12:
@crides The listed stack usage seem ok, but the main stack and ISR stack are not being listed. Please increase `CONFIG_MAIN_STACK_SIZE` and `CONFIG_ISR_STACK_SIZE` both to 2048 bytes and check again if the thread analyzer can list their usage.

I want to rule out stack overflows which can lead to undetermined behaviors.

@carlescufi who can help with, why thread analyzer does not list main and ISR stack usage?

crides, 2022-04-13:
Not sure if this helps, but I slightly dug into what happens right before the assert happens. I added a logging line right after [here](https://github.com/petejohanson/zephyr/blob/v3.0.0%2Bzmk-fixes/subsys/bluetooth/controller/ll_sw/ull.c#L1865):
```c
    LOG_WRN("avail %d %d", MFIFO_AVAIL_COUNT_GET(prep), e != 0);
```
, and it produces the following results (I'm not sure if I'm understanding it incorrectly, but I felt like the `AVAIL_COUNT` is reversed?):
```
[00:00:08.829,010] <dbg> zmk: split_listener: 
[00:00:08.842,376] <wrn> bt_ctlr_ull: avail 0 1
[00:00:08.849,884] <wrn> bt_ctlr_ull: avail 1 1
[00:00:08.857,391] <wrn> bt_ctlr_ull: avail 2 1
[00:00:08.864,868] <wrn> bt_ctlr_ull: avail 3 1
[00:00:08.872,375] <wrn> bt_ctlr_ull: avail 4 1
[00:00:08.879,882] <wrn> bt_ctlr_ull: avail 5 1
[00:00:08.887,390] <wrn> bt_ctlr_ull: avail 6 1
[00:00:08.894,866] <wrn> bt_ctlr_ull: avail 7 0
ASSERTION FAIL [next] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:653

[00:00:08.894,897] <err> os: r0/a1:  0x00000003  r1/a2:  0x0000000c  r2/a3:  0x0000000b
[00:00:08.894,927] <err> os: r3/a4:  0x200036f0 r12/ip:  0xdfffdbbc r14/lr:  0x0001c01d
[00:00:08.894,927] <err> os:  xpsr:  0x61000028
[00:00:08.894,927] <err> os: Faulting instruction address (r15/pc): 0x0001c028
[00:00:08.894,958] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:08.894,989] <err> os: Fault during interrupt handling

[00:00:08.895,019] <err> os: Current thread: 0x20004390 (idle 00)
[00:00:09.378,784] <err> os: Halting system
```

Notice each log line is separated by 7.5ms, i.e. 1 connection interval. So for some reason the `prep` LLL event queue is not getting consumed, and I can't figure out what caused that now (there's too much radio related code and the search paths diverge really quickly :sweat_smile:)

crides, 2022-04-13:
> [00:00:08.829,010] zmk: split_listener:

So this is a split keyboard, and there are 2 devices. The peripheral side establishes connection to the central side to send keypress changes using `bt_gatt_notify`

cvinayak, 2022-04-14:
@crides I have another PR I was thinking was good, but had no reason to upstream it but please check if the changes in #42678 helps avoid the assertion you see. Thank you in advance for all the help.

crides, 2022-04-14:
I'm still in the process of merging the PR, as now LVGL is moved as a module, I'll need to see how to get this working in ZMK (as currently the way it's setup is a bit weird). I'll let you know the results later

cvinayak, 2022-04-18:
@crides You should cherry-pick the 2 commits https://github.com/zephyrproject-rtos/zephyr/pull/42678/commits/f6aea92dacbe184f476f115103a6647ab64e9c91 and https://github.com/zephyrproject-rtos/zephyr/pull/42678/commits/8fb0be52b59db702ec5201e9e54197c11e67a76a so that there is minimal change to the source base used to reproduce the issue. Will also avoid migrating the application to other upstream changes.

crides, 2022-04-20:
Okay that works, but the assertion remains the same (now it's on line 654 due to the commits).

cvinayak, 2022-04-20:
@crides ok, please provide your `zephyr.map` file to review if Controller's global variables could be overrun into.

crides, 2022-04-20:
[zephyr.map.txt](https://github.com/zephyrproject-rtos/zephyr/files/8517521/zephyr.map.txt)

cvinayak, 2022-04-20:
I do not see anything suspicious in the map file.

crides, 2022-04-24:
So, the problem is resolved.. The cause is that an [application stack size](https://github.com/crides/zmk/blob/fissure/app/src/split/bluetooth/service.c#L103) [is set too low](https://github.com/crides/zmk/blob/fissure/app/Kconfig#L193), basically throwing the BLE stack stuff into the void.

Now my question would be, is there any related debugging/error reporting utility related to this? If Zephyr can know the stack usage (somehow), it shouldn't be hard to detect a "stack overflow" when switching out of a task? The stack analyzer seems useful for _tuning down_ the stack allocation, but not for determining if it's enough.

closed 2022-04-24 by crides

cvinayak, 2022-04-29:
@crides Thanks for following up and help resolve the issue. I do not think there is any runtime detection of the stack overflow other than SoC/Architecture supported `CONFIG_MPU_STACK_GUARD`, please check if this option catches your issue.

crides, 2022-05-02:
Yep it does:

```
[00:00:05.407,745] <dbg> zmk: split_listener: 
[00:00:05.407,836] <err> os: ***** MPU FAULT *****
[00:00:05.407,836] <err> os:   Stacking error (context area might be not valid)
[00:00:05.407,836] <err> os:   Data Access Violation
[00:00:05.407,867] <err> os:   MMFAR Address: 0x20007238
[00:00:05.407,867] <err> os: r0/a1:  0x2000c780  r1/a2:  0x20008000  r2/a3:  0x00000013
[00:00:05.407,897] <err> os: r3/a4:  0x0000c00d r12/ip:  0x00000002 r14/lr:  0x00000000
[00:00:05.407,897] <err> os:  xpsr:  0x61000000
[00:00:05.407,897] <err> os: Faulting instruction address (r15/pc): 0x0002a6b6
[00:00:05.407,958] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:05.407,989] <err> os: Current thread: 0x20001270 (unknown)
[00:00:05.766,113] <err> os: Halting system
```

Would still need some digging after this point to know what exactly caused the problem, but I think this would be a good start to find the problem. Thanks!