zephyrproject-rtos/zephyr #44374: Twister: Non-intact handler.log files when running tests and samples folders

smrtos, 2022-03-30:
**Describe the bug**
For each sample or test app, if there's serial log, the log should start with the zephyr banner like below:

*** Booting Zephyr OS build zephyr-v3.0.0-1971-g6d17bf6aa7e2  ***

Use twister to run both tests/ and samples/ folder on reel board. And check the generated handler.log files.
Many of them are non-intact in one of the following 2 ways:

- The handler.log doesn't begin with the zephyr banner.
- The handler.log begins with unexpected serial log content from other test/application.

This will jeopardize the judgement of test pass/failure.

Below is the summary of the non-intact handler.log files for _one_ reel_board run.
The list may vary for different runs. **But the percentage is worrisome.**

**Total: 327, Not intact: 45, Percentage: 13.761467889908257%%**
```
Warning: twister-out/reel_board/tests/application_development/gen_inc_file/buildsystem.include_file/handler.log
Warning: twister-out/reel_board/tests/arch/arm/arm_ramfunc/arch.arm.ramfunc/handler.log
Warning: twister-out/reel_board/tests/arch/arm/arm_thread_swap/arch.arm.swap.common/handler.log
Warning: twister-out/reel_board/tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing/handler.log
Warning: twister-out/reel_board/tests/kernel/common/kernel.common.nano64/handler.log
Warning: twister-out/reel_board/tests/kernel/common/kernel.common.tls/handler.log
Warning: twister-out/reel_board/tests/kernel/common/kernel.common/handler.log
Warning: twister-out/reel_board/tests/kernel/common/kernel.common.nano32/handler.log
Warning: twister-out/reel_board/tests/subsys/usb/bos/usb.bos/handler.log
Warning: twister-out/reel_board/tests/subsys/usb/device/usb.device/handler.log
Warning: twister-out/reel_board/tests/subsys/usb/desc_sections/usb.desc_sections/handler.log
Warning: twister-out/reel_board/tests/subsys/portability/cmsis_rtos_v2/portability.cmsis_rtos_v2/handler.log
Warning: twister-out/reel_board/tests/net/buf/net.buf/handler.log
Warning: twister-out/reel_board/tests/net/ieee802154/fragment/net.ieee802154.fragment/handler.log
Warning: twister-out/reel_board/tests/net/ieee802154/l2/net.ieee802154.l2/handler.log
Warning: twister-out/reel_board/tests/net/ieee802154/custom_l2/net.ieee802154.custom_l2/handler.log
Warning: twister-out/reel_board/tests/benchmarks/latency_measure/benchmark.kernel.latency/handler.log
Warning: twister-out/reel_board/samples/userspace/prod_consumer/sample.userspace.prod_consumer/handler.log
Warning: twister-out/reel_board/samples/posix/eventfd/sample.posix.eventfd/handler.log
Warning: twister-out/reel_board/samples/kernel/condition_variables/condvar/sample.kernel.cond_var/handler.log
Warning: twister-out/reel_board/samples/kernel/metairq_dispatch/sample.kernel.metairq_dispatch/handler.log
Warning: twister-out/reel_board/samples/subsys/tracing/tracing.osawareness.openocd/handler.log
Warning: twister-out/reel_board/samples/subsys/tracing/tracing.user/handler.log
Warning: twister-out/reel_board/samples/subsys/usb/hid/sample.usb.hid/handler.log
Warning: twister-out/reel_board/samples/subsys/usb/mass/sample.usb.mass_ram_none/handler.log
Warning: twister-out/reel_board/samples/subsys/usb/audio/headphones_microphone/sample.usb.audio.headphones_microphone/handler.log
Warning: twister-out/reel_board/samples/subsys/usb/audio/headset/sample.usb.audio.headset/handler.log
Warning: twister-out/reel_board/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.same_prio/handler.log
Warning: twister-out/reel_board/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher/handler.log
Warning: twister-out/reel_board/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.semaphores/handler.log
Warning: twister-out/reel_board/samples/subsys/portability/cmsis_rtos_v1/timer_synchronization/sample.portability.cmsis_rtos_v1.timer_synchronization/handler.log
Warning: twister-out/reel_board/samples/subsys/portability/cmsis_rtos_v1/philosophers/sample.portability.cmsis_rtos_v1.philosopher.same_prio/handler.log
Warning: twister-out/reel_board/samples/subsys/logging/logger/sample.logger.basic/handler.log
Warning: twister-out/reel_board/samples/drivers/crypto/sample.drivers.crypto.mbedtls.micro/handler.log
Warning: twister-out/reel_board/samples/drivers/crypto/sample.drivers.crypto.mbedtls/handler.log
Warning: twister-out/reel_board/samples/synchronization/sample.kernel.synchronization/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.same_prio/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.lifos/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.stacks/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.coop_only/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.fifos/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher.semaphores/handler.log
Warning: twister-out/reel_board/samples/philosophers/sample.kernel.philosopher/handler.log
Warning: twister-out/reel_board/samples/modules/nanopb/sample.modules.nanopb/handler.log
Warning: twister-out/reel_board/samples/compression/lz4/sample.compression.lz4/handler.log
```

Some non-intact handler.log samples:

Incomplete banner
```
* Booting Zephyr OS build zephyr-v3.0.0-1971-g6d17bf6aa7e2  ***
Running test suite arm_ramfunc
===================================================================
START - test_arm_ramfunc
PASS - test_arm_ramfunc in 0.1 seconds
===================================================================
Test suite arm_ramfunc succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

Totally missing banner
```

Philosopher 4 [C:-1]        STARVING       
Philosopher 5 [C:-2]    DROPPED ONE FORK   
Philosopher 5 [C:-2]  THINKING [  725 ms ] 
Philosopher 0 [P: 3]    HOLDING ONE FORK   
Philosopher 4 [C:-1]    HOLDING ONE FORK   
Philosopher 4 [C:-1]   EATING  [  300 ms ] 

```

Unexpected serial log from other test.
```
, consectetur adipiscing elit. Quisque sodales lorem lorem, sed endisse convallis, ipsum nec rhoncus aliquam, ex augue ultrices nisl, id aliquet mi diam quis ante. Pellentesque venenatis ornare ultrices. Quisque et porttitor lectus. Ut venenatis nunc et urna imperdiet porttitor non laoreet massa.Donec eleifend eros in mi sagittis egestas. Sed et mi nunc. Nunc vulputate,mauris non ullamcorper viverra, lorem nulla vulputate diam, et congue dui velit non erat. Duis interdum leo et ipsum tempor consequat. In faucibus enim quis purus vulputate nullam.
*** Booting Zephyr OS build zephyr-v3.0.0-1971-g6d17bf6aa7e2  ***
Philosopher 4 [C:-1]        STARVING       
Philosopher 4 [C:-1]    HOLDING ONE FORK   
Philosopher 4 [C:-1]   EATING  [  400 ms ] 
Philosopher 5 [C:-2]        STARVING       
Philosopher 5 [C:-2]    HOLDING ONE FORK   
Philosopher 3 [P: 0]        STARVING       
Philosopher 3 [P: 0]    HOLDING ONE FORK   
Philosopher 2 [P: 1]        STARVING       
Philosopher 2 [P: 1]    HOLDING ONE FORK   
Philosopher 1 [P: 2]        STARVING       
Philosopher 1 [P: 2]    HOLDING ONE FORK   
Philosopher 0 [P: 3]        STARVING       
Philosopher 4 [C:-1]    DROPPED ONE FORK   
Philosopher 4 [C:-1]  THINKING [  100 ms ] 
```

**Expected behavior**
The handler.log file should begin with Zephyr banner.

**Impact**
Test pass/failure judgement will be jeopardized.
Life gets hard for the daily test.

nashif, 2022-03-31:
This is not a bug of twister, twister captures what it sees and gets from the serial console. If you want some syncronization, you will have to delay the test execution with CONFIG_BOOT_DELAY and if some output from previous test is present, you will need to reset the board completely.

Identifying tests and consistency of test results has been addressed with a resent change that verify test suite names at the beginning and end of the test and flags those if there is a mismatch, so this is already addressed in twister.

nashif, 2022-03-31:
try some of the options above and see of this resolved the issue, reopen of not.

closed 2022-03-31 by nashif

smrtos, 2022-04-03:
More information:

For tests, the test harness expect leading messages like "Running test suite XXX". This message can also be non-intact sporadically while the test actually passed. Adding CONFIG_BOOT_DELAY can mitigate this a lot. But we still see some non-intact serial output like below one. Note the missing "u" in the "suite".

[image]

smrtos, 2022-04-03:
I think we are making life difficult for ourselves.
Why check test suite name at the beginning since we know it can be unreliable?
Why not add the test suite name at the end. *
Like put it into the ending message “PROJECT EXECUTION SUCCESSFUL”?
At least so far I didn’t see that broken.
All we want is to be confident about the test suite been run.
It’s much more reliable to catch that in the ending log than in the beginning.
And this way we don’t need to add extra boot delay, which can avoid wasting more time.