zephyrproject-rtos/zephyr #44391: tests-ci : peripheral: gpio: 1pin test Timeout

hakehuang, 2022-03-31:
**Describe the bug**
1pin test is Timeout on zephyr-v3.0.0-1964-g199048b2ed45 on mimxrt1015_evk

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1015_evk  --sub-test peripheral.gpio
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.0.0-1964-g199048b2ed45  ***
Running test suite gpio_api_1pin_test
===================================================================
START - test_gpio_pin_configure_push_pull
Running test on port=GPIO_3, pin=21
ASSERTION FAIL [pin < config->mux_count] @ WEST_TOPDIR/zephyr/drivers/gpio/gpio_mcux_igpio.c:142
E: r0/a1:  0x00000004  r1/a2:  0x0000008e  r2/a3:  0x40184000
E: r3/a4:  0x60006131 r12/ip:  0x00000000 r14/lr:  0x60006269
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x60008c6c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000248 (test_gpio_pin_configure_push_pu)
E: Halting system

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.0.0-1964-g199048b2ed45

hakehuang, 2022-03-31:
@danieldegrasse , @dleach02 , @mmahadevan108

hakehuang, 2022-04-08:
@danieldegrasse looks like the igpio does not compatible with latest pinmux code

hakehuang, 2022-04-08:
[image]

danieldegrasse, 2022-04-13:
@hakehuang See #44858, I reproduced and fixed the issue on the RT1015, but was not able to see a failure on the RT1060. What console output did you see on that platform, if any?

hakehuang, 2022-04-14:
@danieldegrasse , rt1060 is ok, but the case requires some manual settings. connect gpio 23 to 22.

PR #44858 (2022-05-18): Add workaround to GPIO MCUX driver for non contiguous pins
Some iMX RT parts have a non contiguous set of pins available on certain GPIO ports (for example, GPIO0_0-GPIO0_10 are defined, as well as GPIO0_20-GPIO0_31, but GPIO0_11-GPIO0_19 are absent). Add a workaround to the iMX RT gpio driver to correctly select the GPIO pinmux register in these cases.

This PR also fixes the GPIO active flags on the RT1024 and RT1170 EVK LEDs, as these were discovered to be incorrect in testing.

Fixes #44391
commit f7986835 (2022-05-18): drivers: gpio: gpio_mcux_igpio: add workaround for non contiguous pins

some iMX RT SOCs have non contiguous sets of gpio pins available, which
caused issues when selecting appropriate pinmux for these parts. Add
workaround code to adjust offset of pinmux settings when configuring
these pins.

Fixes #44391
closed 2022-05-18 by carlescufi