zephyrproject-rtos/zephyr #44394: net: route: add IPv4 routing table

tpennors, 2022-03-31:
**Is your enhancement proposal related to a problem? Please describe.**
Currently there is no support for IPv4 routing table only IPv6 is supported.

**Describe the solution you'd like**
It would be good to have IPv4 support with the same features as what is done with IPv6

**Describe alternatives you've considered**
Currently I am forced to use option SO_BINDTODEVICE at socket level but a system level solution with routing table would be much nicer.

avigreen1978, 2022-05-03:
This feature is really needed, any update?

jukkar, 2024-02-09:
This request was lost with other issues, sorry about that. I have not heard of anyone implementing this and it is not in my near future to do list. This would be useful feature to have so we can keep the issue open. Patches are welcome of course.