zephyrproject-rtos/zephyr #44397: twister: test case error number discrepancy in the result

Zhaoningx, 2022-03-31:
**Describe the bug**

The content in the test report files is a discrepancy in some platforms after the commit https://github.com/zephyrproject-rtos/zephyr/commit/340da2ecf3122a1d0b3031f3b4708754e49485fa got merged. I used the platform to describe as a demo to describe this issue.

**To Reproduce**
twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM0 -T tests/ztest/ -v

**twister console output:**

```
ztest@ztest-nuc1-sam-e70b-xplained:~/zephyrproject/zephyr$ twister -p sam_e70b_xplained --device-testing --device-serial /dev/ttyACM0 -T tests/ztest/ -v
INFO    - Zephyr version: zephyr-v3.0.0-1998-gd8df801fe427
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 17 test scenarios (15 configurations) selected, 3 configurations discarded due to filters.

Device testing on:

| Platform          | ID   | Serial device   |
|-------------------|------|-----------------|
| sam_e70b_xplained |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  4/15 sam_e70b_xplained         tests/ztest/mock_fff/testing.ztest.mock_fff        SKIPPED (filter)
INFO    -  5/15 sam_e70b_xplained         tests/ztest/error_hook/testing.ztest.error_hook.no_userspace SKIPPED (filter)
INFO    -  6/15 sam_e70b_xplained         tests/ztest/mock_fff/testing.ztest.mock_fff_global SKIPPED (filter)
INFO    -  7/15 sam_e70b_xplained         tests/ztest/ztress/testing.ztest.ztress            SKIPPED (filter)
INFO    -  8/15 sam_e70b_xplained         tests/ztest/register/testing.ztest.register        PASSED (device 3.200s)
INFO    -  9/15 sam_e70b_xplained         tests/ztest/automain/testing.ztest.automain        PASSED (device 2.582s)
INFO    - 10/15 sam_e70b_xplained         tests/ztest/base/testing.ztest.verbose_1           PASSED (device 2.624s)
INFO    - 11/15 sam_e70b_xplained         tests/ztest/base/testing.ztest.verbose_2           PASSED (device 2.779s)
INFO    - 12/15 sam_e70b_xplained         tests/ztest/base/testing.ztest.verbose_0           PASSED (device 2.603s)
INFO    - 13/15 sam_e70b_xplained         tests/ztest/custom_output/testing.ztest.customized_output PASSED (device 2.485s)
INFO    - 14/15 sam_e70b_xplained         tests/ztest/error_hook/testing.ztest.error_hook    PASSED (device 5.090s)
INFO    - 15/15 sam_e70b_xplained         tests/ztest/custom_output/testing.ztest.regular_output PASSED (device 2.830s)

INFO    - 8 of 15 test configurations passed (100.00%), 0 failed, 7 skipped with 0 warnings in 48.29 seconds
INFO    - In total 39 test cases were executed, 17 skipped on 1 out of total 449 platforms (0.22%)
INFO    - 8 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board             | ID   |   Counter |
|-------------------|------|-----------|
| sam_e70b_xplained |      |         8 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

**In the twister.log file:**
```
2022-04-01 00:03:31,569 - twister - DEBUG - run status: sam_e70b_xplained/tests/ztest/custom_output/testing.ztest.regular_output passed
2022-04-01 00:03:31,570 - twister - INFO - 15/15 sam_e70b_xplained         tests/ztest/custom_output/testing.ztest.regular_output ^[[32mPASSED^[[39m (device 2.830s)
2022-04-01 00:03:31,576 - twister - DEBUG - running footprint_reports
2022-04-01 00:03:31,694 - twister - DEBUG - --------------------------------
2022-04-01 00:03:31,694 - twister - DEBUG - Total Test suites: 15
2022-04-01 00:03:31,694 - twister - DEBUG - Total Test cases: 56
2022-04-01 00:03:31,694 - twister - DEBUG - Skipped test cases: 17
2022-04-01 00:03:31,695 - twister - DEBUG - Completed Testsuites: 12
2022-04-01 00:03:31,695 - twister - DEBUG - Passing Testsuites: 8
2022-04-01 00:03:31,695 - twister - DEBUG - Failing Testsuites: 0
2022-04-01 00:03:31,695 - twister - DEBUG - Skipped Testsuites: 7
2022-04-01 00:03:31,695 - twister - DEBUG - Skipped Testsuites (runtime): 4
2022-04-01 00:03:31,695 - twister - DEBUG - Skipped Testsuites (filter): 3
2022-04-01 00:03:31,695 - twister - DEBUG - Errors: 0
2022-04-01 00:03:31,695 - twister - DEBUG - --------------------------------
2022-04-01 00:03:31,695 - twister - INFO - ^[[32m8 of 15^[[39m test configurations passed (100.00%), ^[[39m0^[[39m failed, 7 skipped with ^[[39m0^[[39m warnings in 48.29 seconds
2022-04-01 00:03:31,695 - twister - INFO - In total 39 test cases were executed, 17 skipped on 1 out of total 449 platforms (0.22%)
2022-04-01 00:03:31,695 - twister - INFO - ^[[32m8^[[39m test configurations executed on platforms, ^[[31m0^[[39m test configurations were only built.
2022-04-01 00:03:31,695 - twister - INFO - Saving reports...
2022-04-01 00:03:31,695 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
2022-04-01 00:03:31,696 - twister - INFO - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
2022-04-01 00:03:31,697 - twister - INFO - Run completed
```
**In the twister_report.xml file:**
```
<testsuites><testsuite name="sam_e70b_xplained" time="24.192772" tests="42" failures="0" errors="2" skipped="17"><properties>
<property name="version" value="zephyr-v3.0.0-1998-gd8df801fe427" /></properties><testcase classname="testing.ztest" 
name="testing.ztest.busy_sim.busy_sim" time="0.000000"><skipped type="skipped" message="Not in testcase platform allow list" />
</testcase><testcase classname="testing.ztest" name="testing.ztest.busy_sim_nrf52840dk_pin.busy_sim" time="0.000000"><skipped 
type="skipped" message="Not in testcase platform allow list" /></testcase><testcase classname="testing.ztest" 
name="testing.ztest.mock.parameter_tests" time="0.000000"><skipped type="skipped" message="In test case arch exclude" />
</testcase><testcase classname="testing.ztest" name="testing.ztest.mock.return_value_tests" time="0.000000"><skipped 
type="skipped" message="In test case arch exclude" /></testcase><testcase classname="testing.ztest" 
name="testing.ztest.mock.multi_value_tests" time="0.000000"><skipped type="skipped" message="In test case arch exclude" />
</testcase><testcase classname="testing.ztest" name="testing.ztest.mock.return_data_tests" time="0.000000"><skipped 
type="skipped" message="In test case arch exclude" /></testcase><testcase classname="testing.ztest" 
name="testing.ztest.regular_output.assert_pass" time="2.830154" /><testcase classname="testing.ztest" 
name="testing.ztest.regular_output.assert_skip" time="2.830154"><skipped type="skipped" message="ztest skip" /></testcase>
<testcase classname="testing.ztest" name="testing.ztest.customized_output.assert_pass" time="2.484563"><error type="failure" message="Unknown">*** Booting Zephyr OS build zephyr-v3.0.0-1998-gd8df801fe427  ***
Running test suite framework_tests
1: Test [test_assert_pass] reported (PASS) no. 1
2: Test [test_assert_pass] reported (PASS) no. 2
3: Test [test_assert_pass] reported (PASS) no. 3
4: Test [test_assert_skip] reported {SKIPPED} no. 1
5: Test [test_assert_skip] reported {SKIPPED} no. 2
Test suite framework_tests succeeded
PROJECT EXECUTION SUCCESSFUL
</error></testcase><testcase classname="testing.ztest" name="testing.ztest.customized_output.assert_skip" time="2.484563"><error type="failure" message="Unknown">*** Booting Zephyr OS build zephyr-v3.0.0-1998-gd8df801fe427  ***
Running test suite framework_tests
1: Test [test_assert_pass] reported (PASS) no. 1
2: Test [test_assert_pass] reported (PASS) no. 2
3: Test [test_assert_pass] reported (PASS) no. 3
4: Test [test_assert_skip] reported {SKIPPED} no. 1
5: Test [test_assert_skip] reported {SKIPPED} no. 2
Test suite framework_tests succeeded
PROJECT EXECUTION SUCCESSFUL
```

After comparing these three files, we can find the **error number** is different. 

**Environment (please complete the following information):**
OS: Linux
Toolchain: zephyr SDK 13.2
Commit: https://github.com/zephyrproject-rtos/zephyr/commit/d8df801fe42780b8e4cb177eb0675c546400fef8

Zhaoningx, 2022-03-31:
cc @gopiotr

Zhaoningx, 2022-03-31:
Hi @gopiotr, I find another issue after the commit https://github.com/zephyrproject-rtos/zephyr/commit/340da2ecf3122a1d0b3031f3b4708754e49485fa got merged, so could you help to take a look at this?

gopiotr, 2022-03-31:
@Zhaoningx thank you for let me know :) 

Indeed now I also observed, that my changes impact on final results in `twister_report.xml` report. But to be honest my changes only discover problem with parsing output which existed before my changes. The source of problem is test `testing.ztest.customized_output` from file `tests/ztest/custom_output/testcase.yaml`. It enable in configs `CONFIG_ZTEST_TC_UTIL_USER_OVERRIDE` option which allow users to override how ztest output looks like. In the result Twister is not able to recognize this new ztest output and do not assign any result to performed testcases. Before my changes it wasn't a problem, because when Twister wasn't able to detect testcase's output it do not mark this test as ztest and just verify whole test by output like `PROJECT EXECUTION SUCCESSFUL`. When I added testsuite name verification (and then recognition by this test as ztest) it cause that Twister starts marking testcases not only by final result.

I'll try to fix it in nearest time.

PR #44430 (2022-04-01): twister: harness: fix ztest detection
Decision about whether test should be considered as ztest should be made
after detection ztest testcase - not after detection ztest test suite.

Due to the fact that ztest provides possibility to overwrite some ztest macros by enable this option: `CONFIG_ZTEST_TC_UTIL_USER_OVERRIDE`, then  user can change how ztest output looks like. When he/she do this, then Twister is not able to properly detect testcases. So, final decision about whole test and each testcases results should base in this case only on final communicate `PROJECT EXECUTION SUCCESSFUL` or `P [... cut]
commit 806cc70f (2022-04-01): twister: harness: fix ztest detection

Decision about whether test should be considered as ztest should be made
after detection ztest testcase - not after detection ztest test suite.

Fixes: #44397
closed 2022-04-01 by nashif