zephyrproject-rtos/zephyr #44400: LE Audio: Unicast server stream control

Thalley, 2022-03-31:
**Is your enhancement proposal related to a problem? Please describe.**
The unicast server is currently not able to control the states of the ASEs, as all the `bt_audio_stream_` functions that handle such changes are guarded with 

```c
	role = stream->conn->role;
	if (role != BT_HCI_ROLE_CENTRAL) {
		BT_DBG("Invalid conn role: %u, shall be central", role);
		return -EINVAL;
	}
```

and thus currently only works for the client. 

The unicast server is the owner of the ASEs, and should be able to control the ASEs. 

**Describe the solution you'd like**
Either create a new set of API functions for the unicast server to set ASE states and values, or use the existing `bt_audio_stream_` functions. The latter of which will require a connection to be established first, to perform a role check. 

We need to figure out how much the unicast server can do while disconnected, e.g. it should definitely not be able to go into a streaming state, but what about configuring codecs and QoS? 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

MariuszSkamra, 2022-04-05:
Regarding server going to Enabling state autonomously, this is not allowed by the ASCS specification
[image]
the operation can be initiated by Client only

MariuszSkamra, 2022-04-05:
Regarding the server ASE going to Enabling state autonomously, it's not allowed by the specification
[image]
[image]
Client only can initiate transition to enabling state from QoS state.

closed 2022-07-18 by fabiobaltieri