zephyrproject-rtos/zephyr #44404: Porting stm32h745 for zephyr

GaelRap, 2022-03-31:
Hello everyone, I try to use Zephyr on STM32h745I_DISCO board but the board is not natively ported. So I try making my own configuration file inspired on `nucleo_h745zi_q` configuration files. 

Actually I totally copy/paste the folder `nucleo_h745zi_q` and I rename it `stm32h745i_disco` I also change inside the folder every file containing `nucleo_h745zi_q*` by `stm32h745*`

To try my new configuration I test with the sample `basic/blinky`I change the user leds gpio to fit with what I have seen in the doc

```
		green_led: led_1 {
			gpios = <&gpioi 13 GPIO_ACTIVE_HIGH>;
			label = "User LD1";
		};
```

I have two warnings when compilation, and the binaries didn't work given the led didn't blink. 
```
[94/156] Building C object zephyr/driv...ck_control.dir/clock_stm32_ll_h7.c.obj
/home/gael/Documents/S2OPC/Zephyr/zephyr/drivers/clock_control/clock_stm32_ll_h7.c: In function 'set_up_plls':
/home/gael/Documents/S2OPC/Zephyr/zephyr/drivers/clock_control/clock_stm32_ll_h7.c:542:6: warning: implicit declaration of function 'get_vco_input_range' [-Wimplicit-function-declaration]
  542 |  r = get_vco_input_range(STM32_PLL_M_DIVISOR, &vco_input_range);
      |      ^~~~~~~~~~~~~~~~~~~
/home/gael/Documents/S2OPC/Zephyr/zephyr/drivers/clock_control/clock_stm32_ll_h7.c:547:21: warning: implicit declaration of function 'get_vco_output_range' [-Wimplicit-function-declaration]
  547 |  vco_output_range = get_vco_output_range(vco_input_range);
      |                     ^~~~~~~~~~~~~~~~~~~~

```
Will be very gratefull if someone could help me to figure out where the issue is.

erwango, 2022-03-31:
What is the SHA1 you're using ?
What command line are you using ?
Do you get the same issue when compiling `nucleo_h745zi_q_m7` target on the same sample ?

On my side, I don't get any warning when compling `nucleo_h745zi_q_m7` on following commit: `zephyr-v3.0.0-2081-g99964fab45`

erwango, 2022-03-31:
Alos, about your board configuration, I can see that `nucleo_h745zi_q` clock configuration relies on HSE. Is it also available on disco board ?

GaelRap, 2022-03-31:
I don't get any warning compiling with `nucleo_h745zi_q_m7` on the following commit:  `zephyr 3.0.99 -8fc4e77d5`
I think that there is no HSE on disco boards, I will try to link another source for the clocks
Thank you

erwango, 2022-03-31:
> I come back about the HSE, I found in this datasheet https://www.st.com/resource/en/datasheet/stm32h745xi.pdf page 2 that there is an HSE something that I don't find out in the specific datasheet of stm32h745i_disco https://www.st.com/resource/en/user_manual/dm00547983-discovery-kits-with-stm32h745xi-and-stm32h750xb-microcontrollers-stmicroelectronics.pdf

Ok, then you need to fix dts configuration.
I'm moving this to a discussion, as it seems there is no bug here.