zephyrproject-rtos/zephyr #44410: drivers: modem: shell: `modem send` doesn't honor line ending in modem cmd handler

MattCampbellST, 2022-03-31:
**Describe the bug**
Currently the modem subsystem's shell command to send arbitrary AT commands is hard coded to only send a `\r` line ending. When registering a modem context driver, it sets the preferred line ending in the modem cmd handler, but this isn't used in the modem send shell command.

**To Reproduce**
Build a Zephyr image with the quectel-bg9x driver (which sets the eol to `\r\n`). Use `modem send 0 AT` and you can see on a UART bus sniffer that `AT\r` is sent.

**Expected behavior**
Sending a command with `modem send` should use the line ending configured in the modem context.

**Impact**
This is less of a production issue, but being able to issue arbitrary AT commands to a modem is extremely helpful during development and debugging, and also is useful in a manufacturing test context as well.

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
With the way the modem command handler is currently implemented, the only public interface to it is the `process` function pointer. Everything else is left behind an opaque implementation defined pointer. This prevents someone (for example a modem shell command in this case) who only has the modem context from getting the line ending set within that opaque pointer. I'd suggest that we could change the `void *` to a `struct modem_cmd_handler_data*`. Right now that is the only concrete implementation of a command handler and I can't think of cases where we would benefit from having different handlers, so I don't think the increased coupling hurts us.

```c
struct modem_cmd_handler {
	void (*process)(struct modem_cmd_handler *cmd_handler,
			struct modem_iface *iface);

	/* implementation data */
	
	/* Change this from a void *
	 * void *cmd_handler_data;
	 */
	/* to this */
	struct modem_cmd_handler_data *cmd_handler_data;
};
```

I'm happy to open a PR with the above proposed change if this feels tenable.

@mike-scott might have some thoughts on this.

carlescufi, 2022-03-31:
> I'm happy to open a PR with the above proposed change if this feels tenable.
> 
> @mike-scott might have some thoughts on this.

@rerickson1 is the new maintainer.

MattCampbellST, 2022-03-31:
Quick follow up after thinking a bit more, I have another issue I was going to file about how sending arbitrary AT commands doesn't print the output which feels like a common use case. I was thinking about a solution to that which uses the modem cmd handler to send arbitrary AT commands rather than just sending it out raw using the modem iface. I could see having some helper in the modem subsystem that took care of this and printed all response to the shell just for that command. I'm not sure if there is a good way to do this that isn't tightly coupled to individual driver implementations, but that would also solve the eol issue above by using the cmd handler.

konstk1, 2022-08-29:
Hi, we would like this issue re-opened. I'm from the same company as @MattCampbellST and taking over this effort.

konstk1, 2022-10-30:
Not stale. Still an issue. Are there any plans to address this?

rerickson1, 2022-10-30:
> Not stale. Still an issue. Are there any plans to address this?

Please submit a PR to address the issue.

closed 2023-01-14 by github-actions[bot]

ouaaliw, 2024-03-12:
Hello Avery body, i am exploring the sample cellular modem app  ..\zephyrproject\zephyr\samples\net\cellular_modem and i want to send directly the commandAT using the modem_shell. I added the necessary shell configurations 

# Modem Shell
CONFIG_MODEM_SHELL=y
CONFIG_SHELL=y
CONFIG_MODEM_CONTEXT=y

but when i use the modem list command, the list is empty, there is no modem receiver.
anyone please have an idea how to fix that.
[image]