zephyrproject-rtos/zephyr #44427: SYS_CLOCK_HW_CYCLES_PER_SEC not correct for hifive1_revb / FE310

snematbakhsh, 2022-03-31:
8388bb7c24bca94a5d40dba0a36e79b80647be6b changed SYS_CLOCK_HW_CYCLES_PER_SEC default to 320000 (was 32768). But, 32768 seems to be the correct setting for hifive1_revb. Test code (run on hifive1_revb):

       while (1) {
               printf("cps: %d cycles: %d secs: %d\n",
                               sys_clock_hw_cycles_per_sec(),
                               sys_clock_cycle_get_32(),
                               sys_clock_cycle_get_32() / sys_clock_hw_cycles_per_sec());
               for (int i = 0; i < 5 * 1000 * 1000; ++i) __asm__ volatile("nop;");
       }

With latest head:

cps: 320000 cycles: 1995 secs: 0                                                
cps: 320000 cycles: 33245 secs: 0                                               
cps: 320000 cycles: 64211 secs: 0                                               
cps: 320000 cycles: 95157 secs: 0                                               
cps: 320000 cycles: 126103 secs: 0                                              
cps: 320000 cycles: 157055 secs: 0                                              
cps: 320000 cycles: 188008 secs: 0                                              
cps: 320000 cycles: 218960 secs: 0                                              
cps: 320000 cycles: 249913 secs: 0                                              
cps: 320000 cycles: 280865 secs: 0                                              
cps: 320000 cycles: 311818 secs: 0                                              
cps: 320000 cycles: 342770 secs: 1

(Print rate is approximately 1 print / sec).

With 8388bb7c24bca94a5d40dba0a36e79b80647be6b reverted:

cps: 32768 cycles: 2174 secs: 0                                                 
cps: 32768 cycles: 33232 secs: 1                                                
cps: 32768 cycles: 64172 secs: 1                                                
cps: 32768 cycles: 95111 secs: 2                                                
cps: 32768 cycles: 126051 secs: 3                                               
cps: 32768 cycles: 156997 secs: 4                                               
cps: 32768 cycles: 187943 secs: 5                                               
cps: 32768 cycles: 218889 secs: 6                                               
cps: 32768 cycles: 249835 secs: 7 

Note that FE310 gets HW cycles from mtime, which counts rtcclk cycles (ref: 9.3 "Timer Registers" in FE310 manual). "RTC should be driven by an external 32.768 kHz watch crystal oscillator".

I'm happy to change this back for hifive1_revb (and/or for all FE310), but based on the 8388bb7c24bca94a5d40dba0a36e79b80647be6b commit message, I wonder if the value 32768 broke something else (maybe something in QEMU target)?

snematbakhsh, 2022-03-31:
https://github.com/zephyrproject-rtos/zephyr/pull/44432 addresses the HiFive1 boards (I double checked both schematics to ensure that 32.768 external oscillator is used for LFCLK).

mbolivar-nordic, 2022-05-24:
@snematbakhsh @fkokosinski I just merged https://github.com/zephyrproject-rtos/zephyr/pull/45344. Can we close this issue?

fkokosinski, 2022-05-25:
@mbolivar-nordic Yes, this issue is fixed now, closing.

closed 2022-05-25 by fkokosinski