zephyrproject-rtos/zephyr #44438: tests-ci : arch: interrupt: arm.nmi test Unknown

hakehuang, 2022-04-01:
**Describe the bug**
tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations  test is Unknown on zephyr-v3.0.0-1964-g199048b2ed45 on lpcxpresso55s28

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28  --sub-test arch.interrupt
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
INFO    -  718/1077 mimxrt685_evk_cm33        tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations FAILED Testsuite mismatch (device 3.826s)
INFO    - /home/ubuntu/zephyr_project/zephyr_rt685/zephyr/twister-out/mimxrt685_evk_cm33/tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations/handler.log
ERROR   - arm_runtime_nmi in 10.46 seconds
===================================================================
Test suite arm_runtime_nmi_fn succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.0.0-1964-g199048b2ed45

hakehuang, 2022-04-01:
@gopiotr , looks like the suitename checking has problem when you have differrent suite name but same log output.

gopiotr, 2022-04-01:
> also I find same issue with testing.ztest.customized_output

This issue was fixed with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/44430

gopiotr, 2022-04-01:
@hakehuang thank you for letting me know about this.
1. Could you limit executed testcases by add `-T tests/arch/arm/arm_interrupt -T tests/arch/arm/arm_runtime_nmi/` option to your Twister command? I would like to make sure if this error occur with those specific tests or is the result of run whole long list of testscases (chosen by `--sub-test arch.interrupt` option).
2. I run those specific tests on my local hardware (`nrf52840dk_nrf52840`) by command and I do not observe any error:
```
./scripts/twister --ninja --device-testing --device-serial /dev/ttyACM0 -v -v --platform nrf52840dk_nrf52840 -T tests/arch/arm/arm_interrupt -T tests/arch/arm/arm_runtime_nmi/ --sub-test arch.interrupt
```
3. Moreover we do not observe any problems with `arch.interrupt` tests in our CI (revision `zephyr-v3.0.0-2120-g5a3f52`) - everything pass properly. I do not have possibility to run those tests on `lpcxpresso55s28`, `mimxrt1015_evk`, `mimxrt1060_evk` or `mimxrt1170_evk_cm7`, so I cannot reproduce those error locally. 
4. So, it looks like a problem connected with specific platform (for example with correctly log printing), not with Twister and "test suite name" verification.
5. I cannot conclude anything more from this short log, which you provide in this issue. Please analyze logs or reports once again and try to clarify what exactly fails and how whole logs/output looks like (especially from those failing tests)

hakehuang, 2022-04-02:
> 1. Could you limit executed testcases by add `-T tests/arch/arm/arm_interrupt -T tests/arch/arm/arm_runtime_nmi/` option to your Twister command? I would like to make sure if this error occur with [...]

it is strange that if I only run the -T tests/arch/arm/arm_interrupt or -T tests/arch/arm/arm_runtime_nmi/ the test is OK, but it fails when I run batch mode.

Let me debug this.

nashif, 2022-04-05:
I have seen this issue, it is related to the test itself and the fact it needs a complete hw reset after the test is completed or something done to the device to get it back in shape after this test. There are a few other tests that behave the same, those need to be marked in the test definition to get special treatment or excluded from batch mode. 

I am now considering disabling the testsuite name check until we have identified all problematic tests, fixed them or marked them in a way to allow special treatment or workflow. Right now we are having multiple issues being reported across all vendors due to the new checks.

@gopiotr  Can you please propose a solution for this and identify the problematic tests?

nashif, 2022-04-05:
I suspect this is a duplicate of #44539

closed 2022-04-05 by nashif