zephyrproject-rtos/zephyr #44439: Bluetooth: Controller: Extended and Periodic Advertising HCI Component Conformance Test Coverage

cvinayak, 2022-04-01:
List of pending Extended and Periodic Advertising HCI Component Conformance Test Coverage

- [ ] HCI/CCO/BI-33-C [Invalid LE Set Periodic Advertising Data Parameters]
- [ ] HCI/DDI/BI-01-C [Reject Invalid Extended Advertising Parameters]
- [ ] HCI/DDI/BI-13-C
- [ ] HCI/DDI/BI-15-C
- [ ] HCI/DDI/BI-16-C
- [ ] HCI/DDI/BI-17-C
- [ ] HCI/DDI/BI-18-C
- [ ] HCI/DDI/BI-19-C
- [ ] HCI/DDI/BI-20-C
- [ ] HCI/DDI/BI-21-C
- [ ] HCI/DDI/BI-22-C
- [ ] HCI/DDI/BI-23-C
- [ ] HCI/DDI/BI-24-C
- [ ] HCI/DDI/BI-25-C
- [ ] HCI/DDI/BI-26-C
- [ ] HCI/DDI/BI-27-C
- [ ] HCI/DDI/BI-28-C
- [ ] HCI/DDI/BI-29-C
- [ ] HCI/DDI/BI-30-C
- [ ] HCI/DDI/BI-31-C
- [ ] HCI/DDI/BI-32-C
- [ ] HCI/DDI/BI-33-C
- [ ] HCI/DDI/BI-34-C
- [ ] HCI/DDI/BI-35-C
- [ ] HCI/DDI/BI-36-C
- [ ] HCI/DDI/BI-37-C
- [ ] HCI/DDI/BI-38-C
- [ ] HCI/DDI/BI-39-C
- [ ] HCI/DDI/BI-40-C
- [ ] HCI/DDI/BI-41-C
- [ ] HCI/DDI/BI-42-C
- [ ] HCI/DDI/BI-43-C
- [ ] HCI/DDI/BI-44-C
- [ ] HCI/DDI/BI-45-C
- [ ] HCI/DDI/BI-46-C
- [ ] HCI/DDI/BI-47-C
- [ ] HCI/DDI/BI-50-C
- [ ] HCI/DDI/BI-51-C
- [ ] HCI/DDI/BI-52-C
- [ ] HCI/DDI/BI-53-C
- [ ] HCI/DDI/BI-54-C
- [ ] HCI/DDI/BI-58-C
- [ ] HCI/DDI/BI-59-C
- [ ] HCI/DDI/BI-60-C
- [ ] HCI/DDI/BI-61-C
- [ ] HCI/DDI/BI-62-C
- [ ] HCI/DDI/BI-63-C
- [ ] HCI/DDI/BI-64-C
- [ ] HCI/DDI/BI-65-C
- [ ] HCI/DDI/BI-66-C
- [ ] HCI/DDI/BI-67-C
- [ ] HCI/DDI/BV-08-C

carlescufi, 2022-05-19:
@cvinayak can this be closed now?

cvinayak, 2022-05-19:
@carlescufi I have couple more PRs to send that have conflict with #45205, hence need to wait for that to be merged.

PR #45818 (2022-05-23): Bluetooth: Controller: Fix HCI AD Data fragment operation validations
Add initial support for Periodic Advertising Data fragment
operation support.

This commit is limited to detection of partial data if used
before Periodic Advertising is enabled, which will lead to
HCI error reason of command disallowed.

Fix implementation to reject setting fragment when periodic
advertising is enabled.

Fixes #44439.
closed 2022-05-23 by carlescufi