zephyrproject-rtos/zephyr #44449: qemu_riscv32 DHCP fault

EGeeks, 2022-04-01:
**Describe the bug**
qemu_riscv32 DHCP fault

-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: riscv32
*** Booting Zephyr OS build zephyr-v3.0.0-1944-g9f886c9cf68c  ***

[00:00:00.000,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:07.010,000] <err> os: 
[00:00:07.010,000] <err> os:  mcause: 5, Load access fault
[00:00:07.010,000] <err> os:   mtval: 8
[00:00:07.010,000] <err> os:      a0: 00000000    t0: 8001397c
[00:00:07.010,000] <err> os:      a1: 000000c0    t1: 8d178000
[00:00:07.010,000] <err> os:      a2: 00000006    t2: 00000000
[00:00:07.010,000] <err> os:      a3: 000000d3    t3: 00000000
[00:00:07.010,000] <err> os:      a4: 80027300    t4: 00000001
[00:00:07.010,000] <err> os:      a5: 80025480    t5: aaaaaaaa
[00:00:07.010,000] <err> os:      a6: 00000000    t6: aaaaaaaa
[00:00:07.010,000] <err> os:      a7: 000000ff
[00:00:07.010,000] <err> os:      ra: 800109e6
[00:00:07.010,000] <err> os:    mepc: 8001037c
[00:00:07.010,000] <err> os: mstatus: 00001880
[00:00:07.010,000] <err> os: 
[00:00:07.010,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:07.010,000] <err> os: Current thread: 0x80020cd0 (sysworkq)
[00:00:07.020,000] <err> os: Halting system

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_riscv32 samples/net/dhcpv4_client --pristine
2. west build -t run

yperess, 2022-04-13:
I'm running into this also when adding an iterable data structure and trying to access it from a user thread (https://github.com/zephyrproject-rtos/zephyr/pull/44562)

rlubos, 2022-04-26:
So the crash seesm to be caused by the SLIP driver (or rather underlying uart_pipe) which tries to make use of undefined UART devcie (uart_pipe tries to read the device name from DTS, but it's not defined for qemu_riscv32).

Anyway, it seems that this platform is not capable for emulated networking at the moment. It only seems to have a single UART device, while we need at least 2 if we want to have both, log output and SLIP.

carlescufi, 2022-05-11:
SLIP + logging requires 2 UARTs, so this platform simply doesn't support networking at the moment.

closed 2022-05-11 by carlescufi