zephyrproject-rtos/zephyr #44453: Linker warnings in watchdog samples and tests built for twr_ke18f

anangl, 2022-04-01:
**Describe the bug**
An attempt to build a sample or test that uses watchdog on twr_ke18f ends up with the following linker warnings:
```
155/165] Linking C executable zephyr/zephyr_pre0.elf
/home/andy/zephyrproject/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `CodeQuickAccess' from `modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a(fsl_wdog32.c.obj)' being placed in section `CodeQuickAccess'

[159/165] Linking C executable zephyr/zephyr_pre1.elf
/home/andy/zephyrproject/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `CodeQuickAccess' from `modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a(fsl_wdog32.c.obj)' being placed in section `CodeQuickAccess'

[165/165] Linking C executable zephyr/zephyr.elf
/home/andy/zephyrproject/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `CodeQuickAccess' from `modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a(fsl_wdog32.c.obj)' being placed in section `CodeQuickAccess'
```
I have hit this problem in CI runs in #44276 ([here](https://github.com/zephyrproject-rtos/zephyr/pull/44276/checks?check_run_id=5770904525) is one example).

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/drivers/watchdog`
2. `west build -b twr_ke18f`
3. See error

**Impact**
This prevents successful CI runs for PRs that touch watchdog related samples or tests (hence trigger building them also for twr_ke18f), so it is a showstopper.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.0
 - Commit 35593ce1e93107f9190587d28585e02bb28b3648

henrikbrixandersen, 2022-04-01:
I can confirm the issue. It is caused by the `AT_QUICKACCESS_SECTION_CODE()` macro used in recent MCUX SDK revisions.

dleach02, 2022-04-01:
PR posted

PR #44472 (2022-04-02): hal: nxp: Centralize handling of SDK custom sections
The NXP SDK defines CodeQuickAccess and DataQuickAccess sections
for locating critical items that need faster access. Centralize
the handling of these sections instead of doing it per SOC.

Fixes #44453
commit 314e8484 (2022-04-02): hal: nxp: Centralize handling of SDK custom sections

The NXP SDK defines CodeQuickAccess and DataQuickAccess sections
for locating critical items that need faster access. Centralize
the handling of these sections instead of doing it per SOC.

Fixes #44453
closed 2022-04-02 by carlescufi