zephyrproject-rtos/zephyr #44466: Zephyr misses strict aliasing disabling

evgeniy-paltsev, 2022-04-01:
While debugging issue I've bumped when I was using new ARC GCC11 based toolchain I've surprisingly found that we don't disable strict aliasing (don't pass `-fno-strict-aliasing` option for gcc/llvm based toolchains).

I've seen the places where we violate strict aliasing rule in many places in Zephyr.

---
Here is an example:
In posix threads layer we do cast between `k_tid_t` (which is defined as `typedef struct k_thread* k_tid_t`) 
and the pointer to posix_thread structure, which is defined as 

```
struct posix_thread {
	struct k_thread thread;

    // *snip* //
};

```

Here is an example of code where we do assignment which violates the strict aliasing rule (located in [pthread.c](https://github.com/zephyrproject-rtos/zephyr/blob/35e058bcad97742b7a67cd43aeb2bde671e6e329/lib/posix/pthread.c#L312)

```
int pthread_getschedparam(pthread_t pthread, int *policy,
			  struct sched_param *param)
{
	struct posix_thread *thread = (struct posix_thread *) pthread;
	uint32_t priority;

	if ((thread == NULL) || (thread->state == PTHREAD_TERMINATED)) {
		return ESRCH;
	}

	priority = k_thread_priority_get((k_tid_t) thread);       // <<<<---- incorrect!

	param->sched_priority = zephyr_to_posix_priority(priority, policy);
	return 0;
}
```
---

So, given that 
- initial GCC11 issue goes away when disable strict aliasing
- we have places in code which violates strict aliasing rule
I guess we need do disable strict aliasing.

I'm going to do that for ARC, however I guess it's worth to do it for all architectures.

Any ideas why it isn't disabled now?

evgeniy-paltsev, 2022-04-01:
@abrodkin @ruuddw FYI

ruuddw, 2022-04-01:
> Any ideas why it isn't disabled now?

Performance I assume, with the strict aliasing assumption the compiler can optimize more. But most likely it wasn't a conscious decision.
Are there many violations? Could we do -Wno-strict-aliasing instead? I see '#pragma GCC diagnostic ignored "-Wstrict-aliasing"' in the syscall generation code...

stephanosio, 2022-04-04:
cc @andyross @peter-mitsis

evgeniy-paltsev, 2022-05-03:
Any thoughts about this?

As of today we have undefined behavior in Zephyr kernel code (so bumping priority)

To avoid that undefined behavior we either need to 
 * disable strict aliasing optimization (`-fno-strict-aliasing` option for GCC and LLVM)
or 
 * fix all places in kernel code where we violate strict aliasing rule (IMO, it's hardly achievable)

andyross, 2022-05-03:
Sorry, just seeing this.  I'm not sure I buy that example.  That's just typecasting, there's no aliasing because in the local function, the pointers have the same value and the compiler can prove that.    And if k_thread_priority_get() is inlined, it gets the same treatment.  If it's not inlined, then the pointer gets passed as an argument and aliasing optimization isn't possible anyway.

There's a lot of value in the gcc strict-aliasing optimization, and turning this off is going to have significant impact.  Circumstances where a function **actually** gets two aliased variants of the same memory in a way that it can't prove are actually extremely rare.  You need to work hard to trip over this.  I guess my sense is that absent clear evidence of bugs we're permitting, we should leave this on.  (And yes, I'm aware that Linux made the opposite choice.  But that was back in the days before pervasive static analysis tools.  And in any case Zephyr is less dynamic and does a **lot** less "copying memory around" than Linux.)

stephanosio, 2022-05-24:
Will mark this as an enhancement since this is not necessarily a bug.

stephanosio, 2022-08-10:
It looks like there are some new optimisations added in the GCC 12 (and 11) that assume the strict aliasing rule, and compiling with strict aliasing enabled (which is the default because of `-Os`) with the GCC 12 generates invalid code resulting in tons of run-time failures: see https://github.com/zephyrproject-rtos/zephyr-testing/runs/7752867905.

For example, the `fix_missing_black` function in `rb.c` goes into an infinite loop and breaks the scheduler which uses it. The cbprintf implementation is also affected by this.

At this point, we have no option but to specify `-fno-strict-aliasing` since, as @evgeniy-paltsev noted above, fixing all places that we violate the strict aliasing rule would be practically impossible (at least, as of now).

evgeniy-paltsev, 2022-08-10:
In the collection of affected tests: `samples/philosophers/sample.kernel.philosopher.coop_only` fails on ARC platforms with GCC 12 but passes if we disable strict aliasing.

stephanosio, 2022-08-10:
> In the collection of affected tests: `samples/philosophers/sample.kernel.philosopher.coop_only` fails on ARC platforms with GCC 12 but passes if we disable strict aliasing.

Yes, that test fails on multiple platforms with `-fstrict-aliasing`. It gets stuck because of the `rb.c` issue mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/44466#issuecomment-1211012355.

All the kernel and cbprintf tests, that were failing `-fstrict-aliasing`, now pass with `-fno-strict-aliasing`.

andyross, 2022-08-10:
Huh... that rb.c issue is surprising.  The tree code shouldn't be doing any typewise aliasing at all, the header is a uniform struct and nothing gets unioned.  Still, as pointed out this optimization isn't turnout out to be all that useful[1], so I guess there's no point in going crazy trying to fix everything.

[1] Which is surprising to me too -- I know in the past on previous projects it's been a big win.  But maybe general optimizers have gotten so good that they're able to figure out most of what this optimization used to do anyway.

ruuddw, 2022-08-11:
> The footprint ramifications of setting -fno-strict-aliasing is minimal.

How about performance?

stephanosio, 2022-08-11:
> How about performance?

Looking at the listing outputs, the performance impact should be negligible.

ruuddw, 2022-08-11:
Thanks (though this might be different for user code), then I support disabling the strict aliasing.

andyross, 2022-08-11:
Regarding performance: zephyr (obviously) builds with -Os and favors size optimizations.  Generally where you worry about optimization that trades size vs. speed you're talking about inlining.  And we do almost none of that at the optimizer level (where we care, it's generally hand-managed via ALWAYS_INLINE).  So all other thing being equal, you'd expect Zephyr performance to be linear with code size (smaller Zephyr binaries should be expected to run faster, not slower).

Which is a long winded way of saying I agree with @stephanosio and would expect this to have minimal performance impact.  (Which, again, is counter to my earlier expectation.  But numbers are numbers.)

closed 2022-08-11 by carlescufi

abrodkin, 2022-08-11:
> The footprint ramifications of setting `-fno-strict-aliasing` is minimal.
> 
> Building `samples/philosophers` for `qemu_cortex_m3`:
> 
> `-Os -fstrict-aliasing` (current)
> 
> ```
> Memory region         Used Size  Region Size  %age Used
>            FLASH:       18156 B       256 KB      6.93%
>             SRAM:       17232 B        64 KB     26.29%
>         IDT_LIST:          0 GB         2 KB      0.00%
> ```
> 
> `-Os -fno-strict-aliasing` (to be)
> 
> ```
> Memory region         Used Size  Region Size  %age Used
>            FLASH:       18184 B       256 KB      6.94%
>             SRAM:       17232 B        64 KB     26.29%
>         IDT_LIST:          0 GB         2 KB      0.00%
> ```

Live and learn - use Linux kernel's `bloat-o-meter`:
```
../linux/scripts/bloat-o-meter zephyr-strict-aliasing.elf zephyr-no-strict-aliasing.elf
add/remove: 0/0 grow/shrink: 6/0 up/down: 40/0 (40)
Function                                     old     new   delta
uart_stellaris_init                          126     144     +18
rb_insert                                    296     302      +6
z_sys_init_run_level                          64      68      +4
stack_left_limb                               54      58      +4
remove_timeout                                52      56      +4
rb_remove                                    852     856      +4
Total: Before=33353, After=33393, chg +0.12%
```

So indeed, there's no visible difference!

PR #48889 (2022-08-11): cmake: compiler: Add flag for disabling strict aliasing
This series adds the infrastructure for specifying the flag for disabling the strict aliasing rule.

```
This commit adds a template for specifying the C/C++ compiler flag for
disabling the strict aliasing rule.

It also enables this flag globally because the Zephyr codebase does not
strictly adhere to the aliasing rules specified by the C/C++ standards
and the optimisation strategies that assume these rules may end up
generating invalid code.

For instance, GCC 11 and above tend to optimise more aggressively
assuming the strict adherence [... cut]