zephyrproject-rtos/zephyr #44473: net: tcp: Connection does not properly terminate when connection is lost

ssharks, 2022-04-01:
When during a connection, there is no packet traffic possible any more, this is not handled properly.

A testcase has been added in:
https://github.com/ssharks/zephyr/tree/net/tcp_interface_down

Run using:
https://github.com/ssharks/zephyr/tree/net/tcp_interface_down

carlescufi, 2022-04-01:
@ssharks do you plan to send a PR with the test?

ssharks, 2022-04-01:
During debugging some once in a while issues I noticed when the cellular connection goes down, the TCP connections are not properly terminated. This can result in an application hang.
The testcase simulates this down connection by midway during the connection throwing away all packets between server and client. A send is done, so the socket can observe there is no way any more to reach to other side. A following recv is expected to eventually return with an error code because the connection is being terminated.

@[carlescufi](https://github.com/carlescufi) Is this what you where looking after

rlubos, 2022-04-12:
I think we're mixing up client side and server side errors here. Let's start with the client side:
1) The connection is established with the server, after what we block any traffic at the loopback interface.
2) `send()` call is issued - this call suceeds as we sucesfully queue the data for transmission at the TCP level.
3) The TCP layer however does not receive ACK for the data it sends, which leads to retransmission and eventually the connection is dropped (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/tcp.c#L1134)
4) At the test level, we don't detect that the client-side connection is closed as we're blocked at the server side `recv()` (https://github.com/ssharks/zephyr/blob/net/tcp_interface_down/tests/net/socket/tcp/src/main.c#L436). If we remove the `recv()` on the server socket, the `recv()` on the client socket does return, indicating that the connection is closed (returning 0). This might be something to improve though - I think we should return an error code instead here, as the connection did not really close gracefully. `ETIMEDOUT` would be a better choice here (according to man pages: ETIMEDOUT - The connection timed out during connection establishment, **or due to a transmission  time‐out on active connection.**)

Now what happens on the server side:
1) We accept the connection from the client, which ends up in a new socket being allocated (`new_sock`)
2) The test blocks on the `recv()` call on the `new_sock`. Now from the server side perspective, nothing happens on the connection (as loopback driver drops all the data), no new packets arrive - so it's perfectly normal that the socket remains blocked. I don't see how retransmission issues on the client socket should affect the server socket - while they are both two ends of the same connection, those are two separate TCP contexts, not aware of each other.

Now the above rises a question - should we implement some kind of idle timeout at the TCP level? I read that this is implemented in Linux, but I'm not sure if there actually any RFC covering the topic, perhaps you know @jukkar? Still, such a idle timeout should rather be considered to be counted in minutes rather than seconds, so it's not really something your test would cover. I'd also consider this as an enhancement not a bugfix.

jukkar, 2022-04-12:
If the connection is already established, and if the TCP keepalive (https://tldp.org/HOWTO/TCP-Keepalive-HOWTO/overview.html) is disabled, then there is no "idle timeout" in TCP. Both ends can idle and not transmit anything without the connection being dropped in this case.

rlubos, 2022-04-13:
> If the connection is already established, and if the TCP keepalive (https://tldp.org/HOWTO/TCP-Keepalive-HOWTO/overview.html) is disabled, then there is no "idle timeout" in TCP. Both ends can idle and not transmit anything without the connection being dropped in this case.

Ok, so if I understand correctly we should rarher focus on a keep alive implementation to detect dead peers. But that's already covered with https://github.com/zephyrproject-rtos/zephyr/issues/34044, AFAIR there was even a PR but got abandoned.

jukkar, 2022-04-13:
Perhaps #35318 could be fixed and reused instead of starting from scratch.

ssharks, 2022-04-14:
Thanks for looking into it properly. Find the replies to the different sections below.

3. It is good news that the socket is indeed closed by the failed send. This is the intended behavior I was trying to test using this test.

4. The blocking at server level was not intended. The timeout should have been set at the new_sock instead of the client socket, to pass this section properly. I updated the pull request to fix this.

If I read multiple sources correctly like (https://man7.org/linux/man-pages/man2/recv.2.html) you have the shutdown(SHUT_WR), recv sequence, which indeed return a 0 on a blocking call when the other half of the socket is closed. This actually the behavior the function test_eof checks.

This is different from a timed out receive, because that will return with an -1 and error code EAGAIN.

Based on this it looks the current implementation is correct, I had to just modify the test case to correctly validate this behavior. After modifying the test I do see it passing properly.

carlescufi, 2022-05-11:
This is being tracked in #34044

closed 2022-05-11 by carlescufi