zephyrproject-rtos/zephyr #44486: nucleo_f429zi: multiple networking tests failing

cfriedt, 2022-04-03:
**Describe the bug**
It looks as though a recent change in the STM32 Ethernet HAL has broken a number of networking tests for `nucleo_f429zi`.

Please also mention any information which could help others to understand
the problem you're facing:
 - **What target platform are you using?**
 Twister in CI
 - **What have you tried to diagnose or workaround this issue?**
 - looked at twister output in CI

**To Reproduce**
Steps to reproduce the behavior:
1. `twister -i -p nucleo_f429zi -s samples/net/civetweb/http_server/sample.net.civetweb.http_server`
2. `twister -i -p nucleo_f429zi -s tests/net/socket/websocket/net.socket.websocket`
3. `twister -i -p nucleo_f429zi -s tests/net/socket/udp/net.socket.udp`
4. `twister -i -p nucleo_f429zi -s tests/net/socket/udp/net.socket.udp.preempt`
5. `twister -i -p nucleo_f429zi -s tests/net/socket/tls/net.socket.tls.preempt`
7. `twister -i -p nucleo_f429zi -s tests/net/socket/tls/net.socket.tls`
8. `twister -i -p nucleo_f429zi -s tests/net/socket/tcp/net.socket.tcp.preempt`
9. `twister -i -p nucleo_f429zi -s tests/net/socket/tcp/net.socket.tcp`
10. `twister -i -p nucleo_f429zi -s tests/net/socket/socketpair/net.socket.socketpair`
11. `twister -i -p nucleo_f429zi -s tests/net/socket/socketpair/net.socket.socketpair.newlib`
12. `twister -i -p nucleo_f429zi -s tests/net/socket/select/net.socket.select.preempt`
13. `twister -i -p nucleo_f429zi -s tests/net/socket/select/net.socket.select`
14. `twister -i -p nucleo_f429zi -s tests/net/socket/register/net.socket.register.tls`
15. `twister -i -p nucleo_f429zi -s tests/net/socket/register/net.socket.register`
16. `twister -i -p nucleo_f429zi -s tests/net/socket/poll/net.socket.poll`
17. `twister -i -p nucleo_f429zi -s tests/net/socket/misc/net.socket.misc.userspace`
18. `twister -i -p nucleo_f429zi -s tests/net/socket/misc/net.socket.misc`
19. `twister -i -p nucleo_f429zi -s tests/net/socket/af_packet_ipproto_raw/net.socket.af_packet.ipproto_raw`
20. `twister -i -p nucleo_f429zi -s tests/net/socket/af_packet/net.socket.af_packet`
21. `twister -i -p nucleo_f429zi -s samples/net/sockets/websocket_client/sample.net.sockets.websocket_client`
22. `twister -i -p nucleo_f429zi -s samples/net/sockets/socketpair/sample.net.sockets.socketpair`
...
See errors

There are 35 errors in total, and they are all on `nucleo_f429zi`.

```
- 583 of 668 test configurations passed (94.34%), 35 failed, 50 skipped with 0 warnings in 3190.26 seconds
```

**Expected behavior**
This should build without error.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
FAILED: zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_stm32_hal.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F429xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../../../../../../../samples/net/civetweb/http_server/../common/include -I../../../../../../../include -Izephyr/include/generated -I../../../../../../../soc/arm/st_stm32/stm32f4 -I../../../../../../../include/posix -I../../../../../../../drivers -I../../../../../../../soc/arm/st_stm32/common -I../../../../../../../subsys/net/l2 -I../../../../../../../subsys/net/lib/sockets/. -I../../../../../../../subsys/net/lib/dns/. -I/__w/zephyr/modules/lib/civetweb/include -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/modules/hal/atmel/include -I/__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/__w/zephyr/modules/hal/stm32/stm32cube/common_ll/include -isystem ../../../../../../../lib/libc/minimal/include -isystem /opt/toolchains/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /opt/toolchains/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Os -imacros /__w/zephyr/zephyr/twister-out/nucleo_f429zi/samples/net/civetweb/http_server/sample.net.civetweb.http_server/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /__w/zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -Werror -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/__w/zephyr/zephyr/samples/net/civetweb/http_server=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -include /__w/zephyr/zephyr/samples/net/civetweb/http_server/../common/include/libc_extensions.h -MD -MT zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_stm32_hal.c.obj -MF zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_stm32_hal.c.obj.d -o zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_stm32_hal.c.obj -c /__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c: In function 'read_eth_phy_register':
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c:105:47: error: passing argument 3 of 'HAL_ETH_ReadPHYRegister' makes integer from pointer without a cast [-Werror=int-conversion]
  105 |  return HAL_ETH_ReadPHYRegister(heth, PHYReg, RegVal);
      |                                               ^~~~~~
      |                                               |
      |                                               uint32_t * {aka unsigned int *}
In file included from /__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_conf.h:332,
                 from /__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal.h:29,
                 from /__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/soc/stm32f4xx.h:287,
                 from ../../../../../../../soc/arm/st_stm32/stm32f4/soc.h:23,
                 from ../../../../../../../include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from ../../../../../../../include/arch/arm/aarch32/mpu/arm_mpu_v7m.h:11,
                 from ../../../../../../../include/arch/arm/aarch32/mpu/arm_mpu.h:14,
                 from ../../../../../../../include/arch/arm/aarch32/arch.h:194,
                 from ../../../../../../../include/arch/cpu.h:19,
                 from ../../../../../../../include/kernel_includes.h:33,
                 from ../../../../../../../include/kernel.h:17,
                 from ../../../../../../../include/zephyr.h:18,
[... 154 lines cut ...]
/__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_eth.h:1338:39: error: initialization of 'ETH_DMADescTypeDef *' from 'unsigned int' makes pointer from integer without a cast [-Werror=int-conversion]
 1338 | #define ETH_CHECKSUM_BY_SOFTWARE      0x00000001U
      |                                       ^~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c:1096:20: note: in expansion of macro 'ETH_CHECKSUM_BY_SOFTWARE'
 1096 |    .ChecksumMode = ETH_CHECKSUM_BY_SOFTWARE,
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_eth.h:1338:39: note: (near initialization for 'eth0_data.heth.Init.RxDesc')
 1338 | #define ETH_CHECKSUM_BY_SOFTWARE      0x00000001U
      |                                       ^~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c:1096:20: note: in expansion of macro 'ETH_CHECKSUM_BY_SOFTWARE'
 1096 |    .ChecksumMode = ETH_CHECKSUM_BY_SOFTWARE,
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c: In function 'read_eth_phy_register':
/__w/zephyr/zephyr/drivers/ethernet/eth_stm32_hal.c:107:1: error: control reaches end of non-void function [-Werror=return-type]
  107 | }
      | ^
cc1: all warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: Linux / macOS / CI
 - Toolchain: Zephyr SDK v0.14.0
 - Commit SHA or Version used: 018f59e46fcfb1cf90d9ba44735ebc00590cef2e

**Additional context**

erwango, 2022-04-04:
I'm wonder how this has not been detected during our CI tests.

erwango, 2022-04-04:
> I'm wonder how this has not been detected during our CI tests.

Got it: Network tests are quite time consuming (several hours) while they don't actually test the ethernet driver. To fasten our daily CI, and because there is no easy way to add a single test to a twister test plan, we had to remove them fully. Issue is that now we rely on manual operation to verify the driver is still building...

Their only usefulness is to build the driver..

JordanYates, 2022-04-06:
This problem could have been picked up by the `drivers/build_all/ethernet` test, without actually running anything: `west build -b nucleo_f429zi zephyr/tests/drivers/build_all/ethernet/`.

erwango, 2022-04-06:
> This problem could have been picked up by the `drivers/build_all/ethernet` test, without actually running anything: `west build -b nucleo_f429zi zephyr/tests/drivers/build_all/ethernet/`.

Ok, I missed that one. Issue is that this one isn't used when launching twister session w/o build_only (and here I made another fault which is not launching a build_only session).
Anyway, I'll try to take advantage of this somehow on our bench. Thanks for pointing it to me.

closed 2022-04-06 by carlescufi