zephyrproject-rtos/zephyr #44504: net: tcp: Context still open after timeout on connect

ssharks, 2022-04-04:
When a TCP connect fails with a timeout, because there is no communication possible, the context is not cleaned up properly.

Can be reproduced using the testcase of:
https://github.com/zephyrproject-rtos/zephyr/pull/44501

Run the test using
./scripts/twister -T tests/net/socket/tcp -p qemu_x86

12csw1, 2022-04-04:
Should this issue be labeled as priority: low?

We are doing research about “Issue Priority Prediction". We have developed a tool to automatically label the priority of an issue. The basic idea is to employ Bert model to encode the textual and code descriptions of the issue, and incorporate the issue’s attributes in the model, then apply multi-task learning to facilitate other information boosting the prediction performance. 
 We have conducted experiments on four GitHub projects with 21700 issues, and results show that [79 accuracy]() can be achieved.

Our tool shows that this issue has a very large possibility to be labeled as priority: low 
Could you please tell us whether it should be labeled as priority: low?
And if you have any comments or suggestions with this tool, could you please tell us? 
Thank you so much.

rlubos, 2022-04-13:
> So if I understand you correctly, a socket should be closed even when the connect fails?

That's right, once opened the only way to clean up its resources is to close it. Anyway, it's generally expected that you always close a socket that is no longer used, it's not only about net_context but also the file description needs to be freed etc.

closed 2022-04-14 by ssharks