zephyrproject-rtos/zephyr #44507: net: tcp: No retries of a TCP FIN message

ssharks, 2022-04-04:
When a connection is opened and a FIN message does not reach the server, it is never retried (no ACK received). The context is properly closed, by the connection stays open at the server because it did not receive the FIN message.

A test to show the issue:
https://github.com/ssharks/zephyr/tree/net/tcp_obstructed_close

Run using:
./scripts/twister -T tests/net/socket/tcp -p qemu_x86

ssharks, 2022-04-07:
@jukkar I agree with you on the priority the other 2 open issues are more urgent. It is a situation that should be able to resolve automatically by the other side of the connection. The timeout of stale connections can vary between ~4 minutes on Azure or 2 hours for default Linux. When you are connecting to another embedded system it is more critical as the number of resources there is limited.

On the number of FIN messages. The close message brings the state diagram from ESTABLISHED to FIN_WAIT1 state. Until is receives an ACK or FIN|ACK message, it stays there. The current implementation jumps to CLOSED state after a second. The server side however never got the FIN message, so that stays in the ESTABLISHED state.

Lets go extreme: if the packet loss rate is 5%, without retries, there is a 5% chance the server side connection becomes stale. With 3 retries (assuming independent packet loss) the chance of not at least 1 packet arriving is reduced to 0.05 * 0.05 * 0.05 = 0.0125%.
Take into account that the FIN could be lost, but also the following ACK message.

The downside is that when it keep retrying a long time, this is a potential power drain on an embedded device.

rlubos, 2022-04-14:
Actually, after investigating the issue a bit it seems that the FIN retransmission **is** supported correctly. The retransmission is handled by `conn->send_timer` and takes place after every `CONFIG_NET_TCP_INIT_RETRANSMISSION_TIMEOUT`. 

I will comment on the test case PR on what's wrong with the test.

carlescufi, 2022-05-11:
Closing as per [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/44507#issuecomment-1099066019)

closed 2022-05-11 by carlescufi