zephyrproject-rtos/zephyr #44525: Replacing zephyr driver/subsys implementations

gregshue, 2022-04-04:
## Introduction

Zephyr Project platinum members need to be able to reuse a zephyr release (at a specific specific SHA) so that the available security audit artifacts can be applied to their projects. They also need to be able to address driver and subsystem implementation issues that exist within that reused repository without changing the SHA of the audited repository. This means that through the build/configuration system each subsystem and driver implementation must be individually replaceable with alternate implementations. This is already possible using the Kconfig, but the capability is not universally implemented and a standard implementation pattern has not specified.

### Problem description

See Introduction

### Proposed change

Align on the need (requirement) for this functionality.
Specify a naming pattern for Kconfigs to control including/excluding a specific implementation.
Update the review process to verify this requirement is met.
Update every subsystem/driver with this pattern.

## Detailed RFC

See Proposed change.  

### Proposed change (Detailed)

See Proposed Change

### Dependencies

This may impact every driver and subsystem, and the projects using it. The scope depends on the specific pattern identified.

### Concerns and Unresolved Questions

None known.

## Alternatives

None identified.

Laczen, 2022-04-05:
@gregshue, since this only seems important for "Zephyr Project platinum members", I propose they assign resources to this to follow it up and keep in line with their requirements. No need to require the complete community to follow something if it only is important to some.

gregshue, 2022-04-05:
@Laczen Thanks for the comment.  Apparently I need to enhance the description, because **any downstream user may have the same need** when they are not allowed to change the forked repositories for some reason and an issue in one or more read-only repositories must be resolved (e.g., symbol conflict between repositories, newly discovered logical error, etc).

The specific reason downstream users may not be allowed to change the forked repositories is beyond our control, not up for debate, and not relevant since the Platinum Members face a similar need.

Does the edited description now indicate why this is a broad need?

nashif, 2022-04-05:
@gregshue please remove the language about member tiers and make this RFC generic. Nothing in upstream Zephyr should be implemented to satisfy only the need of one tier.

gregshue, 2022-04-05:
I have rephrased the need to not reference a member tier.

> Nothing in upstream Zephyr should be implemented to satisfy only the need of one tier.

This statement is contradictory to the nature of requirements. Requirement management must capture unique user needs, whether or not that exists for only one tier.

nashif, 2022-04-05:
> This statement is contradictory to the nature of requirements. Requirement management must capture unique user needs, whether or not that exists for only one tier.

sure, but users in this aspect should not be based on tiers of the project. Please leave this out and talk about your specific needs as a user and do not speak for "platinum member".

gregshue, 2022-04-05:
It looks like I missed selecting "update comment" on the description.  It is updated now.