zephyrproject-rtos/zephyr #44530: xtensa xcc build usb stack fail (newlib)

hongshui3000, 2022-04-05:
**Describe the bug**
XCC newlib does not support ECANCELED
usb code：usb_transfer.c
```
void usb_cancel_transfer(uint8_t ep)
{
	struct usb_transfer_data *trans;
	unsigned int key;

	key = irq_lock();

	trans = usb_ep_get_transfer(ep);
	if (!trans) {
		goto done;
	}

	if (trans->status != -EBUSY) {
		goto done;
	}

	trans->status = -ECANCELED; --------------------------------------------------->newlib build fail
	k_work_submit_to_queue(&USB_WORK_Q, &trans->work);

done:
	irq_unlock(key);
}
```

mbolivar-nordic, 2022-04-05:
@stephanosio to check which version of newlib introduced ECANCELED

stephanosio, 2022-04-05:
`ECANCELED` was added to the newlib back in 2007:
https://github.com/bminor/newlib/commit/6699fabb916055563b105cda4d0789e862c0760a#diff-c8661ca0a63aaa7035272bbbcaf836f850d36b484bf5682e182b381dd768de60R151

Noting that XCC is based on the GCC 4.x, which was released as early as 2005, it is possible that the version of the newlib included in the XCC is just as old and does not have `ECANCELED` defined in the `errno.h`.

mbolivar-nordic, 2022-04-15:
So what do we do here? Close as "won't fix"? According to [errno(3)](https://www.man7.org/linux/man-pages/man3/errno.3.html), ECANCELED was standardized in POSIX.1-2001, so it seems like fair game in 2022 to assume it's available in errno.h.

nashif, 2022-04-19:
closing this, there is really not much we can do here. This should be addressed by the user of the toolchain which does not have this support.

closed 2022-04-19 by nashif