zephyrproject-rtos/zephyr #44531: bl654_usb without mcuboot maximum image size is not limited

lairdjm, 2022-04-05:
**Describe the bug**
When using the bl654_usb board without mcuboot, the size of the image is not limited and so large applications can be generated that actually impose on the bootloader's flash space.

**To Reproduce**
Build any application with an output size of > 892KB for bl654_usb

**Expected behavior**
To prevent successful firmware builds

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.0
 - Commit SHA or Version used: e8a369e7da50f5fe9699e9c6b90e5a5e3fdc4e15

12csw1, 2022-04-05:
Should this issue be labeled as priority: low?

We are doing research about “Issue Priority Prediction". We have developed a tool to automatically label the priority of an issue. The basic idea is to employ Bert model to encode the textual and code descriptions of the issue, and incorporate the issue’s attributes in the model, then apply multi-task learning to facilitate other information boosting the prediction performance.
We have conducted experiments on four GitHub projects with 21700 issues, and results show that [79 accuracy](https://github.com/zephyrproject-rtos/zephyr/issues/44504) can be achieved.

Our tool shows that this issue has a very large possibility to be labeled as priority: low
Could you please tell us whether it should be labeled as priority: low?
And if you have any comments or suggestions with this tool, could you please tell us?
Thank you so much.

stephanosio, 2022-04-05:
@12csw1 Instead of commenting on each issue, please create a dedicated issue for this study and ask feedback in that issue, with references to the related issues.

closed 2022-04-06 by carlescufi