zephyrproject-rtos/zephyr #44539: twister fails on several stm32 boards with tests/arch/arm testcases

FRASTM, 2022-04-05:
**Describe the bug**
It appears that several (various) test cases of the tests/arch/arm running on the nucleo_wb55rg or nucleo_f103rb 
are failing due to **FAILED: Testsuite mismatch**

twister reports failed test : 
`twister - ERROR - nucleo_wb55rg             tests/arch/arm/arm_irq_advanced_features/arch.arm.irq_advanced_features FAILED : Testsuite mismatch`

handler.log reports:
```
================================================================
PROJECT EXECUTION SUCCESSFUL
```

--> It appears that reverting the https://github.com/zephyrproject-rtos/zephyr/pull/42482/
can fix the pb

**To Reproduce**
Steps to reproduce the behavior:
1. twister --device-testing - --hardware-map ../map.yaml --no-update  -T tests/arch/arm

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Toolchain : zephyr-sdk-0.13.2
 - Commit SHA : db3a6e17faedf1030055352edc05e123cecc4d3e

PerMac, 2022-04-05:
I think it is related to changes introduced to ztest. #42482 shouldn't be reverted. This patch exposed tests that where incorrectly handled by twister. This patch verifies if the proper test suite was run on hw. If it would run correctly you should get sth like
```
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.0.0-2155-g4e9d9f2ef7a1  (delayed boot 500ms) ***
Running test suite arm_irq_advanced_features
===================================================================
[resuts of test cases]
===================================================================
Test suite common succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

CC @gopiotr @yperess

nashif, 2022-04-05:
> I think it is related to changes introduced to ztest.

I do not see how is this related to ztest changes, the output looks completely sane to me. This could be related to the same issue in #44438 and order of execution.

@FRASTM  do you get the same results if you run twister only on this test and nothing else?

nashif, 2022-04-05:
I think the culprit here is `tests/arch/arm/arm_runtime_nmi` which require some after-action to have tests running after that pass.

nashif, 2022-04-05:
with this change

```diff
diff --git a/scripts/pylib/twister/twisterlib.py b/scripts/pylib/twister/twisterlib.py
index e8cdf0b643..53a4b515d4 100755
--- a/scripts/pylib/twister/twisterlib.py
+++ b/scripts/pylib/twister/twisterlib.py
@@ -612,7 +612,7 @@ class BinaryHandler(Handler):
         if self.call_make_run:
             command = [self.generator_cmd, "run"]
         elif self.call_west_flash:
-            command = ["west", "flash", "--skip-rebuild", "-d", self.build_dir]
+            command = ["west", "flash", "--erase", "--skip-rebuild", "-d", self.build_dir]
         else:
             command = [self.binary]

```

it works fine.

PerMac, 2022-04-06:
>I do not see how is this related to ztest changes, the output looks completely sane to me

In the first output only `PROJECT EXECUTION SUCCESSFUL` is given. That's why I thought it might be this error when a empty test suite were build instead of proper ones, like here #42396

gopiotr, 2022-04-06:
@nashif thank you for letting me know about possible source of problem. I'll try to look at this today and propose some changes. Please do not revert "test suite verification" yet. It will be hard to verify if my propositions could fix those problem - it occur only in specific situations and I am not able to reproduce them easily locally or in our CI.

FRASTM, 2022-04-06:
>

FRASTM, 2022-04-06:
@nashif thanks. Trying with this change in scripts/pylib/twister/twisterlib.py

gopiotr, 2022-04-06:
> with this change
> ...
> it works fine.

@nashif how changes into BinaryHandler could help solve any problem during testing on real hardware? I guess that you think about changes in DeviceHadnler here:

```
diff --git a/scripts/pylib/twister/twisterlib.py b/scripts/pylib/twister/twisterlib.py
index e8cdf0b643..790d02f688 100755
--- a/scripts/pylib/twister/twisterlib.py
+++ b/scripts/pylib/twister/twisterlib.py
@@ -825,7 +825,7 @@ class DeviceHandler(Handler):
         logger.debug(f"Using serial device {serial_device} @ {hardware.baud} baud")
 
         if (self.suite.west_flash is not None) or runner:
-            command = ["west", "flash", "--skip-rebuild", "-d", self.build_dir]
+            command = ["west", "flash", "--erase", "--skip-rebuild", "-d", self.build_dir]
             command_extra_args = []
 
             # There are three ways this option is used.

```

BTW @nashif @FRASTM you can achieve the same result by add `--west-flash="--erase"` argument to Twister call instead of change Twister source code. For example you can run Twister by this command:
```
twister --device-testing - --hardware-map ../map.yaml -T tests/arch/arm  --west-flash="--erase"
```

gopiotr, 2022-04-06:
@FRASTM @hakehuang @nashif 

I am still not able to reproduce this error locally. Could you be so kind and run Twister on your hardware by this command (change platform name of course):
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nrf52840dk_nrf52840 -T tests/arch/arm -v -v -j 1
```
And then attach all `twister.log` file here? Without this I am not able to verify what is the order of tests which you run and what is the full output of performed tests.

FRASTM, 2022-04-06:
running scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg  -T tests/arch/arm -v -v -j 1
[twister.log](https://github.com/zephyrproject-rtos/zephyr/files/8426441/twister.log)

gopiotr, 2022-04-06:
@FRASTM Thank you very much for this log! Now it is much more cleaner for me and now I am able to say something more.

In your log it can be observed, that at first `tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi` test is build and executed:
```
2022-04-06 13:08:47,443 - twister - DEBUG - Running cmake on /home/r/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for nucleo_wb55rg
2022-04-06 13:08:47,444 - twister - DEBUG - Calling cmake with arguments: ['-B/home/r/zephyrproject/zephyr/twister-out/nucleo_wb55rg/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi', '-S/home/r/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi', '-DEXTRA_CFLAGS=-Werror', '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DBOARD=nucleo_wb55rg']
2022-04-06 13:09:00,265 - twister - DEBUG - Finished building /home/r/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for nucleo_wb55rg
2022-04-06 13:09:00,268 - twister - DEBUG - build test: nucleo_wb55rg/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
2022-04-06 13:09:00,268 - twister - DEBUG - Building /home/r/zephyrproject/zephyr/tests/arch/arm/arm_runtime_nmi for nucleo_wb55rg
2022-04-06 13:09:47,130 - twister - DEBUG - run test: nucleo_wb55rg/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi
2022-04-06 13:09:47,131 - twister - DEBUG - Using serial device /dev/ttyACM0 @ 115200 baud
2022-04-06 13:09:47,164 - twister - DEBUG - Flash command: ['make', '-C', '/home/r/zephyrproject/zephyr/twister-out/nucleo_wb55rg/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi', 'flash']
2022-04-06 13:09:54,768 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-2317-g56dc490afe36  ***
2022-04-06 13:09:54,771 - twister - DEBUG - DEVICE: Running test suite arm_runtime_nmi_fn
2022-04-06 13:09:54,776 - twister - DEBUG - DEVICE: ===================================================================
2022-04-06 13:09:54,780 - twister - DEBUG - DEVICE: START - test_arm_runtime_nmi
2022-04-06 13:09:54,786 - twister - DEBUG - DEVICE: START - nmi_test_isr
2022-04-06 13:09:54,787 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 0 s
2022-04-06 13:09:55,787 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 1 s
2022-04-06 13:09:56,790 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 2 s
2022-04-06 13:09:57,791 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 3 s
2022-04-06 13:09:58,793 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 4 s
2022-04-06 13:09:59,795 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 5 s
2022-04-06 13:10:00,797 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 6 s
2022-04-06 13:10:01,799 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 7 s
2022-04-06 13:10:02,801 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 8 s
2022-04-06 13:10:03,803 - twister - DEBUG - DEVICE: Trigger NMI in 10s: 9 s
2022-04-06 13:10:04,806 - twister - DEBUG - DEVICE: NMI received (test_handler_isr)! Rebooting...
2022-04-06 13:10:04,810 - twister - DEBUG - DEVICE: PASS - nmi_test_isr in 10.26 seconds
2022-04-06 13:10:04,816 - twister - DEBUG - DEVICE: ===================================================================
2022-04-06 13:10:04,822 - twister - DEBUG - DEVICE: ===================================================================
2022-04-06 13:10:04,824 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
2022-04-06 13:10:04,842 - twister - DEBUG - run status: nucleo_wb55rg/tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi passed
```

And then after that, `tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info` test is built and run as follow:
```
2022-04-06 13:10:04,845 - twister - DEBUG - Running cmake on /home/r/zephyrproject/zephyr/tests/arch/arm/arm_interrupt for nucleo_wb55rg
2022-04-06 13:10:04,846 - twister - DEBUG - Calling cmake with arguments: ['-B/home/r/zephyrproject/zephyr/twister-out/nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info', '-S/home/r/zephyrproject/zephyr/tests/arch/arm/arm_interrupt', '-DEXTRA_CFLAGS=-Werror', '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DOVERLAY_CONFIG=/home/r/zephyrproject/zephyr/twister-out/nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info/twister/testcase_extra.conf', '-DBOARD=nucleo_wb55rg']
2022-04-06 13:10:14,317 - twister - DEBUG - Finished building /home/r/zephyrproject/zephyr/tests/arch/arm/arm_interrupt for nucleo_wb55rg
2022-04-06 13:10:14,319 - twister - DEBUG - build test: nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info
2022-04-06 13:10:14,319 - twister - DEBUG - Building /home/r/zephyrproject/zephyr/tests/arch/arm/arm_interrupt for nucleo_wb55rg
2022-04-06 13:11:19,007 - twister - DEBUG - run test: nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info
2022-04-06 13:11:19,007 - twister - DEBUG - Using serial device /dev/ttyACM0 @ 115200 baud
2022-04-06 13:11:19,010 - twister - DEBUG - Flash command: ['make', '-C', '/home/r/zephyrproject/zephyr/twister-out/nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info', 'flash']
2022-04-06 13:11:19,015 - twister - DEBUG - DEVICE: ================
2022-04-06 13:11:19,018 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
[... cut ...]
Here, you can observe, that Twister do not see full test output (perhaps your device was not able to print full output after run `arch.interrupt.arm.nmi`). So, for me it is now clear, that this test is marked as `failed`, because Twister is not able to recognize if proper test was executed (`PROJECT EXECUTION SUCCESSFUL` info is not enough to make sure, that exactly `arch.interrupt.extra_exception_info` was performed).

@FRASTM to be sure: could you paste here the output logs from this file:
`twister-out/nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info/handler.log`
?

gopiotr, 2022-04-06:
@FRASTM one more question: do you observe `arch.interrupt.extra_exception_info` test as failed every time you run this command:
`scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -T tests/arch/arm -v -v -j 1`
or this is random behavior?

nashif, 2022-04-06:
> @nashif how changes into BinaryHandler could help solve any problem during testing on real hardware? I guess that you think about changes in DeviceHadnler here:

yes, the change was to DeviceHandler, I had to do it quickly to be posted to this issue and did not notice.

nashif, 2022-04-06:
btw, the nmi test has other issues, ie..:

```
2022-04-06 13:09:54,780 - twister - DEBUG - DEVICE: START - test_arm_runtime_nmi
2022-04-06 13:09:54,786 - twister - DEBUG - DEVICE: START - nmi_test_isr
```
We are manually printing that a certain test has started, so we have a discrepancy there as well. 
But the main issue I observed is with anything running after this test, the device basically becomes unusable and output is not consistent...

gopiotr, 2022-04-06:
> btw, the nmi test has other issues, ie..:

@nashif yes you are right. This test shouldn't print additionally `START - nmi_test_isr`. I observed, that `TC_START()` macro is used in situation, when ztest is not used, like there:
https://github.com/zephyrproject-rtos/zephyr/blob/d10a8f13af8d7a83dbc91be6e381df896820e5b2/tests/benchmarks/latency_measure/src/main.c#L42

But in this `nmi` test it has no sense to use additionally `TC_START()` macro, because it is already used, when `ztest_run_test_suite()` function is called.

gopiotr, 2022-04-06:
> It seems the failure is always on twister-out/nucleo_wb55rg/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info :
> 
> handler.log

@FRASTM okey, so for me now it is clear, that this is not a problem with Twister and "testsuite name verification" it is rather problem with `nmi` test. In my opinion we shouldn't revert "testsuite name verification", because it help us to discover problem with this test. Now this test should be fixed. I don't know how `NMI` works and how this test should be changed to not impact on other one. Eventually it could be a problem with `nucleo_wb55rg` board, but probably not because this issue was observed on other platforms yet.

@FRASTM could you create new issue relevant to this `nmi` test, and describe exactly what to do to reproduce this error by run
`scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -T tests/arch/arm -v -v -j 1`

nashif, 2022-04-06:
> @FRASTM okey, so for me now it is clear, that this is not a problem with Twister and "testsuite name verification" it is rather problem with `nmi` test.

This is just one test of many, the issue is not limited to this test. I also noticed this with the coredump feature.

> In my opinion we shouldn't revert "testsuite name verification", because it help us to discover problem with this test (it probably exists before introduction this additional verification).

Nobody is talking about a revert, but we need to deal with the issue and have a way to deal with such cases. There is not necessarily a problem with the test case itself, it is the type of test and what is being tested that introduce issues in the workflow. We will get more of those when we start testing power management related features more aggressively. 

> Eventually it could be a problem with `nucleo_wb55rg` board, but probably not because this issue was observed on other platforms yet.

No, this is not board related. I am also able to reproduce this on nordic boards.

> @FRASTM could you create new issue relevant to this `nmi` test, and describe exactly what to do to reproduce this error by run
> `scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -T tests/arch/arm -v -v -j 1`

as I said, this is not a test problem, this is a test execution and special test handling problem that needs to be addressed in twister and how we test such cases.

gopiotr, 2022-04-07:
@nashif 

> This is just one test of many, the issue is not limited to this test. I also noticed this with the coredump feature.

You still mention, that "many" tests behave similar to this `nmi` test, but could you indicate which one exactly do this? Which particular `coredump` feature test? I asked because I still cannot reproduce such behavior on my local boards and trust me - it is hard to me to try "fix" somehow this issue, when I do not have any way to test some solutions.

> as I said, this is not a test problem, this is a test execution and special test handling problem that needs to be addressed in twister and how we test such cases.

I disagree with you. In my opinion it is definitely test problem. Every test scenario in yaml file should be standalone and shouldn't impact on another tests. Without such assumption we will never be able to make stable CI process. 

How do you imagine handling such situation by Twister? How Twister could recognize if such output:
```
================================================================
PROJECT EXECUTION SUCCESSFUL
```
is okay and this test should pass? Twister do not have such "special" knowledge about performed test and especially it do not have any knowledge, that previous test could impact on output from next one test...

You mentioned, that it helps, when you add `--erase` option to west command. So at this moment I have two propositions:
1. Short-term:
All problematic tests (like `nmi`) could be marked with tag like `require_erase`. Then such tests could be called by this command:
```
scripts/twister --tag require_erase --west-flash="--erase"
```
Rest of test could be run just by:
```
scripts/twister --exclude-tag require_erase
```
2. Long-term:
It would be nice to have command like `west flash --erase` just for erasing (without flashing any program). Then it could be added to yaml testcase file possibility to append entry like `post_erase: true`. For example `nmi` test could looks like this:
```
tests:
  arch.interrupt.arm.nmi:
    filter: CONFIG_ARMV6_M_ARMV8_M_BASELINE or CONFIG_ARMV7_M_ARMV8_M_MAINLINE
    arch_allow: arm
    tags: nmi interrupt arm
    post_erase: true
```
This could be indication for Twister, that after execution of such test it should call `west flash --erase` command.

@nashif What do you think about such propositions?

gopiotr, 2022-04-07:
@FRASTM could you run Twister once again with this command and look if adding `--west-flash="--erase"` will help to solve problem with `nmi` tests?
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -T tests/arch/arm -v -v -j 1 --west-flash="--erase"
```

FRASTM, 2022-04-07:
adding --west-flash="--erase" to the twister command makes the test fail with "FAILED Testsuite mismatch"   on the first run. The next run will give "FAILED  Flash error"

nashif, 2022-04-07:
> You still mention, that "many" tests behave similar to this `nmi` test, but could you indicate which one exactly do this? Which particular `coredump` feature test? I asked because I still cannot [...]

```
i9:zephyr(main): twister   -p reel_board --device-testing --hardware-map ~/map.yaml -v -T tests/arch/arm/
Renaming output directory to /home/nashif/zephyrproject/zephyr/twister-out.5
INFO    - Zephyr version: zephyr-v3.0.0-2457-ge74e8b2048d1
INFO    - JOBS: 20
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 19 test scenarios (19 configurations) selected, 1 configurations discarded due to filters.

Device testing on:

| Platform   | ID                                               | Serial device   |
|------------|--------------------------------------------------|-----------------|
| reel_board | OSHW000032254e4500128002ab98002784d1000097969900 | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  2/19 reel_board                tests/arch/arm/arm_mem_protect/arch.arm.mem_protect.syscalls SKIPPED (filter)
INFO    -  3/19 reel_board                tests/arch/arm/arm_irq_advanced_features/arch.arm.irq_advanced_features.secure_fw SKIPPED (filter)
INFO    -  4/19 reel_board                tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz SKIPPED (filter)
INFO    -  5/19 reel_board                tests/arch/arm/arm_thread_swap_tz/arch.arm.swap.tz_off SKIPPED (filter)
INFO    -  6/19 reel_board                tests/arch/arm/arm_irq_vector_table/arch.arm.irq_vector_table PASSED (device 4.339s)
INFO    -  7/19 reel_board                tests/arch/arm/arm_runtime_nmi/arch.interrupt.arm.nmi PASSED (device 14.201s)
INFO    -  8/19 reel_board                tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info FAILED Testsuite mismatch (device 6.782s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info/handler.log
INFO    -  9/19 reel_board                tests/arch/arm/arm_hardfault_validation/arch.interrupt.arm.hardfault_validation FAILED Testsuite mismatch (device 4.412s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/arch/arm/arm_hardfault_validation/arch.interrupt.arm.hardfault_validation/handler.log
INFO    - 10/19 reel_board                tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations FAILED Testsuite mismatch (device 11.976s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations/handler.log
INFO    - 11/19 reel_board                tests/arch/arm/arm_thread_swap/arch.arm.swap.common FAILED Testsuite mismatch (device 6.809s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/arch/arm/arm_thread_swap/arch.arm.swap.common/handler.log
```

what is the problem? 

```
i9:zephyr(main): cat /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info/handler.log
==========================================
Test suite arm_runtime_nmi_fn succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

but this testsuite is somewhere else...

```
i9:zephyr(main): git grep arm_runtime_nmi_fn
tests/arch/arm/arm_runtime_nmi/src/main.c:      ztest_test_suite(arm_runtime_nmi_fn,
tests/arch/arm/arm_runtime_nmi/src/main.c:      ztest_run_test_suite(arm_runtime_nmi_fn);
```

nashif, 2022-04-07:
> I disagree with you. In my opinion it is definitely test problem. Every test scenario in yaml file should be standalone and shouldn't impact on another tests. Without such assumption we will never be able to make stable CI process.

Sure, if you disagree then please provide a solution within the test itself to address this problem. 

Depending on the hardware feature you are testing, further steps are needed to cleanup the environment to allow additional tests to run. We have seen this happening with features like power management, watchdig, interrupts and so on. 

> How do you imagine handling such situation by Twister? How Twister could recognize if such output:
> ================================================================
PROJECT EXECUTION SUCCESSFUL

This output is not Ok and is not enough to evaluate the issue, we agree on this.

A test runner, read twister should do the setup/teardown for each test to guarantee that the environment is sane and not corrupted by previous tests. Twister need to know how to deal with such cases. The extreme case could be a hardware/power reset after each test, and complete erase of the flash after each test and a power reset, or something along those lines. The above output can come from any test and until now this was acceptable. If we enforce verification of the test being run we need to do more than btw... Your solution assume that if the testsuite name at the beginning and at the end match, then it is ok, but you could be parsing some console output of some other test that is stuck in the hardware, so we are not fully green here, but I guess such situations do not happen very often, however, it is something we need to fix.

This goes back to what I suggested some time ago where each test is assigned an ID during build which is tracked during execution, meaning that twister knows what to expect, because it is the source of the ID, so to illustrate this:
1. twister builds the test with a unique ID
2. test executes and prints the ID on the screen
3. twister matches the id of that test with the one printed on the screen
4. twister only passes the test if the id matches, an old output will have a non-matching ID which would fail the test.

nashif, 2022-04-07:
@gopiotr  re coredump:

```
i9:zephyr(main): twister   -p reel_board --device-testing --hardware-map ~/map.yaml -v -T ./tests/subsys/debug/ --west-flash="--erase"
Renaming output directory to /home/nashif/zephyrproject/zephyr/twister-out.7
INFO    - Zephyr version: zephyr-v3.0.0-2457-ge74e8b2048d1
INFO    - JOBS: 20
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 1 configurations discarded due to filters.

Device testing on:

| Platform   | ID                                               | Serial device   |
|------------|--------------------------------------------------|-----------------|
| reel_board | OSHW000032254e4500128002ab98002784d1000097969900 | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 2/4 reel_board                tests/subsys/debug/coredump/coredump.logging_backend PASSED (device 5.459s)
INFO    - 3/4 reel_board                tests/subsys/debug/coredump_backends/coredump.backends.logging PASSED (device 3.656s)
INFO    - 4/4 reel_board                tests/subsys/debug/coredump_backends/coredump.backends.other FAILED Testsuite mismatch (device 3.552s)
ERROR   - see: /home/nashif/zephyrproject/zephyr/twister-out/reel_board/tests/subsys/debug/coredump_backends/coredump.backends.other/handler.log

INFO    - 2 of 4 test configurations passed (66.67%), 1 failed, 1 skipped with 0 warnings in 21.74 seconds
INFO    - In total 7 test cases were executed, 3 skipped on 1 out of total 450 platforms (0.22%)
INFO    - 3 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board      | ID                                               |   Counter |
|------------|--------------------------------------------------|-----------|
| reel_board | OSHW000032254e4500128002ab98002784d1000097969900 |         3 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
(reverse-i-search)`fla': twister   -p reel_board --device-testing --hardware-map ~/map.yaml -v -T ./tests/subsys/debug/ --west-^Cash="--erase"
i9:zephyr(main): west flash --erase
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner pyocd
-- runners.pyocd: mass erase requested
-- runners.pyocd: Flashing file: /home/nashif/zephyrproject/zephyr/build/zephyr/zephyr.hex
0001050 I Loading /home/nashif/zephyrproject/zephyr/build/zephyr/zephyr.hex [load_cmd]
[==================================================] 100%
0003089 I Erased chip, programmed 24576 bytes (6 pages), skipped 0 bytes (0 pages) at 11.95 kB/s [loader]
i9:zephyr(main): twister   -p reel_board --device-testing --hardware-map ~/map.yaml -v -T ./tests/subsys/debug/ --west-flash="--erase"
Renaming output directory to /home/nashif/zephyrproject/zephyr/twister-out.8
INFO    - Zephyr version: zephyr-v3.0.0-2457-ge74e8b2048d1
INFO    - JOBS: 20
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 1 configurations discarded due to filters.

Device testing on:

| Platform   | ID                                               | Serial device   |
|------------|--------------------------------------------------|-----------------|
| reel_board | OSHW000032254e4500128002ab98002784d1000097969900 | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 2/4 reel_board                tests/subsys/debug/coredump/coredump.logging_backend PASSED (device 5.439s)
INFO    - 3/4 reel_board                tests/subsys/debug/coredump_backends/coredump.backends.other PASSED (device 3.495s)
INFO    - 4/4 reel_board                tests/subsys/debug/coredump_backends/coredump.backends.logging PASSED (device 3.705s)

INFO    - 3 of 4 test configurations passed (100.00%), 0 failed, 1 skipped with 0 warnings in 21.99 seconds
INFO    - In total 7 test cases were executed, 3 skipped on 1 out of total 450 platforms (0.22%)
INFO    - 3 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board      | ID                                               |   Counter |
|------------|--------------------------------------------------|-----------|
| reel_board | OSHW000032254e4500128002ab98002784d1000097969900 |         3 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/nashif/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
i9:zephyr(main):
```

nashif, 2022-04-07:
@gopiotr  quick draft of what I was talking about above: https://github.com/zephyrproject-rtos/zephyr/pull/44661

FRASTM, 2022-04-08:
I have tested with the PR above, The result is much better, except for nucleo_wb55rg and nucleo_f103rb 
For both boards, some test fail, and handler.log is _shifted_ and _truncated_ For example the handler of the /tests/arch/arm/arm_hardfault_validation/arch.interrupt.arm.hardfault_validation/handler.log is not the hardfault_validation one but 
```
phyr-v3.0.0-2459-gdfdfc43fafa1  ***
Running test suite tz_wrap_func
===================================================================
START - test_tz_wrap_func
PASS - test_tz_wrap_func in 0.1 seconds
===================================================================
Test suite tz_wrap_func succeeded
===================================================================
RunID: 15946839231837786741
PROJECT EXECUTION SUCCESSFUL
```

nashif, 2022-04-08:
To summarize the type of issues we are seeing after the introduction of test suite name checking and also the type of issues it detects.

- we had some cases where tests were marked as passed although nothing has run or the list of the testcases were skipped, this afaik was the main motivation for implementing the suite name checking.
- this solution detects such issues based on output on the console and heavily relies on the output from the screen which has some shortcomings and exposed issues where different platforms did not have the complete output available for twister to analyze (out of sync, not connecting in time to serial, full buffer and dropped messages and so on...). So even if the test was fully run and completed, we report errors.
- this solution also does not cover the case where a test is "stuck" in the device and old output appears after flashing with new tests as mentioned here: https://github.com/zephyrproject-rtos/zephyr/issues/44539#issuecomment-1092086252

So, here are the options:
- consider the RunID solution to cover the cases we see with the interrupt tests and architecture related tests that require a clean reset (erase) of the system
- consider change ztest to better account for executed tests and deliver final verdict based on actual tests being run, if no tests are run for whatever reason, it should fail. Ideally we should consider to print a test summary at the end with pass/fail/skip numbers as well, this will provide more confidence about numbers in one or few lines toward the end of the test, so even if console output is dropped, we still have a verdict and do not have to guess.
- the suite name checking becomes a pedantic option we can leave enabled by default but with the option to disable it once we have the above.

FRASTM, 2022-04-11:
> @FRASTM meh... okay... so to be sure, that the source of problem is in `nmi` test could you run once again this command (it exclude `nmi` test):
> 
> ```
> scripts/twister --device-testing --device-serial /dev/ttyACM0 -p nucleo_wb55rg -v -v -j 1 -T tests/arch/arm/arm_hardfault_validation -T tests/arch/arm/arm_interrupt -T tests/arch/arm/arm_irq_advanced_features -T tests/arch/arm/arm_irq_vector_table -T tests/arch/arm/arm_mem_protect -T tests/arch/arm/arm_no_multithreading -T tests/arch/arm/arm_ramfunc -T tests/arch/arm/arm_sw_vector_relay -T tests/arch/arm/arm_thread_swap -T tests/arch/arm/arm_thread_swap_tz -T tests/arch/arm/arm_tz_wrap_func
> ```

It gives a test configurations passed (100.00%),

gopiotr, 2022-04-12:
> the suite name checking becomes a pedantic option we can leave enabled by default but with the option to disable it once we have the above.

@nashif 
https://github.com/zephyrproject-rtos/zephyr/pull/44800

FRASTM, 2022-04-15:
on the main branch c5b59282d6bcbcc5eedaf29b386c90c77036440b 
(with Anas's PR #44800 is merged)
running the twister command on nucleo_w55rg gives a 100% passed:

[twister.log](https://github.com/zephyrproject-rtos/zephyr/files/8497097/twister.log)

gopiotr, 2022-04-21:
@FRASTM are you satisfied with introduced solutions (add `RunID` verification and option `--disable-suite-name-check`)? Is it acceptable to you to run tests with permanently added option `--disable-suite-name-check`?

Is it enough for you? To be honest I don't have any idea what else can be done in this situation. Especially that when I ask you for run Twister with `--west-flash="--erase"` option you said that it cause only "FAILED Flash error". From this I conclude that your boards do not support `west flash --erase` command and adding any "post-erase" actions will not solve your problem.

@nashif what do you think about this case? Should something more be done?

erwango, 2022-04-26:
> @FRASTM are you satisfied with introduced solutions (add `RunID` verification and option `--disable-suite-name-check`)? Is it acceptable to you to run tests with permanently added option `--disable-suite-name-check`?

It works, thanks.

Though, this raises an issue to integrate this in the bench as we're also running V2.7.x branch tests on our bench.
And on a v2.7.x branch, this prevents twister to run.

How to deal with this? I can see 2 solutions for now:
- Patch the bench to detect which version is being tested (based VERSION file)
- Get twister v2.7.0 not to fail on this option and just discard it.

Other idea?

gopiotr, 2022-04-28:
@erwango - I don't see any other way rather than modify Twister's command depend on Zephyr version (taken from VERSION file)

closed 2022-04-28 by gopiotr