zephyrproject-rtos/zephyr #44544: shell_module/sample.shell.shell_module.usb fails for thingy53_nrf5340_cpuapp_ns

nordic-krch, 2022-04-05:
**Describe the bug**
Build ` samples/subsys/shell/shell_module/sample.shell.shell_module.usb` for `thingy53_nrf5340_cpuapp_ns` fails.

**To Reproduce**
`twister -c -p thingy53_nrf5340_cpuapp_ns -s samples/subsys/shell/shell_module/sample.shell.shell_module.usb`

fails with:
```
[ 94%] Linking C executable zephyr_pre0.elf
/home/krch/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: drivers/hwinfo/libdrivers__hwinfo.a(hwinfo_nrf.c.obj): in function `soc_secure_read_deviceid':
/home/krch/vork/zephyr/soc/arm/nordic_nrf/common/./soc_secure.h:37: undefined reference to `soc_secure_mem_read'
```

**Environment (please complete the following information):**
 - Commit SHA: e4aed1cfbd0760ec0af524d1b5796ba0ef38b61b

jfischer-no, 2022-04-05:
@nordic-krch Is it not fixed by https://github.com/zephyrproject-rtos/zephyr/pull/44405/commits/11228d5d0e9da05bf7ccb20391f278af64dd3bb2 ?

nordic-krch, 2022-04-05:
@jfischer-no seems that it not fixed the issue with that board.

carlescufi, 2022-06-01:
This is no longer failing on `main`:

```
$ scripts/twister -c -p thingy53_nrf5340_cpuapp_ns -s samples/subsys/shell/shell_module/sample.shell.shell_module.usb
ZEPHYR_BASE unset, using "/home/carles/src/zephyr/zephyr"
INFO    - Zephyr version: v3.1.0-rc3-14-gbb255bc97512
INFO    - JOBS: 4
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/carles/src/zephyr/zephyr/twister-out/testplan.json
INFO    - Using 'zephyr' toolchain.
INFO    - 1 test scenarios (1 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 62.03 seconds
INFO    - In total 0 test cases were executed, 1 skipped on 1 out of total 456 platforms (0.22%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/carles/src/zephyr/zephyr/twister-out/twister.json
INFO    - Using 'zephyr' toolchain.
INFO    - Writing xunit report /home/carles/src/zephyr/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/carles/src/zephyr/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

closed 2022-06-01 by carlescufi