zephyrproject-rtos/zephyr #44546: Bluetooth: ISO: Provide stream established information

Thalley, 2022-04-05:
**Is your enhancement proposal related to a problem? Please describe.**
Several, if not all, fields of the CIS established or BIG sync established events are useful or needed for the application to properly use isochronous channels (this is especially true for Audio cases). 

**Describe the solution you'd like**
Several, if not all, fields from the events should be stored and be available to the application. 

**Describe alternatives you've considered**
N/A

**Additional context**
The change should also apply for `bt_audio_stream`s.

GatCode, 2022-04-06:
@Thalley, wouldn't it make sense to include the e.g. `bt_hci_evt_le_big_sync_established` / `bt_hci_evt_le_cis_established` structs in the `bt_iso_chan` struct since this struct would be accessible to the application on multiple points (recv callback, connected cb, ...)?

```
/** @brief ISO Channel structure. */
struct bt_iso_chan {
	/** Channel connection reference */
	struct bt_conn			*iso;
	/** Channel operations reference */
	struct bt_iso_chan_ops		*ops;
	/** Channel QoS reference */
	struct bt_iso_chan_qos		*qos;
	enum bt_iso_state		state;
	bt_security_t			required_sec_level;
	/** Node used internally by the stack */
	sys_snode_t node;
};```

Thalley, 2022-04-06:
> @Thalley, wouldn't it make sense to include the e.g. `bt_hci_evt_le_big_sync_established` / `bt_hci_evt_le_cis_established` structs in the `bt_iso_chan` struct since this struct would be accessible to the application on multiple points (recv callback, connected cb, ...)?
> 

Generally we avoid exposing HCI events directly. This is partly due that they are `__packed` (generally avoided in the public API as well), but also that something like the latency is defined as 
```c
	uint8_t  c_latency[3];
	uint8_t  p_latency[3];
```

in `struct bt_hci_evt_le_cis_established`, where it makes a lot more sense to expose this as a `uint32_t` value.

closed 2022-04-26 by carlescufi