zephyrproject-rtos/zephyr #44573: Do we have complete RNDIS stack available for STM32 controller in zephyr ?

enthunilu, 2022-04-06:
**Describe the bug**
Hi, this is not a bug. I could not find correct option to post this question. So, logging ticket under bug category.

I am looking for integration of RNDIS for my STM32 controller. And I saw RNDIS USB class function available under zephyr subsystem 
https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/usb/device/class/netusb

I am new to this. So, not sure how to understand it and check complete support of RNDIS. 
So, I wanted to know does zephyr supports complete stack for RNDIS ?
If yes, where can I get documentation for the same ?
Also, do we have any test environment available to ensure functionality on zephyr ?

My system:
Board: STM32 dev board
Machine: Windows 10

Thanks in advance!!

carlescufi, 2022-04-06:
RNDIS is supported over USB in Zephyr: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/device/class/netusb/function_rndis.c. This is vendor-independent code so as long as STM32 supports USB (it does) then this should work as well.

closed 2022-04-06 by carlescufi

enthunilu, 2022-06-06:
Hello @carlescufi thanks for your support.
If possible can you please share example code or documentation for USB RNDIS?
I am new to this, so wanted to understand how I can use it.

Thanks!!

carlescufi, 2022-06-30:
> Hello @carlescufi thanks for your support. If possible can you please share example code or documentation for USB RNDIS? I am new to this, so wanted to understand how I can use it.
> 
> Thanks!!

I am sorry, I do not really have any insight there. Perhaps @finikorg knows?

finikorg, 2022-06-30:
> > Hello @carlescufi thanks for your support. If possible can you please share example code or documentation for USB RNDIS? I am new to this, so wanted to understand how I can use it.
> > Thanks!!
> 
> I am sorry, I do not really have any insight there. Perhaps @finikorg knows?

Basically @enthunilu, you can take some networking sample and setup netusb, it can be done with overlay file like: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/dumb_http_server/overlay-netusb.conf

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/sockets/dumb_http_server/overlay-netusb.conf#L1-L9

There is also one configuration for RNDIS left for zperf here:

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/zperf/sample.yaml#L24-L30

enthunilu, 2022-07-04:
Thank you @carlescufi and @finikorg. This is a great help to start with.

:-)

enthunilu, 2022-07-22:
Just wanted to share I am able to test usb-rndis functionality by using below example code and 
[https://github.com/zephyrproject-rtos/zephyr/tree/f9739ab41dde9c56e3bc038fe0bebf65b5ac1cfa/samples/net/sockets/dumb_http_server](url)

with some minor changes in code.

Thanks again @finikorg for sharing inputs.

roygedge, 2023-11-08:
Hey!
I'm experiencing the same difficulties with `stm32f411e_disco` , I'm trying to compile the `echo_server` sample with the netusb overlay, however it throws me linking errors such as: 

> /home/c456913/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd:` zephyr/subsys/usb/device/class/netusb/libsubsys__usb__device__class__netusb.a(function_ecm.c.obj): in function `ecm_read_cb':
> /home/c456913/zephyrproject/zephyr/subsys/usb/device/class/netusb/function_ecm.c:295:` undefined reference to `net_pkt_rx_alloc_with_buffer'
> /home/c456913/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd:` /home/c456913/zephyrproject/zephyr/subsys/usb/device/class/netusb/function_ecm.c:302: undefined reference to `net_pkt_write'
> /home/c456913/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /home/c456913/zephyrproject/zephyr/subsys/usb/device/class/netusb/function_ecm.c:304: undefined reference to `net_pkt_unref'
> /home/c456913/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/subsys/usb/device/class/netusb/libsubsys__usb__device__class__netusb.a(function_ecm.c.obj): in function `ecm_send':
> /home/c456913/zephyrproject/zephyr/subsys/usb/device/class/netusb/function_ecm.c:260: undefined reference to `net_pkt_read'

Does that mean this board doesn't support netusb and if so how can I fix that?