zephyrproject-rtos/zephyr #44579: MCC: Discovery cannot complete with success

MariuszSkamra, 2022-04-06:
**Describe the bug**
The Client fails to complete MCS and OTS discovery. The discovery is aborted in the middle due to lack of resources in ATT (to store the `bt_att_req`) and the discovery is aborted after 30 seconds of ATT Timeout (see `req = bt_att_req_alloc(BT_ATT_TIMEOUT);` in `gatt_req_alloc`).

The Media Control Client discovery implementation is not efficient in terms of CCC subscription. The `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` is enabled, so the `ATT: Find Information Request` is used to find CCC's.
The implementation sends bunch of `ATT: Find Information Request` to find a CCC for a given characteristic. Each `ATT: Find Information Response` contains multiple Handle-UUID pairs that could be used to find CCC of multiple characteristics, but are discarded instead. 

I'm not sure whether it's MCC implementation only bug or rather `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` feature that could be more efficient or used differently to be more efficient.

**To Reproduce**
* Use shell with audio.conf `west build -b native_posix  ~/zephyrproject/zephyr/tests/bluetooth/shell/ --pristine -- -DOVERLAY_CONFIG=audio.conf`
* `mcc init`
* Connect to MCS device
* `mcc discover_mcs`

**Expected behavior**
Discovery completes with no error.

**Logs and console output**
```
[00:02:38.782,000] <dbg> bt_gatt: gatt_ccc_discover: CCC Discovery failed (err -12)
[00:02:38.782,000] <dbg> bt_gatt: gatt_read_type: start_handle 0x00e4 end_handle 0x00f0
[00:02:38.782,000] <dbg> bt_att: bt_att_req_alloc: req 0x81a39a0
[00:02:38.782,000] <dbg> bt_att: bt_att_req_free: req 0x81a39a0
[00:02:38.782,000] <dbg> bt_mcc: discover_mcs_char_func: Setup complete for GMCS
[00:02:38.782,000] <dbg> bt_mcc: discover_mcs_char_func: Start discovery of included services
[00:02:38.782,000] <dbg> bt_gatt: gatt_read_type: start_handle 0x00b3 end_handle 0x00f0
[00:02:38.782,000] <dbg> bt_att: bt_att_req_alloc: req 0x81a39a0
[00:02:38.782,000] <dbg> bt_att: bt_att_req_free: req 0x81a39a0
[00:02:38.782,000] <dbg> bt_mcc: discover_mcs_char_func: Discover of included service failed (err -12)
Discovery failed (-12)
```

Regarding inefficient CCC discovery:
```
> ACL Data RX: Handle 64 flags 0x01 dlen 2                                                                                                         #6301 [hci1] 4148.333478
      ATT: Read By Type Response (0x09) len 64
        Attribute data length: 7
        Attribute data list: 9 entries
        Handle: 0x00b4
        Value: 12b500932b
        Name: MCS: Media Player Name (0x2b93)
        Handle: 0x00b7
        Value: 02b800942b
        Name: MCS: Media Player Icon Object ID (0x2b94)
        Handle: 0x00b9
        Value: 02ba00952b
        Name: MCS: Media Player Icon URL (0x2b95)
        Handle: 0x00bb
        Value: 10bc00962b
        Name: MCS: Track Changed (0x2b96)
        Handle: 0x00be
        Value: 12bf00972b
        Name: MCS: Track Title (0x2b97)
        Handle: 0x00c1
        Value: 12c200982b
        Name: MCS: Track Duration (0x2b98)
        Handle: 0x00c4
        Value: 1ec500992b
        Name: MCS: Track Position (0x2b99)
        Handle: 0x00c7
        Value: 1ec8009a2b
        Name: MCS: Playback Speed (0x2b9a)
        Handle: 0x00ca
        Value: 12cb009b2b
        Name: MCS: Seeking Speed (0x2b9b)
         new_handle_db_entry, MCS: Media Player Name handle=0x00b4, v_handle=0x00b5 
         new_handle_db_entry, MCS: Media Player Icon Object ID handle=0x00b7, v_handle=0x00b8 
         new_handle_db_entry, MCS: Media Player Icon URL handle=0x00b9, v_handle=0x00ba 
         new_handle_db_entry, MCS: Track Changed handle=0x00bb, v_handle=0x00bc 
         new_handle_db_entry, MCS: Track Title handle=0x00be, v_handle=0x00bf 
         new_handle_db_entry, MCS: Track Duration handle=0x00c1, v_handle=0x00c2 
         new_handle_db_entry, MCS: Track Position handle=0x00c4, v_handle=0x00c5 
         new_handle_db_entry, MCS: Playback Speed handle=0x00c7, v_handle=0x00c8 
         new_handle_db_entry, MCS: Seeking Speed handle=0x00ca, v_handle=0x00cb 
< ACL Data TX: Handle 64 flags 0x00 dlen 9                                                                                                         #6304 [hci1] 4148.347813
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x00b5-0x00f0
> ACL Data RX: Handle 64 flags 0x02 dlen 66                                                                                                        #6306 [hci1] 4148.429478
      ATT: Find Information Response (0x05) len 61
        Format: UUID-16 (0x01)
        Handle: 0x00b5
        UUID: MCS: Media Player Name (0x2b93)
        Handle: 0x00b6
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00b7
        UUID: Characteristic (0x2803)
        Handle: 0x00b8
        UUID: MCS: Media Player Icon Object ID (0x2b94)
        Handle: 0x00b9
        UUID: Characteristic (0x2803)
        Handle: 0x00ba
        UUID: MCS: Media Player Icon URL (0x2b95)
        Handle: 0x00bb
        UUID: Characteristic (0x2803)
        Handle: 0x00bc
        UUID: MCS: Track Changed (0x2b96)
        Handle: 0x00bd
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00be
        UUID: Characteristic (0x2803)
        Handle: 0x00bf
        UUID: MCS: Track Title (0x2b97)
        Handle: 0x00c0
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00c1
        UUID: Characteristic (0x2803)
        Handle: 0x00c2
        UUID: MCS: Track Duration (0x2b98)
        Handle: 0x00c3
        UUID: Client Characteristic Configuration (0x2902)
< ACL Data TX: Handle 64 flags 0x00 dlen 9                                                                                                         #6308 [hci1] 4148.433407
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x00bc-0x00f0
> ACL Data RX: Handle 64 flags 0x02 dlen 66                                                                                                        #6310 [hci1] 4148.525465
      ATT: Find Information Response (0x05) len 61
        Format: UUID-16 (0x01)
        Handle: 0x00bc
        UUID: MCS: Track Changed (0x2b96)
        Handle: 0x00bd
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00be
        UUID: Characteristic (0x2803)
        Handle: 0x00bf
        UUID: MCS: Track Title (0x2b97)
        Handle: 0x00c0
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00c1
        UUID: Characteristic (0x2803)
        Handle: 0x00c2
        UUID: MCS: Track Duration (0x2b98)
        Handle: 0x00c3
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00c4
        UUID: Characteristic (0x2803)
        Handle: 0x00c5
        UUID: MCS: Track Position (0x2b99)
        Handle: 0x00c6
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00c7
        UUID: Characteristic (0x2803)
        Handle: 0x00c8
        UUID: MCS: Playback Speed (0x2b9a)
        Handle: 0x00c9
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x00ca
        UUID: Characteristic (0x2803)
< ACL Data TX: Handle 64 flags 0x00 dlen 9                                                                                                         #6312 [hci1] 4148.529065
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x00bf-0x00f0
(...)
```

Attached console output and HCI traces.
[mcc_dicovery_hci.log](https://github.com/zephyrproject-rtos/zephyr/files/8425678/mcc_dicovery_hci.log)
[mcc_discovery.log](https://github.com/zephyrproject-rtos/zephyr/files/8425679/mcc_discovery.log)

Thalley, 2022-04-06:
I think the `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` discussion is better fitted for a different issue. `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` will only find a single CCCD, and it will do a discovery for each CCC to subscribe to (of which there are many in MCS). There is room for improvement there, but it isn't any different than us doing `bt_gatt_discover` for each service instead of a single `bt_gatt_discover` for multiple services (assuming that the MTU supports that). 

@MariuszSkamra Was this run with the `/tests/bluetooth/shell/`? 
One thing we haven't considered is that the number of TX buffers required in many of the implementations will increase by the MTU: If the entire MCS service is returned in a single read, then we will likely attempt to subscribe to all the CCCs based on that single read. If the MTU is large (such as 64 in this case), then we should actually probably be able to do a single write to multiple CCCDs. 

The alternative is of course to adapt the less (time) efficient approach of doing
1. Discover service
2. Discover specific characteristic
3. [Optionally] subscribe to specific characteristic
4. Repeat 2 and 3 until entire service characteristic range is discovered

This approach will require less memory as we never need more than a single discover parameter, but it will likewise also increase the number of GATT operations to be performed for discovery. IIRC this is approach of the HAS client, right @MariuszSkamra ?

MariuszSkamra, 2022-04-06:
@Thalley Yes, `/tests/bluetooth/shell/` was used.
@Thalley Indeed, this approach is used in HAS client, but I think more efficient way to do this is such a big service like this is:

1. Discover service
2. Discover all characteristics within service range and store handles.
3. Once the characteristic discovery is complete, subscribe to.

But still there is a possibility that some of the information might be lost in `ATT: Find Information Response`. when `bt_gatt_subscribe` is used with `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` option enabled. 

So instead of that I would suggest using `bt_gatt_discover` with type `BT_GATT_DISCOVER_ATTRIBUTE` and handle range <service start handle, service end handle> to find all of the attributes within range. By doing that one gets all of the characteristic value handles + all of the descriptors. So,

1. Discover service
2. Discover attributes
3. Subscribe to

Thalley, 2022-04-06:
Indeed, that will be more efficient. 

The `CONFIG_BT_GATT_AUTO_DISCOVER_CCC` was defined to be an easy way of doing it, not an efficient way :D 

It will require additional effect for the service client implementation to match the CCID handles with characteristic handles, but will certainly be an optimization (in both air time and memory).

asbjornsabo, 2022-04-21:
The issue seems to be somehow related to the ATT MTU size or the ATT MTU exchange.

The discovery shell command started failing after commit 6eafc64855192fba71b43a95ce85f12c3d5e1b24  Bluetooth: audio: Enable auto ATT MTU Exchange for BASS and unicast client

For the commit before this the discovery command succeeds.  But it fails if "gatt exchange-mtu" is done before the discovery.

The BabbleSim test for mcc does not fail, however, even though that also does the ATT AUTO MTU exchange.

Thalley, 2022-04-21:
> The issue seems to be somehow related to the ATT MTU size or the ATT MTU exchange.
> 
> The discovery shell command started failing after commit [6eafc64](https://github.com/zephyrproject-rtos/zephyr/commit/6eafc64855192fba71b43a95ce85f12c3d5e1b24) Bluetooth: audio: Enable auto ATT MTU Exchange for BASS and unicast client
> 
> For the commit before this the discovery command succeeds. But it fails if "gatt exchange-mtu" is done before the discovery.
> 
> The BabbleSim test for mcc does not fail, however, even though that also does the ATT AUTO MTU exchange.

Correct. 

With a larger MTU, we get more characteristics in a single discovery GATT command response, thus triggering more subscribe requests while handling the discovery response. 

The quick workaround for this is to increase the number of TX buffers to the number of characteristics we'll subscribe to (expensive, and will still cause issues if e.g. doing any other discovery procedure or GATT procedure).

The proper workaround is to do these subscribe requests _after_ discovery as suggested by @MariuszSkamra, which would keep the number of buffers to a minimum, but will require additional handling in the MCC.

closed 2022-07-08 by carlescufi