zephyrproject-rtos/zephyr #44587: Bluetooth: Host: Automatically write the Client Supported Features Characteristic

hermabe, 2022-04-06:
**Is your enhancement proposal related to a problem? Please describe.**
To be able to use certain features like Robust Caching or Multiple Handle Value Notifications, the GATT client must write to the Client Supported Features Characteristic. The host should automatically do this so that the application only needs to enable the relevant Kconfigs to use the features.

**Describe the solution you'd like**
After connection to a GATT server, the Client automatically discovers and writes the relevant bits to the Client Supported Features Characteristic. The application should be able to enable/disable this functionality using a Kconfig. I am thinking something along the lines of:
```Kconfig
config BT_GATT_AUTO_CSF
    bool "Automatically write to the Client Supported Feature Characteristic"
    depends on BT_EATT || BT_GATT_CACHING || BT_GATT_NOTIFY_MULTIPLE
    default BT_EATT || BT_GATT_CACHING || BT_GATT_NOTIFY_MULTIPLE
```

**Describe alternatives you've considered**
Make the feature bit definitions from `gatt.c` public and depend on the application developer to do the correct initialization to enable these features.

**Additional context**
N/A

jori-nordic, 2024-02-09:
> Make the feature bit definitions from gatt.c public and depend on the application developer to do the correct initialization to enable these features.

Since the host does no automatic discovery, I would prefer this to be the solution to this issue
Note that a discovery module would be very helpful. It is a big feature though, and I'd rather we clean up the GATT module before embarking on this.

hermabe, 2024-02-09:
> Note that a discovery module would be very helpful. It is a big feature though, and I'd rather we clean up the GATT module before embarking on this.

Yeah, this is connected to some other enhancement issues relating to discovery, service changed and caching.