zephyrproject-rtos/zephyr #44623: [backport v2.7-branch] backport of #44532 failed

zephyrbot, 2022-04-07:
This issue was created automatically because the backport of #44532 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-44532-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 54e6d8435a8a58a0ea4c393ca93c3956ce8d5097
# Push it to GitHub
git push --set-upstream origin backport-44532-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-44532-to-v2.7-branch`.

cfriedt, 2022-04-07:
@zephyrproject-rtos/release-team 👀

lairdjm, 2022-04-07:
Issue is that because of a different commit, some of the structure in the file changed and so it fails to apply the diff, would be good to see that error in the description. Not sure how to fix that as I can't push to this repo

mbolivar-nordic, 2022-04-07:
> Not sure how to fix that as I can't push to this repo

Just resolve conflicts manually and create a separate backport

closed 2022-06-07 by lairdjm