zephyrproject-rtos/zephyr #44649: Introduce reset api for k_poll signals used in a struct

ddavidebor, 2022-04-07:
**Is your enhancement proposal related to a problem? Please describe.**
It's easy to forget to properly reset a k_poll signal placed in a loop. As a workaround, the following has to be done:

```c
events[0].signal->signaled = 0;
events[0].state = K_POLL_STATE_NOT_READY;
```

This works and is reasonably documented , however, it is in conflict with the principle that users should not fiddle with the content of the event struct. 
**Describe the solution you'd like**

Introduce `void k_poll_signal_loop_reset(struct k_poll_signal *sig)` in addition to  `void k_poll_signal_reset(struct k_poll_signal *sig)`.

The API would behave identically, but also set state to `K_POLL_STATE_NOT_READY`.

Document when it is appropriate to use one instead of the other.

**Describe alternatives you've considered**
status quo. It's ok. 

---

If the Zephyr gods are ok with this, I'd be happy to PR

andyross, 2022-04-07:
No complaints from me if you want to try this.  I will say though that k_poll() needs some more extensive rework than just this, IMHO.  The process you're describing is in fact a race condition in most circumstances, because the user is responsible for making sure that the critical section between that reset and the previous/next calls to k_poll() is atomic vs. whoever is going to be trying to signal it.

Personally I think we actually want to move in the direction of an entirely new level-triggered signal mechanism that automatically resets, then mark the old one deprecated, etc...

But there's no reason not to evolve what we have.

ddavidebor, 2024-02-20:
We've got a new firmware engineer coming soon, I'll task her to sharpen her claws on this issue

ddavidebor, 2024-02-20:
I'll close this issue, if anyone wants to work on if feel free to reopen