zephyrproject-rtos/zephyr #44671: tests-ci : kernel: scheduler: deadline test failed

hakehuang, 2022-04-08:
**Describe the bug**
    deadline test is failed on zephyr-v3.0.0-2262-ge8a369e7da50 on mimxrt685_evk_cm33

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt685_evk_cm33      --sub-test kernel.scheduler
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**

   ```

    *** Booting Zephyr OS build zephyr-v3.0.0-2262-ge8a369e7da50  ***
Running test suite suite_deadline
===================================================================
START - test_deadline
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/deadline/src/main.c:103: test_deadline: (d0 <= d1 is false)
threads ran in wrong order
FAIL - test_deadline in 0.114 seconds
===================================================================
START - test_yield
        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.0.0-2262-ge8a369e7da50

hakehuang, 2022-04-08:
@dleach02 , @danieldegrasse , @mmahadevan108

mmahadevan108, 2022-04-19:
Unable to reproduce this with `zephyr-v3.0.0-2805-gc6b05afc9255`
```
*** Booting Zephyr OS build zephyr-v3.0.0-2805-gc6b05afc9255  ***
Running test suite suite_deadline
===================================================================
START - test_deadline
 PASS - test_deadline in 0.101 seconds
===================================================================
START - test_yield
 PASS - test_yield in 0.101 seconds
===================================================================
START - test_unqueued
 PASS - test_unqueued in 0.151 seconds
===================================================================
Test suite suite_deadline succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

closed 2022-04-20 by hakehuang