zephyrproject-rtos/zephyr #44672: tests-ci : drivers: counter: basic_api test failed

hakehuang, 2022-04-08:
**Describe the bug**
    basic_api test is failed on zephyr-v3.0.0-2262-ge8a369e7da50 on mimxrt685_evk_cm33

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt685_evk_cm33      --sub-test drivers.counter
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**

```
 
build zephyr-v3.0.0-2262-ge8a369e7da50  ***
Running test suite test_counter
===================================================================
START - test_multiple_alarms
Testing CTIMER_0
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:569: test_multiple_alarms_instance: (2 not equal to cnt)
CTIMER_0: Invalid number of callbacks 1 (expected: 2)
FAIL - test_multiple_alarms in 268.463 seconds
===================================================================
START - test_all_channels
Testing CTIMER_0
E: channel already in use
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:636: test_all_channels_instance: (0 not equal to 1)
CTIMER_0: Unexpected error on setting alarm
FAIL - test_all_channels in 0.22 seconds

```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: zephyr-v3.0.0-2262-ge8a369e7da50

hakehuang, 2022-04-08:
@dleach02 @danieldegrasse @mmahadevan108

danieldegrasse, 2022-04-28:
@hakehuang This issue should be able to be closed since #45075 merged, correct?

hakehuang, 2022-04-29:
@danieldegrasse correct

closed 2022-04-29 by hakehuang