zephyrproject-rtos/zephyr #44676: mimxrt1050_evk_qspi freeze when writing to flash

jeppenodgaard, 2022-04-08:
**Describe the bug**
Accessing the flash causes the system to hang forever or crash depending on what application is run.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b mimxrt1050_evk_qspi zephyr/samples/drivers/flash_shell/ && west flash
2. flash page_count

**Expected behavior**
Shell outputs `Flash device contains 2048 pages.`

**Impact**
Sheel outputs
```
[00:00:19.245,025] <err> os: ***** MPU FAULT *****
[00:00:19.245,086] <err> os:   Instruction Access Violation
[00:00:19.245,117] <err> os: r0/a1:  0x6000de48  r1/a2:  0x80002ab8  r2/a3:  0x80002abc
[00:00:19.245,117] <err> os: r3/a4:  0xf011d418 r12/ip:  0x00000000 r14/lr:  0x6000d0cf
[00:00:19.245,117] <err> os:  xpsr:  0x60000000
[00:00:19.245,117] <err> os: Faulting instruction address (r15/pc): 0xf011d418
[00:00:19.245,208] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:19.245,208] <err> os: Current thread: 0x80000620 (shell_uart)
[00:00:19.308,502] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Version 3.0.0 used

**Additional context**
Changing to `zephyr,flash-controller = &is25wp064; ` from https://github.com/zephyrproject-rtos/zephyr/blob/b0089162e91adee685a5d13d724fb60c048b7634/boards/arm/mimxrt1050_evk/mimxrt1050_evk_qspi.dts#L13 seems to resolve the issue. It also aligns with mimxrt1050_evk.dts https://github.com/zephyrproject-rtos/zephyr/blob/f1e75dea501630343bf259cdac3070baa238724b/boards/arm/mimxrt1050_evk/mimxrt1050_evk.dts#L29

mmahadevan108, 2022-04-08:
@jeppenodgaard . Could you please submit a PR with the proposed fix you mentioned.

danieldegrasse, 2022-04-08:
@jeppenodgaard after some internal discussion, we're going to fix this issue as part of a broader PR since it is present on several of our evaluation kits. Are you ok using the fix you have for the time being while that PR is created and tested?

jeppenodgaard, 2022-04-08:
@danieldegrasse Sure! Is there a link to the PR or is it to be created?

danieldegrasse, 2022-04-08:
@jeppenodgaard to be created-likely sometime today or Monday.

danieldegrasse, 2022-04-11:
@jeppenodgaard #44762 should fix the `flash-controller` to use the correct node label

jeppenodgaard, 2022-04-12:
@danieldegrasse thanks, I'll try it out 😊

jeppenodgaard, 2022-04-12:
Cannot reproduce with your branch :+1:

closed 2022-04-12 by jeppenodgaard