zephyrproject-rtos/zephyr #44690: ST kit b_u585i_iot02a and OCTOSPI flash support

JiiZR, 2022-04-08:
**Is your enhancement proposal related to a problem? Please describe.**
ST kit b_u585i_iot02a has flash MX25LM51245GXDI005 connected to the OCTOSPI2. I was not able to find any support for Octo-SPI.  Currently kit's external flash can not be used. 

**Describe the solution you'd like**
Add ST Octo-SPI support to the Zephyr.

**Describe alternatives you've considered**
If Octo-SPI can be used as SPI (MISO/MOSI) then example configuration for MX25LM51245GXDI005 would be nice.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2022-04-08:
@JiiZR Please check https://github.com/zephyrproject-rtos/zephyr/pull/39770.
Note that this is a work in progress, not sure a bout the status though

JiiZR, 2022-04-08:
Thanks, I have missed this even I tried to search all octospi items.

FRASTM, 2022-04-11:
@JiiZR  also take the https://github.com/zephyrproject-rtos/hal_stm32/pull/122

FRASTM, 2022-05-06:
tested with _west build -p auto -b b_u585i_iot02a samples/drivers/spi_flash_
```
*** Booting Zephyr OS build zephyr-v3.0.0-3319-g2e038739fd22  ***

JEDEC OSPI-NOR SPI flash testing
==========================

Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
```
Data read matches data written. Good!!

in SPI/STR or OPI  STR / DTR modes with interrupt 
or with (GP)DMA transfer with https://github.com/FRASTM/zephyr/tree/ospi_dma_u5

closed 2022-05-12 by MaureenHelm