zephyrproject-rtos/zephyr #44691: west sign fails to find header size or padding

thamilv91, 2022-04-08:
I am using west v0.12.0, Zephyr 3.0.99 would like to sign an image for mcuboot.The following command works on my desktop, but fails on linux ubuntu machine: west sign -t imgtool -- --key <key>.pem

=== image configuration:
partition offset: 65536 (0x10000)
partition size: 262144 (0x40000)
rom start offset: 0 (0x0)
=== signing binaries
unsigned bin: /home/thamizh/ncs/zephyr/samples/subsys/mgmt/updatehub/build/zephyr/zephyr.bin
signed bin:   /home/thamizh/ncs/zephyr/samples/subsys/mgmt/updatehub/build/zephyr/zephyr.signed.bin
Usage: imgtool sign [OPTIONS] INFILE OUTFILE

Error: Invalid value for "-H" / "--header-size": Minimum value for -H/--header-size is 32
FATAL ERROR: command exited with status 2: /home/thamizh/.local/bin/imgtool sign --version 0.0.0+0 --align 4 --header-size 0 --slot-size 262144 --key home/thamizh/ncs/bootloader/mcuboot/root-rsa-2048.pem /home/thamizh/ncs/zephyr/samples/subsys/mgmt/updatehub/build/zephyr/zephyr.bin /home/thamizh/ncs/zephyr/samples/subsys/mgmt/updatehub/build/zephyr/zephyr.signed.bin

lairdjm, 2022-04-12:
The header size is defined in the python code as:
```
        # The vector table offset is set in Kconfig:
        vtoff = self.get_cfg(command, build_conf, 'CONFIG_ROM_START_OFFSET')
```
This means it is likely you do not have a ROM start offset defined for the build, did you build with mcuboot support enabled and the correct Kconfig to link it in the application slot?

mbolivar-nordic, 2022-04-12:
>  Zephyr 3.0.99

This is not an actual version number. Please share the exact SHA you're using.

> The following command works on my desktop, but fails on linux ubuntu machine: west sign -t imgtool -- --key .pem

Can you please share exact steps to reproduce? This is just the west sign command, but as @lairdjm says you also need to share the build commands.

thamilv91, 2022-04-16:
> This is not an actual version number. Please share the exact SHA you're using.
>
>
> Can you please share exact steps to reproduce? This is just the west sign command, but as @lairdjm says you also need [...]

Version: SHA-256

**Step:1 mcuboot Build and flash** 
thamizh@Thamizh:~/ncs/bootloader/mcuboot$  
west build -b nrf5340dk_nrf5340_cpuapp  ./boot/zephyr/ 
West flash 

**Step:2 updatehub Build and flash** 
thamizh@Thamizh:~/ncs/zephyr/samples/subsys/mgmt/updatehub$   
west build -b nrf5340dk_nrf5340_cpuapp 
**west 	sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem** 

we got the problem when we execute the last command (please ref below snap)  
[image]

lairdjm, 2022-04-19:
That isn't a commit sha, go to the zephyr directory and do `git log -n 1` and get the commit hash from that. Works for me:
```
cmake -GNinja -DBOARD=nrf5340dk_nrf5340_cpuapp -DCONFIG_TEST_RANDOM_GENERATOR=y ..
...
[231/231] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      130368 B       256 KB     49.73%
            SRAM:       52775 B       448 KB     11.50%
        IDT_LIST:          0 GB         2 KB      0.00%

west sign -t imgtool
=== image configuration:
partition offset: 65536 (0x10000)
partition size: 262144 (0x40000)
rom start offset: 512 (0x200)
=== signing binaries
unsigned bin: /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.bin
signed bin:   /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.signed.bin
unsigned hex: /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.hex
signed hex:   /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.signed.hex

west sign -t imgtool -- --key /tmp/aa/bootloader/mcuboot/root-rsa-2048.pem
=== image configuration:
partition offset: 65536 (0x10000)
partition size: 262144 (0x40000)
rom start offset: 512 (0x200)
=== signing binaries
unsigned bin: /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.bin
signed bin:   /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.signed.bin
unsigned hex: /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.hex
signed hex:   /tmp/bb/zephyr/samples/subsys/mgmt/updatehub/_AA/zephyr/zephyr.signed.hex
```
As can be seen from your output, the rom start offset is 0, which means it is not leaving space for an mcuboot header

mbolivar-nordic, 2022-05-25:
I'm closing since there has been no response. @thamilv91 please reopen this issue if you believe there is a bug here.

closed 2022-05-25 by mbolivar-nordic