zephyrproject-rtos/zephyr #44701: advertising with multiple advertising sets fails with BT_HCI_ERR_MEM_CAPACITY_EXCEEDED

CodingGhost, 2022-04-08:
I am trying wo advertise on 1M and coded phy at the same time, but the second advertisement fails with BT_HCI_ERR_MEM_CAPACITY_EXCEEDED
I am on Zephyr 2.7.1
and have set 
```
CONFIG_BT_CTLR_ADV_EXT=y
CONFIG_BT_EXT_ADV=y
CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_USER_PHY_UPDATE=y
CONFIG_BT_EXT_ADV_LEGACY_SUPPORT=y
CONFIG_BT_EXT_ADV_MAX_ADV_SET=2
CONFIG_BT_CTLR_ADV_SET=2
CONFIG_BT_CTLR_ADV_AUX_SET=2
```

this is the function to start one advertisement;
```
nwv_error_t BluetoothHandler::startExtendedAdvertising(int txpwr)
{
    Utils utils;
    int err = 0;
    err = set_tx_power(BT_HCI_VS_LL_HANDLE_TYPE_ADV, 0, txpwr);
    ERRCHECK((nwv_error_t)err);
    const struct bt_le_adv_param extAdvParams
    {
        .id = 0,
        .sid = 0,
        .options = BT_LE_ADV_OPT_CONNECTABLE | BT_LE_ADV_OPT_USE_NAME | BT_LE_ADV_OPT_EXT_ADV | BT_LE_ADV_OPT_CODED,
        .interval_min = BT_GAP_ADV_FAST_INT_MIN_2,
        .interval_max = BT_GAP_ADV_FAST_INT_MAX_2,
    };
    static struct bt_le_ext_adv_start_param extAdvStartParams
    {
        .timeout = 0,
        .num_events = 0,
    };

    static uint8_t mfg_data[] = {'N', 'W', 'V'}; // not sure if we need this..
    static const struct bt_data ad[] = {
        BT_DATA_BYTES(BT_DATA_FLAGS, (BT_LE_AD_GENERAL | BT_LE_AD_NO_BREDR)),
        BT_DATA(BT_DATA_MANUFACTURER_DATA, mfg_data, 3),
        BT_DATA_BYTES(BT_DATA_UUID128_ALL, 0x84, 0xaa, 0x60, 0x74, 0x52, 0x8a, 0x8b, 0x86, 0xd3, 0x4c, 0xb7, 0x1d, 0x1d, 0xdc, 0x53, 0x8d)};

    if (!extAdv)
    {
        err = bt_le_ext_adv_create(&extAdvParams, NULL, &extAdv);
        if (err)
        {
            LOG_ERR("Failed to create ext advertising set (err %d)\n", err);
        }
    }

    err = bt_le_ext_adv_start(extAdv, &extAdvStartParams);

    if (err)
    {
        if (err == -EALREADY || err == -ENOMEM)
        {
            LOG_DBG("extendet advertising already started");
            return NWV_ERROR_NONE;
        }
        LOG_ERR("Advertising failed to start (err %d)\n", err);
        return static_cast<nwv_error_t>(err);
    }
    return NWV_ERROR_NONE;
}
```

this one is for the next one:

```
nwv_error_t BluetoothHandler::startLegacyAdvertising(int txpwr)
{
    Utils utils;
    int err = 0;
    err = set_tx_power(BT_HCI_VS_LL_HANDLE_TYPE_ADV, 0, txpwr);
    ERRCHECK((nwv_error_t)err);
    const struct bt_le_adv_param legacyAdvParams
    {
        .id = 0,
        .sid = 1,
        .options = BT_LE_ADV_OPT_CONNECTABLE | BT_LE_ADV_OPT_USE_NAME,
        .interval_min = BT_GAP_ADV_FAST_INT_MIN_2,
        .interval_max = BT_GAP_ADV_FAST_INT_MAX_2,
    };
    static struct bt_le_ext_adv_start_param extAdvStartParams
    {
        .timeout = 0,
        .num_events = 0,
    };

    static uint8_t mfg_data[] = {'N', 'W', 'V'}; // not sure if we need this..
    static const struct bt_data ad[] = {
        BT_DATA_BYTES(BT_DATA_FLAGS, (BT_LE_AD_GENERAL | BT_LE_AD_NO_BREDR)),
        BT_DATA(BT_DATA_MANUFACTURER_DATA, mfg_data, 3),
        BT_DATA_BYTES(BT_DATA_UUID128_ALL, 0x84, 0xaa, 0x60, 0x74, 0x52, 0x8a, 0x8b, 0x86, 0xd3, 0x4c, 0xb7, 0x1d, 0x1d, 0xdc, 0x53, 0x8d)};

    if (!legacyAdv)
    {
        err = bt_le_ext_adv_create(&legacyAdvParams, NULL, &legacyAdv);
        if (err)
        {
            LOG_ERR("Failed to create le advertising set (err %d)\n", err);
        }
    }

    err = bt_le_ext_adv_start(legacyAdv, &extAdvStartParams);

    if (err)
    {
        if (err == -EALREADY || err == -ENOMEM)
        {
            LOG_DBG("advertising already started");
            return NWV_ERROR_NONE;
        }
        LOG_ERR("legacy Advertising failed to start (err %d)\n", err);
        return static_cast<nwv_error_t>(err);
    }

    return NWV_ERROR_NONE;
}
```
I am getting <wrn> bt_hci_core: opcode 0x2039 status 0x07 on the second advertisement.

if I stop one advertisement, then start the other one, everything works as expected.

cvinayak, 2022-04-12:
Add the below to increase Data buffers, and let me know. Also, try to check on latest upstream/main whether if you face the same issue.

```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_ADV_DATA_BUF_MAX=6
```

carlescufi, 2022-04-21:
Closing as per comment above, please reopen if you still see the issue.

closed 2022-04-21 by carlescufi

CodingGhost, 2022-04-21:
I just tested it, but the Problem did not change. I currently cannot test on the latest upstream due to some complications with our setup, but I am still on the LTS I guess? its v20701.

CodingGhost, 2022-04-21:
CONFIG_BT_MAX_CONN=2

cvinayak, 2022-04-21:
ok, that does not explain the error. Advertising Extensions support is experimental in v2.7, you should try the latest upstream which had many changes/fixes.

cvinayak, 2022-04-23:
Multiple advertising sets with legacy and extended advertising seems to work, check this chat snippet answered by @Thalley 

[image]