zephyrproject-rtos/zephyr #44704: Bootloader linking error building for RPI_PICO

arturohernandez10, 2022-04-08:
**Describe the bug**

When compiling the blinky sample for rpi_pico in windows using the main branch commit: a52aa7f2ab98cdd5da248adc405dca515c32c9c2. I get the following error:

```
c:/progra~1/zephyr~1.0/arm-ze~1/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.exe: unrecognized option '--major-image-version'
```

In the `rules.ninja` file I found the following code:

```
#############################################
# Rule for linking ASM executable.

rule ASM_EXECUTABLE_LINKER__bootstage2
  command = cmd.exe /C "$PRE_LINK && C:\PROGRA~1\ZEPHYR~1.0\ARM-ZE~1\bin\ARM-ZE~4.EXE $FLAGS $LINK_FLAGS $in -o $TARGET_FILE -Wl,--out-implib,$TARGET_IMPLIB -Wl,--major-image-version,0,--minor-image-version,0 $LINK_PATH $LINK_LIBRARIES && $POST_BUILD"
  description = Linking ASM executable $TARGET_FILE
  restat = $RESTAT 
```

I could not find where the major and minor version flags come from, eventually I removed and the compilation continued as normal. I got a binary file. I was able to throw it in the raspberry pico and the led started blinking as expected.

**To Reproduce**
1. Install zephyr on a windows computer
2. west build -p auto -b rpi_pico samples\basic\blinky
4. See error

**Expected behavior**
See a completed completed compilation message

**Impact**
It is annoying having to modify the ninja.rules file every time.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK 
 - Commit SHA: a52aa7f2ab98cdd5da248adc405dca515c32c9c2

**Logs and console output**
```
C:\Projects\zephyrproject\zephyr (main -> origin)
λ west build -p auto -b rpi_pico samples\basic\blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Projects/zephyrproject/zephyr/samples/basic/blinky
-- Found Python3: C:/Python38/python.exe (found suitable exact version "3.8.0") found components: Interpreter
-- Cache files will be written to: C:/Projects/zephyrproject/zephyr/.cache
-- Zephyr version: 3.0.99 (C:/Projects/zephyrproject/zephyr)
-- Found west (found suitable version "0.13.0", minimum required is "0.7.1")
-- Board: rpi_pico
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.0 (C:/Program Files/zephyr-sdk-0.14.0)
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.0 (C:/Program Files/zephyr-sdk-0.14.0)
-- Found BOARD.dts: C:/Projects/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico.dts
-- Generated zephyr.dts: C:/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Projects/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/Projects/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: C:/Projects/zephyrproject/zephyr/build/zephyr/dts.cmake
C:/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts:60.34-65.5: Warning (simple_bus_reg): /soc/peripheral-clk: missing or empty reg/ranges property
C:/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts:66.26-71.5: Warning (simple_bus_reg): /soc/system-clk: missing or empty reg/ranges property
C:/Projects/zephyrproject/zephyr/build/zephyr/zephyr.dts:81.36-112.5: Warning (unique_unit_address_if_enabled): /soc/pin-controller@40014000: duplicate unit-address (also used in node /soc/gpio@40014000)
Parsing C:/Projects/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Projects/zephyrproject/zephyr/boards/arm/rpi_pico/rpi_pico_defconfig'
Merged configuration 'C:/Projects/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Projects/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Projects/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Projects/zephyrproject/zephyr/build
-- west build: building application
[4/158] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (C:/Projects/zephyrproject/zephyr), build: zephyr-v3.0.0-2353-ga52aa7f2ab98
[93/158] Performing configure step for 'second_stage_bootloader'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/zephyr-sdk-0.14.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Projects/zephyrproject/zephyr/build/bootloader
[143/158] Performing build step for 'second_stage_bootloader'
FAILED: modules/hal_rpi_pico/second_stage_bootloader-prefix/src/second_stage_bootloader-stamp/second_stage_bootloader-build bootloader/boot_stage2.S C:/Projects/zephyrproject/zephyr/build/modules/hal_rpi_pico/second_stage_bootloader-prefix/src/second_stage_bootloader-stamp/second_stage_bootloader-build C:/Projects/zephyrproject/zephyr/build/bootloader/boot_stage2.S
cmd.exe /C "cd /D C:\Projects\zephyrproject\zephyr\build\bootloader && "C:\Program Files\CMake\bin\cmake.exe" --build ."
[1/2] Building ASM object CMakeFiles/boot_stage2.dir/C_/Projects/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_stage2/boot2_w25q080.S.obj
[2/2] Linking ASM executable boot_stage2.exe
FAILED: boot_stage2.exe boot_stage2.bin boot_stage2.S C:/Projects/zephyrproject/zephyr/build/bootloader/boot_stage2.bin C:/Projects/zephyrproject/zephyr/build/bootloader/boot_stage2.S
cmd.exe /C "cd . && C:\PROGRA~1\ZEPHYR~1.0\ARM-ZE~1\bin\ARM-ZE~4.EXE  -nostartfiles --specs=nosys.specs -Wl,--script=C:/Projects/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_stage2/boot_stage2.ld CMakeFiles/boot_stage2.dir/C_/Projects/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_stage2/boot2_w25q080.S.obj -o boot_stage2.exe -Wl,--out-implib,libboot_stage2.dll.a -Wl,--major-image-version,0,--minor-image-version,0   && cmd.exe /C "cd /D C:\Projects\zephyrproject\zephyr\build\bootloader && "C:\Program Files\zephyr-sdk-0.14.0\arm-zephyr-eabi\bin\arm-zephyr-eabi-objcopy.exe" -Obinary C:/Projects/zephyrproject/zephyr/build/bootloader/boot_stage2.exe boot_stage2.bin && cd /D C:\Projects\zephyrproject\zephyr\build\bootloader && C:\Python38\python.exe C:/Projects/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_stage2/pad_checksum -s 0xffffffff boot_stage2.bin boot_stage2.S""
c:/progra~1/zephyr~1.0/arm-ze~1/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.exe: unrecognized option '--major-image-version'
c:/progra~1/zephyr~1.0/arm-ze~1/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.exe: use the --help option for usage information
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Projects\zephyrproject\zephyr\build'
```

stephanosio, 2022-04-08:
cc @yonsch

yonsch, 2022-04-10:
Hi @arturohernandez10 , from a quick search online, it seems like this is a problem with the build environment. Since I don't develop on Windows, and don't have a Windows setup, I can't point you directly to the problem.
Did you install Zephyr exactly as described in the getting started guide? Do you get the same error for other ARM boards, or does it happen only for the rpi pico?

rivimey, 2022-04-24:
I think I may have the same problem. Asked Q on Discord: https://discord.com/channels/720317445772017664/938474761405726800/967870175351275521
I'm using MacOS not windows. As for whether I installed it right, at this point I don't know! It's been so complicated to figure out all the bits&pieces.

PR #45271 (2022-05-05): rpi_pico: Fix compilation for non Linux systems
On non Linux system, the compilation failed because the main
CMake did not pass the CMAKE_SYSTEM_NAME to the bootloader's
CMake. This caused the variable to be empty, instead of being
"Generic".

Fixes #44704
closed 2022-05-05 by MaureenHelm