zephyrproject-rtos/zephyr #44713: A Study about Issue Priority Prediction

12csw1, 2022-04-09:
We are doing research about “Issue Priority Prediction". We have developed a tool to automatically label the priority of an issue. The basic idea is to employ Bert model to encode the textual and code descriptions of the issue, and incorporate the issue’s attributes in the model, then apply multi-task learning to facilitate other information boosting the prediction performance. 
We have conducted experiments on four GitHub projects with 21700 issues, and results show that 79 accuracy can be achieved. In terms of zephyr project, it can reach 72 f1.
Our tool shows that  issue #44018 and issue #44704 has a very large possibility to be labeled as priority: medium.
Could you please tell us whether it should be labeled as priority: medium.?
And if you have any comments or suggestions with this tool, could you please tell us? 
Thank you so much.

ddavidebor, 2022-04-09:
I think https://github.com/zephyrproject-rtos/zephyr/issues/44704 is quite low priority, as the RP2040 support is still somewhat new and experimental.

12csw1, 2022-04-11:
> I think #44704 is quite low priority, as the RP2040 support is still somewhat new and experimental.

Thank you very much for your comment. This may be due to the fact that our model does not capture the deeper information in this issue. Do you have any other suggestions for our model?