zephyrproject-rtos/zephyr #44730: zcbor ARRAY_SIZE conflict with zephyr include

lairdjm, 2022-04-11:
**Describe the bug**
Seen whilst updating a mcumgr PR:
```
In file included from /tmp/aa/zephyr/include/zephyr/kernel/sched_priq.h:9,
                 from /tmp/aa/zephyr/include/zephyr/kernel_includes.h:23,
                 from /tmp/aa/zephyr/include/zephyr/kernel.h:17,
                 from /tmp/aa/zephyr/include/zephyr/zephyr.h:18,
                 from /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/cmd/fs_mgmt/include/fs_mgmt/hash_checksum_mgmt.h:11,
                 from /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/cmd/fs_mgmt/src/hash_checksum_mgmt.c:10:
/tmp/aa/zephyr/include/zephyr/sys/util.h:108: warning: "ARRAY_SIZE" redefined
  108 | #define ARRAY_SIZE(array) \
      | 
In file included from /tmp/aa/zephyr/include/zephyr/mgmt/mcumgr/buf.h:11,
                 from /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/mgmt/include/mgmt/mgmt.h:11,
                 from /tmp/aa/zephyr/subsys/mgmt/mcumgr/lib/cmd/fs_mgmt/src/hash_checksum_mgmt.c:9:
/tmp/aa/modules/lib/zcbor/include/zcbor_common.h:74: note: this is the location of the previous definition
   74 | #define ARRAY_SIZE(arr) (sizeof(arr) / sizeof((arr)[0]))
```

**To Reproduce**
Build something using both includes

**Expected behavior**
No warning to be emitted

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.0
 - Commit SHA or Version used: Adding commits on top of 70c48417c7a0d7db139accc2368bec7f4aa9c762

de-nordic, 2022-04-11:
Reverse order of includes. the zcbor does have #fndef guard, over ARRAY_SIZE definition, that the util.h, from Zephyr, lacks.

de-nordic, 2022-04-11:
I got hit by the same problem, but it is not unique to zcbor as ARRAY_SIZE is now redefined 4 times within modules.

lairdjm, 2022-04-11:
Reversing order and adding `#include <zephyr.h>` to files without it resolves it

closed 2022-04-11 by lairdjm