zephyrproject-rtos/zephyr #44734: regression in GATT/SR/GAS/BV-06-C qualification test case

sjanc, 2022-04-11:
Following commit breaks GATT/SR/GAS/BV-06-C qualification test.

commit 60d42966e82610aa1a13caf888573bf056486140
Author: Herman Berget <herman.berget@nordicsemi.no>
Date:   Tue Feb 15 14:31:52 2022 +0100

    Bluetooth: Host: Update change-aware requirements for Bluetooth 5.3
    
    Core spec version 5.2 introduced EATT, and multiple ATT channels on one
    connection is now possible. The requirements for when a client
    becomes change-aware were updated to reflect this.
    
    Signed-off-by: Herman Berget <herman.berget@nordicsemi.no>

carlescufi, 2022-04-11:
@hermabe can you please send a fix?

hermabe, 2022-04-11:
> @hermabe can you please send a fix?

Investigating

sjanc, 2022-04-11:
FYI GATT/SR/GAS/BV-07-C is also affected by this

hermabe, 2022-04-11:
The `CF_OUT_OF_SYNC` flag was not set correctly. With https://github.com/hermabe/zephyr/commit/ef171f6e3f01fe56cc12eef6c26b5c03e4c1c7e7, the behaviour should be the same as it was before https://github.com/zephyrproject-rtos/zephyr/commit/60d42966e82610aa1a13caf888573bf056486140 was merged when no EATT channels are connected.

However, I think the proper fix is to not check the `CF_OUT_OF_SYNC` flag here at all, and unconditionally set the client as change-aware. This matches better with the quote from the spec. (https://github.com/hermabe/zephyr/commit/feaa03b96bd3a6d0fc5f2df37f134a737b8bbdc3)

carlescufi, 2022-04-21:
Fixed by #44648

closed 2022-04-21 by carlescufi