zephyrproject-rtos/zephyr #44738: CSIP: Allow user to register for the different CSIS instances

MariuszSkamra, 2022-04-11:
**Is your enhancement proposal related to a problem? Please describe.**
There are 2 problems with current CSIS/P APIs.

1. As per specification, each CSIS instance should have context, and this context is represented by the service which includes CSIS instance. Each context should be possible to handle by different application. For the moment, there can be only one user of the CSIS module and there is no way to learn about the context of the CSIS.

2. For LE Audio case we care about CAS context. This context can be used by different LeAudio profiles e.g. HAP, VC, TMAP, Broadcast Assistant. In order to provide it to this user, it is probably CAP who should provide the list of the CSIS members sorted by rank (in both directions)

Thalley, 2022-04-26:
For the server side, you can see the way CSIS is handled with CAP in https://github.com/zephyrproject-rtos/zephyr/pull/44522

Thalley, 2024-01-05:
@MariuszSkamra can you please respond to the above? I'm not sure what is being requested here

Thalley, 2024-01-23:
Closing due to missing response, and that I think what is being requested already exists.

closed 2024-01-23 by Thalley