zephyrproject-rtos/zephyr #44783: stm32h7: zephyr flash-controller support on M4 core

GiancarloPerroni, 2022-04-12:
Good Morning,

trying to use external flash with M4 core I realised that is not supported yet: 

- I would ask when the implementation of flash driver for M4 is planned?
- compared to M7 flash driver what is still missing to have the driver working for M4?

Thank you in advance.

FRASTM, 2022-04-12:
@GiancarloPerroni   Thanks for your request.
There is no plan to adapt the flash driver for the CM4 of the stm32h7 serie. At this time the drivers simply returns 
```
#ifdef CONFIG_CPU_CORTEX_M4
#error Flash driver on M4 core is not supported yet
#endif
```
It does not seems to be far from the existing flash_stm32h7x.c  driver . There is probably few things to control:  
- a BOOT bit in the flash register, 
- HW semaphore (CFG_HW_FLASH_SEMID) to protect flash  access 
- the mapping to define (if required splitting  / sharing the flash area between both CPUs to avoid overlapping ; typically one bank per CPU). 

There are [stm32cube examples](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects) that could help. 

Your contribution would be appreciated for this project.

GiancarloPerroni, 2022-04-12:
@FRASTM Thank you for your attention and information.

ballanor, 2022-04-26:
Hello everyone.
I'm facing the same issue but, maybe, with a different use case.
I'm basically working with an external SPI NOR Flash.
Everything is working fine on M7 core, but, wehen I try to switch to M4 core, the driver throws an exception as suggested by @FRASTM in https://github.com/zephyrproject-rtos/zephyr/issues/44783#issuecomment-1096266036

If I comment out:
```
// #ifdef CONFIG_CPU_CORTEX_M4 
// #error Flash driver on M4 core is not supported yet
// #endif
```
everything is work as expected.
I think, the reason is that I'm not using the internal flash.

This is part of my configuration file:
```
# flash config
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_STM32_QSPI=y
```
The problem in my case is that CONFIG_FLASH_STM32_QSPI depends on CONFIG_FLASH.
Is there a way to remove this dependency?
Is there any specific documentation on this topic?

Even if the (internal) flash driver is not working, the driver for the external QSPI flash is working fine.

I have a quick fix but I would like to contribute.

Trying to add support for M4 core to the flash driver, the very first issue I've found is that:
`#define REAL_FLASH_SIZE_KB	KB(LL_GetFlashSize())`
will produce a BUS FAULT in M4.
```
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x1ff1e880
E: r0/a1:  0x08130b30  r1/a2:  0x24010bec  r2/a3:  0x24010be8
E: r3/a4:  0x1ff1e000 r12/ip:  0x00000000 r14/lr:  0x081130cb
E:  xpsr:  0x610f0000
```
The reason is that `LL_GetFlashSize()` is trying to read memory at 0x1ff1e880

```
__STATIC_INLINE uint32_t LL_GetFlashSize(void)
{
  return (uint16_t)(READ_REG(*((uint32_t *)FLASHSIZE_BASE_ADDRESS)));
}
```
but M4 can not read memory in that range as stated in reference manual
[image]

Any help will be much appreciated 
Regards,

FRASTM, 2024-06-17:
@GiancarloPerroni can you please look at https://github.com/zephyrproject-rtos/zephyr/pull/74121. 
Does it answer your request ?

mkaricheri, 2024-07-02:
> @GiancarloPerroni can you please look at #74121. Does it answer your request ?

Thanks @FRASTM!  Could you review my PR and do a test on your platform? That would help.  We use Flash driver on M4 and faced the issue. So would like to contribute the fix to the community through my PR above. I think the PR addresses the issue in discussion.

FRASTM, 2024-09-26:
Could we close this issue as the PR mentioned is now merged ?

closed 2024-12-17 by FRASTM