zephyrproject-rtos/zephyr #44811: What is the right way to make STRINGIFY work?

vincemulhollon, 2022-04-12:
**Describe the bug**
A clear and concise description of what the bug is.

I've recently done a git pull and west update to the latest.

I have a compilation error:

In file included from /home/vince/zephyrproject/zephyr/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_state.c:16:
/home/vince/zephyrproject/zephyr/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_state.c: In function 'img_mgmt_state_write':
/home/vince/zephyrproject/zephyr/subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h:37:13: warning: implicit declaration of function 'STRINGIFY' [-Wimplicit-function-declaration]
   37 |    .value = STRINGIFY(k),   \
      |             ^~~~~~~~~
/home/vince/zephyrproject/zephyr/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_state.c:276:3: note: in expansion of macro 'ZCBOR_MAP_DECODE_KEY_VAL'
  276 |   ZCBOR_MAP_DECODE_KEY_VAL(hash, zcbor_bstr_decode, &zhash),
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/vince/zephyrproject/zephyr/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt_state.c:276:28: error: 'hash' undeclared (first use in this function); did you mean 'zhash'?
  276 |   ZCBOR_MAP_DECODE_KEY_VAL(hash, zcbor_bstr_decode, &zhash),
      |                            ^~~~

Well, OK, fine, we have no STRINGIFY() so the macro expansion does not work because the macro is trying to turn those bare words like "hash" into a string "hash" and it can't because it has no STRINGIFY.

subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt.c

calls that ZCBOR_MAP_DECODE_KEY macro which is defined in:

#include "zcbor_bulk/zcbor_bulk_priv.h"

and that apparently does not know how to STRINGIFY so around line 13 of that file, I do know where I can find a STRINGIFY and I add:

#include <zephyr/toolchain/common.h>

and now compilation succeeds with no errors.

So my git diff looks like this:

vince@VINCE:~/zephyrproject/zephyr$ git diff
diff --git a/subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h b/subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h
index a15511817c..cf3647a709 100644
--- a/subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h
+++ b/subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h
@@ -10,6 +10,8 @@
 extern "C" {
 #endif
 
+#include <zephyr/toolchain/common.h>
+
 /** @cond INTERNAL_HIDDEN */
 
 struct zcbor_map_decode_key_val {
vince@VINCE:~/zephyrproject/zephyr$ 

I don't understand enough of ZCBOR_MAP_DECODE_KEY_VAL to know what's going on, but it clearly wants to STRINGIFY something so I gave it a STRINGIFY and my firmware seems to work.  I don't know if this is the "right" way to fix this; but it does compile...

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

No errors upon compilation.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

Could not compile Zephyr until I added that include line to subsys/mgmt/mcumgr/lib/util/include/zcbor_bulk/zcbor_bulk_priv.h which may or may not be the "right" way to fix an error like that.  But it did work.

**Logs and console output**

**Environment (please complete the following information):**
Running on Win10 in a WSL running Ubuntu (long story) but I don't think STRINGIFY() not being defined would be platform dependent.

**Additional context**

Anyway best of luck and have a great day.

stephanosio, 2022-04-13:
Please specify how to reproduce this issue as required by this issue template.

de-nordic, 2022-04-13:
I have built the smp_svr for nrf52840dk with all mcumgr commands enabled and no issue.
There is in-review unit test for the zcbor_bulk functions here https://github.com/zephyrproject-rtos/zephyr/pull/44695, and I was able to build it  with default gcc compiler, `west build -d bulk_test -b native_posix zephyr/tests/subsys/mgmt/zcbor_bulk/`, and llvm, `ZEPHYR_TOOLCHAIN_VARIANT=llvm west build -d bulk_test -b native_posix zephyr/tests/subsys/mgmt/zcbor_bulk/`

de-nordic, 2022-04-13:
@vincemulhollon is the https://github.com/zephyrproject-rtos/zephyr/pull/44821 fixing the problem for you?

de-nordic, 2022-04-13:
This can be triggered when switching to `CONFIG__NEWLIB_LIBC`.

vincemulhollon, 2022-04-13:
> This can be triggered when switching to `CONFIG__NEWLIB_LIBC`.

Confirmed.  I was unaware of the cause, but I can verify my prj.conf for this project does contain:

CONFIG_NEWLIB_LIBC=y

CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

Also it compiled just fine "last week" or so and the problem only began after a git pull / west update yesterday.  Which would fit in with the recent change to use zcbor (img mgmt used to use a different library for that purpose until last week or so)

de-nordic, 2022-04-13:
@vincemulhollon can you check if https://github.com/zephyrproject-rtos/zephyr/pull/44821 fixes the problem for you?

commit 4e03e341 (2022-04-13): mgmt/mcumgr/lib: Add missing inclusion of toolchain.h

The zcbor_bulk_priv.h uses STRINGIFY for some of definitions
and, due to lack of inclusion of the toolchain.h, that was causing
compilation errors when NEWLIB would be selected.

Fixes #44811.
closed 2022-04-13 by mbolivar-nordic